teckel12/LuaTelemetry #203: Add French support

d00bld0ze, 2018-10-22:
### language translation file lang_fr.lua + fr wav folder

[lang_fr.lua + .wav.zip](https://github.com/iNavFlight/LuaTelemetry/files/2501474/lang_fr.lua.%2B.wav.zip)
Unzip to SCRIPTS/TELEMETRY/INav
A décompresser dans SCRIPTS/TELEMETRY/INav
## ATTENTION
### Partially tested / Partiellement testé
-Need other French speaking feedback for the translation of some words
-Besoin du feedback d'autre personnes parlant français pour la traduction de certains termes

teckel12, 2018-10-22:
@d00bld0ze Thanks!  Will check into this once (and if) Github starts working again.

teckel12, 2018-11-07:
@d00bld0ze See #206

teckel12, 2018-11-11:
@d00bld0ze French language support has been added to the development build.  To upgrade, follow the [Upgrade to Development Build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) instructions.  If you could test and report back that would be great so I can get v1.4.3 released.  Thanks!

teckel12, 2018-11-14:
Added in v1.4.3 release https://github.com/iNavFlight/LuaTelemetry/releases/tag/1.4.3

closed 2018-11-14 by teckel12