teckel12/LuaTelemetry #204: pull PIDs after autotune

vavisimo, 2018-10-26:
Hi

Would it be possible to pull PIDs after autotune and landing, to back em up ?
like The BF Taranis Lua Script ?

maybe future option to implement over Tx to write PIDs to controller, but that may be dangerous ;)

for now i have to use Bluetooth+ mobile to get it, this would make it - just a thought....

teckel12, 2018-10-28:
@vavisimo Why not just use the Betaflight PID Lua Script?

I've looked into reading/writing values over MSP like the BF PID script, but it would take way too much memory do both that and INAV Lua Telemetry's core goal and functionality. There's just not enough available RAM on the transmitter to do this much.

With some clever programming it may be possible to fit it all in available RAM, but not from my perspective. I'd accept a pull request that accomplished it, but it's out of scope from my perspective of the INAV Lua Telemetry project. Especially considering there's already a script that does it, and it there was enough RAM available on the transmitter you could do both right now.

vavisimo, 2018-10-28:
so is the Betaflight PID Lua Script working with inav? if yes than im all good, i have to try

teckel12, 2018-10-28:
@vavisimo Not sure it shows all the PIDs for INAV vs BF, but it does work with INAV.

I'd say if it doesn't totally work with INAV, the solution is to create an INAV fork. Not much would need to change, just the INAV specific PIDs (if there ever are any).

I've already created a fork that removes all the PID adjustments and is only the VTx control. The reason is the limited memory on the transmitter, so both Lua Telemetry and VTx control can both run at the same time. Even this is with limited success as both scripts are so memory hungry and there's just so little memory available.

vavisimo, 2018-10-29:
yea I thought that ... im not much into coding :( maybe i find someone to fork it or look into. But I still think you are the best person to ask first... :) not needed asap...maybe as time goes and you look int that script to fork it ;)

closed 2018-11-14 by stale[bot]