teckel12/LuaTelemetry #207: Add spanish support

fmartinezo, 2018-11-10:
### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

[lang_es.lua.txt](https://github.com/iNavFlight/LuaTelemetry/files/2568420/lang_es.lua.txt)
[es-taranis.zip](https://github.com/iNavFlight/LuaTelemetry/files/2568422/es-taranis.zip)

fmartinezo, 2018-11-10:
This is a feature request to add spanish language. I enclose lang.es.lua and one zip with all voices

teckel12, 2018-11-10:
@fmartinezo Awesome! Will add this later today or tomorrow.

fmartinezo, 2018-11-10:
Double check file lang_es_lua.txt I dot know lua and it is a copy of lua.de translated to spanish.Some terms in the voices and the text I have left in English because they are shorter and their translation into Spanish is complicated to express with a single or two words and are terms well known by aeromodelists. Pej "waypoint", "poshold", etc

teckel12, 2018-11-11:
@fmartinezo See #208

teckel12, 2018-11-11:
@fmartinezo Spanish language support was added to the development build.  To upgrade, follow the [Upgrade to Development Build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) instructions.  If you could test and report back that would be great so I can get v1.4.3 released.  Thanks!

fmartinezo, 2018-11-12:
@teckel12 Tested it and it workes good. Only one question. I have my taranis in spanish mode and lua select automatic spanish lenguage is it ok? or you can change it in lua in order to speak other lenguages i cant found this parameter. So if it is not possibe  it works great for me

fmartinezo, 2018-11-12:
Very very very good work. Thanks so much

teckel12, 2018-11-12:
@fmartinezo Lua Telemetry uses the Taranis settings for language. There's no language settings in Lua Telemetry at all. I have a Wiki document on how to setup your Taranis for different languages.

teckel12, 2018-11-14:
Added in v1.4.3 release.

closed 2018-11-14 by teckel12

fmartinezo, 2018-12-11:
New spanish voices less robotics and less metalic

[es.zip](https://github.com/iNavFlight/LuaTelemetry/files/2669843/es.zip)

teckel12, 2018-12-12:
@fmartinezo I'll update these later today. Thanks!

teckel12, 2018-12-13:
@fmartinezo The ZIP file seems to be corrupt.  At least I can't open it.

fmartinezo, 2018-12-13:
Change "Zip" by "rar"

fmartinezo, 2018-12-13:
@teckel12 I enclose new zip file
[es.zip](https://github.com/iNavFlight/LuaTelemetry/files/2677155/es.zip)
the previous file was compressed with winrar. This is compressed with winzip

teckel12, 2018-12-13:
@fmartinezo Got it!