teckel12/LuaTelemetry #209: Is Betaflight supported?

henfri, 2018-11-11:
Hello
this looks great. Do you know, whether the script works with Betaflight as well, or does Inav have specific modifications?

Regards,
Hendrik

teckel12, 2018-11-12:
@henfri Nope, Betaflight lacks the telemetry info required to work. Not that Betaflight couldn't be modified to be compatible. But GPS and navigation just isn't the focus of Betaflight so I don't think there's really anyone who wants it.

I could submit a pull request to Betaflight for the changes, but would anyone really use it? I just don't see the value, make your case.

teckel12, 2018-11-14:
@henfri I understand that.  But Lua Telemetry monitors flight information to know the flight state.  Without this flight state, much of the information would be wrong.  Also, how the flight status is communicated has been greatly modified in INAV to work with Lua Telemetry.

I'll tell you what, I'll submit a pull request for Betaflight to add enhanced FrSky telemetry support that will allow Betaflight to work with Lua Telemetry.  If there's resistance from Betaflight to merge it, you can make your case there.  I'll let you know the status...

teckel12, 2018-11-15:
@henfri Sorry...  I starting making the changes and then discovered that Betaflight is now using the `Tmp2` telemetry sensor to send yaw/pitch/roll information while INAV uses `Tmp2` for flight modes.  This makes it much more challenging for Betaflight to be made compatible with Lua Telemetry.

For this reason, Betaflight compatibility is being noted as "out of scope" for Lua Telemetry, as Lua Telemetry is designed for INAV and the two projects have diverged enough at this point to be fundamentally incompatible with regards to telemetry.

Sorry.

closed 2018-11-15 by teckel12

henfri, 2018-11-17:
Hi Tim,

I understand. 
I appreciate your efforts though! Thanks!

Two last questions:
-Wouldn't it be possible to make LuaTelemetry compatible (by distinguishing between BF/iNav?
-Can you point me at the relevant part of the LuaTelemetry code, where I can have a look at the usage of Tmp2?

Greetings,
Hendrik

teckel12, 2018-11-19:
@henfri Let me clarify... It's not that Lua Telemetry needs to change to work with BF.  It's that BF would need new features added that INAV already has to be made compatible with Lua Telemetry.

Basically, BF needs to port stuff from INAV, then it would work with Lua Telemetry. Nothing needs to change in Lua Telemetry.

Looking at the BF code, it seems that INAV and BF have diverged with regards to FrSky telemetry. So getting the BF authors to agree to incorporate the INAV changes (and therefore breaking BF stuff) would be difficult or not possible at all.

If you'd like to make the changes in BF required to work with Lua Telemetry that's fine. But that's what I already looked at and I don't think the BF folks would allow the changes, so it would be a waste of your time.

Let me reiterate. BF needs to change to work with Lua Telemetry, not the other way around.