teckel12/LuaTelemetry #214: German translations

teckel12, 2018-12-03:
@Peschi90 Due to adding support for Horus transmitters, new German translations would be helpful:

-- Max 10 characters
Fuel
Battery
Current
Altitude
Distance

teckel12, 2018-12-04:
I'll take a stab at it:

Fuel = Kapazitaet
Battery = Batterie
Current = Ampere / Strom?
Altitude = Hoehe
Distance = Entfernung

I've learned the best way to get a reply on the Internet is not to ask for help but instead post something that's incorrect 😁

Peschi90, 2018-12-04:
Sorry mate i saw it to late.
Your Translation is correct current means Strom in German Ampere is the value like Meter or miles. 

Great Job!

teckel12, 2018-12-04:
@Peschi90 No problem, I didn't need the answer quickly, just attempting to make it easier.

closed 2018-12-17 by teckel12