teckel12/LuaTelemetry #221: Small typo errors in README.md

MRC3742, 2018-12-27:
Thanks for this great resource to make iNav telemetry much more use-able.
While reviewing the latest release, I've noticed a few small errors.

There are two identical typos in the README.md files.
Under the Requirements heading you have "FkSky" listed instead of "FrSky"
And the same typo again under Notes heading.
These errors follow through all branches of your repo.

Thanks Again for all your time and effort on this project - Rich

teckel12, 2018-12-27:
@MRC3742 What file are you referring to? The root readme seems correct.

MRC3742, 2018-12-27:
The root README.md

Line 52:    _FkSky_ X-series, R9 series or D-series telemetry receiver: X4RSB, X8R, XSR, R-XSR, XSR-M, XSR-E, RX4R, RX6R, XM, XM+, R9, R9 Slim, R9 Slim+, R9 Mini, R9 MM, D8R-II plus, D8R-XP, D4R-II, etc.

Line 64:    INAV v2.0+ is required for _FkSky_ D-series telemetry and proper GPS accuracy (HDOP) display

teckel12, 2018-12-27:
@MRC3742 Sorted out, thanks!

closed 2018-12-27 by MRC3742