teckel12/LuaTelemetry #238: Horus: Benchmark drawing speed and update accordingly

teckel12, 2019-01-06:
Lots of drawing to the screen with the Horus view. Benchmark different drawing techniques to see which are faster and use those. For example, is it faster to draw two parallel lines, a rectangle two pixels wide, or a filled rectangle two pixels wide.

Time the drawing speed for each section to see where most of the drawing time is spent. Then, focus on the slowest sections for optimizations.

For attitude display, maybe calculate pitch once vs. horizon and then extrapolate each pitch line. For example, calc horizon and 50°, then just divide and subtract for all the pitch lines instead of using slow algabra for each line (this has got to greatly speed things up).

wx4cb, 2019-01-20:
not having a horus i can't test, but if possible maybe just update the things that you need to update slowly say once every second and the attitude a little more ?

teckel12, 2019-01-20:
@wx4cb I considered that, and still am. But most stuff is really fast (like drawing text and non-trig math). There's a lot more trigonometry and drawing of solid areas with the attitude display.

On a simar note, I ran some tests on the Q X7 with Crossfire and the screen updates were REALLY fast and responsive (even with the attitude display). While some of the optimizations I'm doing for the Horus can carry over to Taranis transmitters, it's already so fast that it's really not required.

teckel12, 2019-01-21:
Implemented changes to the Horus view which increase FPS by up to 40%.

closed 2019-03-10 by teckel12