teckel12/LuaTelemetry #239: CRSF: RSSI doesn't work well

teckel12, 2019-01-06:
Should match INAV RSSI in OSD, but doesn't.

teckel12, 2019-01-13:
Changed to use the RSSI values from the receiver (the better one if there's diversity) and then scale it to the 130dB sensitivity of Crossfire.

It works well, but I'd really like to get the calculations Crossfire uses to send RSSI/link quality back to the FC via channel 8/12.

ZinhoRacing, 2019-01-16:
Hi @teckel12 .
Did you got any answer from TBS about RSSI calculation? I'm very interested too.

teckel12, 2019-01-17:
@ZinhoRacing They mentioned the latest Crossfire firmware exposing the RSSI value. But I upgraded and don't see it. I've got a reply back trying to clarify.

The way I'm doing it now makes a lot of sense. It takes the best of the two RSSI values, subtracts it from 130dB, and uses that value for RSSI. It doesn't match what Crossfire is sending back to INAV over channel 8 or 12, but it at lease makes sense and seems fairly linear.

I'm good with going with the way it is, it can always improved in the future from TBS feedback.

teckel12, 2019-01-22:
After discussion with the TBS engineer, the RSSI value is coming from the RSNR telemetry sensor, using the calculation he provided.  More info: #244

wx4cb, 2019-01-22:
that's cool... I just loaded it onto my mini drak (F405 wing) and also updated the script on my taranis X9DSE.

I can tell you right now the pilot display with the artificial horizon is updating WAYYY faster. I'll find out in the morning when i go flying to see how it runs.

teckel12, 2019-01-23:
@wx4cb Note that It will slow down when Crossfire goes to 50Hz, but that's not the script's fault. The script shows the Crossfire transmission speed in the title at the far right, so you can monitor it now. 

There's more speed improvements I could still make, but it's at the cost of using more memory, which is limited on Taranis. Probably won't do more speed tweaking till VTx control is built in, which is scheduled for the next major release.

PR #243 (2019-02-26): V1.6.0
* Adds Crossfire support (requires INAV v2.1.0-RC3 or later)
* Horus: Enhanced display speed (up to 40% faster)
* Horus: Script variables initialize on flight/telemetry reset
* Compatible with Betaflight using FrSky X or R9 series receivers (with reduced functionality) and TBS Crossfire support with Betaflight 4.0+

Fixes #36
Fixes #235 
Fixes #236 
Fixes #237 
Fixes #239
Fixes #240 
Fixes #241
Fixes #242
Fixes #244
Fixes #245
Fixes #246 
Fixes #248 
Fixes #250
closed 2019-02-26 by teckel12