teckel12/LuaTelemetry #242: Horus: Detect telemetry reset in widget and run reset function

teckel12, 2019-01-06:
Is this possible?

KnuckleUpFPV, 2019-01-15:
https://github.com/yaapu/FrskyTelemetryScript/blob/master/README.md

Check this lua out. He says for telemetry to reset two conditions must be met. Timer 3 must be zeroed out and motor disarmed. I dont know if this will be any help to you.

KnuckleUpFPV, 2019-01-15:
I posted this to show it is possible. Didnt know if his code would help you. Your question was very vague. I just assumed you didnt know if you could reset telemetry on the horus. 

Side note. Development of your lua is working with a 2.0 version of inav downloaded from the configurator. They must have pushed your changes before releasing 2.1 rc1

teckel12, 2019-01-15:
@KnuckleUpFPV Telemetry can be reset via the transmitter, but there's not a hook to tell a Lua script when it happens.  The trick was to find a way to detect when telemetry was reset, and then do a script reset as well.  There's not really a great way of doing it, which is why it was labeled "is possible?".

I've worked it out and it's pushed to the development branch.

Crossfire support for Lua Telemetry is already merged into INAV 2.1.0.  What hasn't merged yet is the MSP over Crossfire.  Still trying to get that approved for 2.1.0 release.

PR #243 (2019-02-26): V1.6.0
* Adds Crossfire support (requires INAV v2.1.0-RC3 or later)
* Horus: Enhanced display speed (up to 40% faster)
* Horus: Script variables initialize on flight/telemetry reset
* Compatible with Betaflight using FrSky X or R9 series receivers (with reduced functionality) and TBS Crossfire support with Betaflight 4.0+

Fixes #36
Fixes #235 
Fixes #236 
Fixes #237 
Fixes #239
Fixes #240 
Fixes #241
Fixes #242
Fixes #244
Fixes #245
Fixes #246 
Fixes #248 
Fixes #250
closed 2019-02-26 by teckel12