teckel12/LuaTelemetry #251: iNav 2.0.1 LUA 1.5.1 FrSky QX7 + Frsky R9M S.bus+S.port

FPVxfun, 2019-02-06:
Description: while (long) pressing PAGE button for LUA telemetry access, it is shown the enclosed error. As shown in the second picture, telemetry data are correctly received by QX7 

Tx: FrSky QX7S EU (new) running OpenTx 2.2.3, with Tx module R9M, iNav LUA 1.5.1
Rx: FrSky R9M mini
Model: WingFC running iNav 2.0.1

[QX7.zip](https://github.com/iNavFlight/LuaTelemetry/files/2837375/QX7.zip)
[image]
[image]

teckel12, 2019-02-06:
@FPVxfun I ran OpenTX 2.2.3 and your Taranis settings that you included and installed Lua Telemetry 1.5.1 and it worked fine for your WingFC model.

[image]

My guess is that you didn't install all of the Lua Telemetry files or in the correct location.  There's MANY more files that just `iNav.lua`. Please see the `Download & Setup` instructions on the following page: https://github.com/iNavFlight/LuaTelemetry/releases  Specifically, make sure you're copying both folders and all the contents.

FPVxfun, 2019-02-07:
Dear Tim, thank you for your support.

Please consider that all is working ok on my second radio (Horus X12S) and on Q X7 Betaflight scripts are working too.

I could not understand what is going wrong with iNav and Q X7.

Enclosed you can find my SD card content, if necessary.

Could you please send me a SD card working content?

Thanks a lot

[SD.zip](https://github.com/iNavFlight/LuaTelemetry/files/2841610/SD.zip)

FPVxfun, 2019-02-07:
Dear Tim, I've also tried the "development build". It worked (for a while) with a screenshot similar to the one you posted, however when I tried to switch to "graphic view", all crashed once again

FPVxfun, 2019-02-08:
Dear Tim, many thanks for your support.
The problem seems to be few Q X7 RAM resources.
I've disconnected Betaflight LUA script in the 12th menu page and now iNav is working ok.

teckel12, 2019-02-08:
@FPVxfun If you only need the Betaflight script for VTx control, I have a stripped-down version that may work along with Lua Telemetry.

teckel12, 2019-02-08:
@FPVxfun I've reworked the [Tips & Common Problems](https://github.com/iNavFlight/LuaTelemetry/wiki/Tips-&-Common-Problems) Wiki page.  Could you give it a quick look to see if it would now better guide you to resolving your problem?  Thanks!

FPVxfun, 2019-02-09:
Awesome! please don't forget to suggest power cycle on-off-on in order to guarantee real RAM de-allocation

closed 2019-02-11 by teckel12