teckel12/LuaTelemetry #252: Pitch input causes little arrows to appear and value changes on x10s

expipiplus1, 2019-02-10:
### To reproduce

To reproduce:

- Navigate to the luatelemetry widget
- Observe the values displayed in the widget and the lack of little arrows on the screen
- Move the pitch input up or down
- Observe that some values have changed and there are little arrows next to many of them

I guess that this is some kind of variometer mode.

### Expected behavior

Pitch input doesn't change what is displayed

### Screenshots

[image]
[image]

### Radio and model settings

setup:
- x10s
- crossfire
- Latest development release as of 09-02-2019

expipiplus1, 2019-02-10:
Ah, I see what's causing this: `data.showMax` is toggled by the PREV and NEXT events.

Nothing unexplained any more, perhaps I missed this in the docs.

closed 2019-02-10 by expipiplus1

teckel12, 2019-02-10:
@expipiplus1 Ah yes, you're toggling between the current and the min/max values. The arrows are designed to indicate this. When you arm it goes to the current values. But once disarmed, it's sometimes nice to know your max speed, altitude, amps, or whatever.

teckel12, 2019-02-10:
@expipiplus1 I'd also like to get some feedback on Crossfire and the Horus (both new additions). Thanks!

expipiplus1, 2019-02-11:
@teckel12 It seems to be working for the most part, thanks for the great resource!

I've only had one flight with this setup and spent all of it watching the plane :)

I did manage to get into a state earlier where all I could see was the text "disabled" on the screen instead of the widget, not quite sure how that happened though.

teckel12, 2019-02-11:
@expipiplus1 That's what happens when the Horus has any kind of problem with a Widget. I've only seen it when first setting up Lua Telemetry as a widget and not making it full-screen.

A restart of the transmitter will resolve it (maybe changing models would also work). I should, however, add that to the tips wiki page as it could be quite common while setting it up.

expipiplus1, 2019-02-11:
Yeah, I figured as much :)

Please feel free to remind me to give some more feedback in a month if I've not done so already by then  once I've had a chance to use this some more!

expipiplus1, 2019-02-13:
I encountered the "disabled" message again on my second flight, I'll open an issue about this.