teckel12/LuaTelemetry #253: Widget becomes disabled on x10s

expipiplus1, 2019-02-13:
### To reproduce

- Navigate to the luatelemetry widget
- Observe that everything is working as expected
- Fly around for a bit (I haven't encountered this on the bench)
- Observe that the widget has stopped working and OpenTX is just showing `disabled`

### Expected behavior

The widget keeps working

### Radio and model settings

setup:
- x10s
- crossfire
- Latest development release as of 09-02-2019

expipiplus1, 2019-02-13:
I understand that this isn't much to go on. I'll try to keep an eye out in the future for this along with any other useful information.

If there's any kind of debug build/mode I should try please let me know.

teckel12, 2019-02-13:
@expipiplus1 OpenTX will disable a Lua script for any number of reasons. It could be an error, and then again it could be too many CPU cycles were being used for non-critical operations.

I believe the open an issue has instructions on dumping your OpenTX settings and Lua Telemetry config. If I could get those, I may be able to isolate or duplicate the issue.

teckel12, 2019-02-13:
@expipiplus1 I was on my phone for my previous message so I couldn't be more specific as to what I was looking for.  Here's the instructions for exporting radio and model settings that are needed to simulate (and hopefully reproduce) your problem:

1. Connect transmitter to computer using standard method (trim in, power, connect USB) see: https://youtu.be/GAETmZsKqwY?t=2m13s
2. From your transmitter's '\SCRIPTS\TELEMETRY\iNav\cfg' folder, copy the contents '*.dat' to your computer
3. Run OpenTX Companion
4. From 'Read/Write' menu, select 'Read Models and Settings From Radio'
5. Note the model number/name you're using when having the issue
6. From 'File' menu, select 'Save As...' and save radio models and settings to OTX file
7. ZIP the OTX file and the '*.dat' files and attach it to this issue
8. Make sure you specify the model number/name from your transmitter when having the issue

teckel12, 2019-02-17:
@expipiplus1 My guess is that your transmitter has decided the script is using too many CPU resources and has disabled it.  I haven't experienced this, but that is how OpenTX works.  I could limit the FPS or alternate between calculation and display or something like that.  But, I'm only guessing that this is the problem.  It seems to run very fast (around 10 FPS) but who knows when or why OpenTX decides to disable a script with such a generic message like "disabled".

skyherokokkie, 2019-02-20:
Does this maybe got to do withe the 4000k mod or no mod.

Sorry to brake in this conversation, but i am gonna install tomorrow using crfs and X10S

teckel12, 2019-02-21:
I have the mod installed on my X10S and running at 400k.  I haven't had the "disabled" come up while flying, only when initially setting it up (which is common with widgets).  But, it could depend on what other things are running on a particular user's transmitter.  Lua Telemetry doesn't seem like it's anywhere near overloading the transmitter, but I guess it could happen if there was a lot else going on.

expipiplus1, 2019-02-26:
I haven't installed the 400k mod yet, I will compare when I've installed
it, thanks for the idea.

teckel12, 2019-02-26:
@expipiplus1 When running Lua Telemetry as a widget, you could long-hold Enter, select (I believe) Stats? And it will show you a bunch of stats including stats on lua scripts (refresh speed, memory used, etc). Probably best to hit Enter to reset stats, go back to Lua Telemetry for a bit, then go back to the stats and take a picture or report what the info is.

I'll do this on my transmitter to get you exact instructions as I believe I'm missing a step. But that's the basic gist of it.

expipiplus1, 2019-03-08:
@teckel12 I'll give that a try today if I remember at the field.

expipiplus1, 2019-03-10:
I forgot to get the info, but I did notice that the widget became disabled very quickly after arming, something I've not been able to reproduce at home.

teckel12, 2019-03-10:
@expipiplus1 My only guess is that you have other things running which are using more CPU resources than typical.  I and others have never experienced this `diabled` issue.  I know it can happen if the widget overloads the CPU, but at least for me, it's not close to that limit.

If you were running FrSky Rx getting the transmitter settings and a telemetry log would allow me to simulate your flight in the OpenTX companion.  However, with Crossfire, different telemetry sensors are used which don't pass through to the companion, making simulation impossible.

Maybe looking at your transmitter settings would help.  Following are the instructions for exporting radio and model settings:

1. Connect transmitter to computer using standard method see: https://youtu.be/2IlSxJ9LNXk?t=132t=2m13s
2. From your transmitter's '\SCRIPTS\TELEMETRY\iNav\cfg' folder, copy the contents '*.dat' to your computer
3. Run OpenTX Companion
4. From 'Read/Write' menu, select 'Read Models and Settings From Radio'
5. Note the model number/name you're using when having the issue
6. From 'File' menu, select 'Save As...' and save radio models and settings to OTX file
7. ZIP the OTX file and the '*.dat' files and attach it to this issue
8. Make sure you specify the model number/name from your transmitter when having the issue

teckel12, 2019-03-15:
I was able to reproduce the `disabled` and I believe I know why.  I'll look into this today.

teckel12, 2019-03-18:
@expipiplus1 The latest development branch should resolve this issue.

expipiplus1, 2019-03-26:
Thanks! I'll try it out ASAP!

PR #261 (2019-03-29): v1.6.1
* Added new option for altitude graph (additional view on Taranis / map overlay on Horus)
* Config option to set max cell voltage for calculating cell count (primary use: LiHV batteries)
* Horus: Fix display bug that caused `disabled` error
* Horus: Further optimized view for faster rendering
* Crossfire: Better fuel percentage estimation when battery isn't full
* Update language files for new config menu options

Fixes #253 
Resolves #254 
Resolves #257 
Fixes #258 
Fixes #259
closed 2019-03-29 by teckel12