teckel12/LuaTelemetry #254: [Feature Request] Altitude graph

wx4cb, 2019-02-17:
This would be a nice feature to have.

https://www.rcgroups.com/forums/showthread.php?3158918-LUA-Program-that-Plots-Altitude-on-Taranis-X9D

teckel12, 2019-02-17:
@wx4cb It's been considered. The main issue is the memory used to log that data. The Taranis has such a small amount of memory allocated to lua scripts that it's difficult fit everything in.

Since you're using Taranis, have you tried to run the script you linked to in addition to Lua Telemetry? It can run on another screen that you would use the page button to cycle through.  If you get an error trying that, it's probably due to running out of memory, which would likely also happen if the functionality was added to Lua Telemetry.

Let me know how it works running both Lua Telemetry and the other script at the same time. No reason to recreate the wheel if another program already does thr job.

I'll still look into it. Seeing how much memory it uses would be the primary issue. If it was added, it would likely be added an additional view on the Taranis. This would also reduce the memory used a bit.  On the Horus, it could be included in the map area.

wx4cb, 2019-02-17:
ahh... i didnt know you could run them at the same time so to speak. i'll
give that a try when i go out next

teckel12, 2019-03-10:
@wx4cb For Taranis, the best solution may be to use an already existing script.  But for Horus, that may not be an option.

In the latest development branch I've added an altitude graph that's displayed in the map area. It shows the altitude over the latest 5 minutes in 5 second increments. Not much styling at this point, but it does seem to work well. More development and sample images to follow...

wx4cb, 2019-03-11:
cool.... i'll take a look at it when i fly next, just installed it.

teckel12, 2019-03-11:
[image]
Sample of altitude graph

teckel12, 2019-03-15:
Due to Taranis memory concerns and already existing scripts that perform this task, this enhancement won't be added on Taranis transmitters.  On Horus transmitters, Lua Telemetry v1.6.1 will add the option of displaying an altitude graph that looks somewhat like what is shown above (grey replaced with green and now shows top of scale indication instead of f/s which is shown in another location).  Will update sample Horus image with addition of altitude graph.

Jetrell, 2019-03-19:
> 
> ahh... i didnt know you could run them at the same time so to speak. i'll give that a try when i go out next

Did you manage to get the Altitude graph script from RCG to run on the X9D?
I didn't have any luck.
I loaded the .LUA file into Scripts/Telemetery/. Then tried to execute it.
It comes up as - "Script syntax error - Attempt to index a nil value."

teckel12, 2019-03-19:
@Jetrell That may be something you should contact the script author about. I didn't try running the script as I don't have a X9D transmitter. I added support only on the Horus as it has the memory (and screen space) to support it.

Jetrell, 2019-03-19:
I'll PM the guy later. 
@teckel12  Could a possibility be....That Lua Telemetry is using most of the X9d"s memory, and there may not be enough left to run the Altitude Graph FW.
The people that have used it with success, may only be using lua scripts with lower memory usage. 
Just conjecture :-)

teckel12, 2019-03-19:
@Jetrell That could very well be the problem.  Easy to test, just remove Lua Telemetry from a page so only the other script is running (you'll need to reboot).  You won't lose settings or anything from Lua Telemetry if you do this.

If it is memory, it's also the reason that adding it to Taranis may not be possible.  I will try adding it to Taranis and do a before and after assessment of the memory used.  No promises if it can be added.

Jetrell, 2019-03-20:
I removed Lua Telemetry. And the AP script still did not start.
I've lost track of the different Script error's that it has display over the last day, as I have tried different things to get it to work.
In the end it's still not functional. 
I'm over it for now :-)

teckel12, 2019-03-20:
@Jetrell Well, I'll add a altitude graph view as a test on Taranis that you can test out.  I really don't know what kind of memory it will use but that will be part of the test.  When I get it working I'll post here along with instructions.

wx4cb, 2019-03-20:
is it possible to put a text field on one of the screens which shows memory
usage ?

teckel12, 2019-03-20:
@wx4cb Not possible as that value isn't exposed to lua scripts. You need to access the Taranis menu to get the free memory. But it's as easy as hitting a few menu options.

wx4cb, 2019-03-20:
it was just a thought for something you could track in any logging you do
for testing

teckel12, 2019-03-22:
@Jetrell I've created an altitude graph view for Taranis transmitters.  To upgrade to the branch with the altitude graph view, use the following instructions:

1. Download [LuaTelemetry-taranis_alt_graph.zip](https://github.com/iNavFlight/LuaTelemetry/archive/taranis_alt_graph.zip)
1. Open or extract the `LuaTelemetry-taranis_alt_graph.zip` file
1. Open the `LuaTelemetry-taranis_alt_graph` folder and navigate to the `dist` folder
1. Copy the contents of the `dist` folder (`SCRIPTS` and `WIDGETS` folders) to the transmitter's SD card's root

Once running this branch on your transmitter (will boot as v1.6.1) hit <Enter> to scroll through the views till you get to the altitude graph (or you can select the altitude view from the config menu).  I haven't yet compared the memory used with and without this view.  I should be able to do that this evening and report back what the results are and if I believe merging this into development for the next release will be possible or not.  The way it was implemented, it may not use much memory (or that's my hope).  We'll see...

Let me know if you like it and it works, etc.  Thanks!

Jetrell, 2019-03-23:
@teckel12 It works great Tim :+1: 
Nicely laid out and not crowded;  Just right!
 
I haven't seen any issues. 

Also.. Well done with the battery percentage fuel gauge. It works great too. Its simple and easy to use. By `Reset Flight` after a partially discharged battery is connected.
As you said....Its not totally accurate. But I'm more than happy with the ball park estimate, than the way it was before. 

Thanks again for your efforts.

teckel12, 2019-03-23:
@Jetrell I still need to review the memory usage, but feel confidant that making altitude graph an additional view will keep memory usage low.

Jetrell, 2019-03-23:
@teckel12 
Been out flying Line of Sight again.
Not trying to rush you Tim...
Just a question or two before you finalize it and right it to the Wiki.

1) When it comes to the graphs `TIME AXIS `. 
    Pressing the + or - buttons causes an UP arrow to appear next to the graph.  Does this Add and 
    Subtrack minutes like the AP.lua?  I found it hard to tell when flying.
    AP.lua says it has a default of 5 minutes. I found that a bit long for the size of this graph. I'm more 
    likely to go for 2 minutes.

2) Also the lua `Config MENU` shows  `Altitude Graph --`. I can't seem to change anything under this 
    field. 
    _@teckel12 wrote: (or you can select the altitude view from the config menu)_
     Will it be configurable later?

teckel12, 2019-03-23:
@Jetrell The arrow has nothing to do with the altitude graph. It's noting that the values are min/max values instead of current values. Basically, noting that the values to the two sides of the altitude graph are min/max and not current. It's currently fixed to a 5 minute graph. I've considered allowing that to be configurable, but not till I'm sure the feature is even going to stay.

The altitude graph option in the menu is only for the Horus. It allows you to turn it on and off. I'm fairly sure I have this listed in the wiki. For Taranis, the altitude graph is selected with the view option. If I do go forward with merging this into development, that option may expand to serve double-duty to not only activate the altitude graph, but also set the time period.

teckel12, 2019-03-24:
@wx4cb @Jetrell Did memory testing on Taranis and making the altitude graph a view doesn't use much memory. I may incorporate a way to totally disable the altitude graph to save a bit more memory but I think it's fine as-is.

I'll also look into different ways of showing different time lengths for the altitude graph. The only real problem is the complexity of the minute markers. It's not that it's difficult to figure out, it's that Lua doesn't like doing loops with lots of calculations. And the added complexity required could make Lua unhappy.  In any case, I'll see what I can do.

I've merged this into the development branch so you can update using the standard update method detailed in the wiki.

Jetrell, 2019-03-24:
Thanks @teckel12 
I understand your issue.
How about setting the Time scale to 2.5 minutes? With each marker being 30 seconds?

The main problem I see with 5minutes. Is that the original AP.lua took up the whole screen width. Which gave better resolution over 5 minutes.
While with the LT version. The graph is only 1/3 the size.

teckel12, 2019-03-26:
@Jetrell The latest development build now includes support for changing the altitude graph length from 1-6 minutes (as well as off by default). The dividers adjust to the minute marks.

Let me know if there's any issues.

Jetrell, 2019-03-26:
Thanks for the updates Tim.
It works great. It now covers everything from short duration quads or F5D pylon. To long range planes that will cover a lot of terrain. 

Great work :+1:

closed 2019-03-29 by teckel12

evandillen, 2021-11-09:
Hello,

i have a taranis x9d+ 2019 and otx 2.3.14
i want to try this alt graph.lua script but i can't find it anymore (error 404)
where can i find it?
thx etienne van dillen

PR #261 (2019-03-29): v1.6.1
* Added new option for altitude graph (additional view on Taranis / map overlay on Horus)
* Config option to set max cell voltage for calculating cell count (primary use: LiHV batteries)
* Horus: Fix display bug that caused `disabled` error
* Horus: Further optimized view for faster rendering
* Crossfire: Better fuel percentage estimation when battery isn't full
* Update language files for new config menu options

Fixes #253 
Resolves #254 
Resolves #257 
Fixes #258 
Fixes #259