teckel12/LuaTelemetry #255: [Feature Request] Logging options

wx4cb, 2019-02-17:
I'm having issues with the taranis not logging all the TLM data using it's SD logs (specifically alt and speed) even though they're all checked in the telemetry sensor options.

Is it possible that you could do logging from the lua script so i can turn all that crap off as it's not working correctly anyways. I'm mainly just interested in date/time/gps posit/alt/speed/heading and maybe rssi/lq.

teckel12, 2019-02-17:
@wx4cb All of that logs for me just fine on my Taranis Q X7. How are you reviewing the logs? Even with a spreadsheet app they're not listed?

In any case, while it would be possible to create a lua script to log telemetry, it would be a duplication of an already working feature.

Have you tried creating an issue in the OpenTX repo?

wx4cb, 2019-02-17:
yup even in excel it's not there - not that i can see there isn't one labeled "alt" or anything like that

Date | Time | 1RSS(dB) | 2RSS(dB) | RQly(%) | RSNR(dB) | RFMD | TPWR(mW) | TRSS(dB) | TQly(%) | TSNR(dB) | GPS |   | Sats | Ptch(rad) | Roll(rad) | Yaw(rad) | FM | RxBt(V) | Curr(A) | Capa(mAh) | GSpd(mph) | Hdg(@) | Rud | Ele | Thr | Ail | S1 | S2 | S3 | LS | RS | SA | SB | SC | SD | SE | SF | SG | SH | LSW | TxBat(V)

Those are the only ones i have. as far as "issue, no i haven't i only just noticed it tbh.

teckel12, 2019-02-17:
My TBS Crossfire logs:

Date,Time,FM,1RSS(dB),2RSS(dB),RQly(%),RSNR(dB),RFMD,TPWR(mW),TRSS(dB),TQly(%),TSNR(dB),RxBt(V),Curr(A),Capa(mAh),GPS,GSpd(mph),**Alt(ft)**,Sats,Ptch(rad),Roll(rad),Yaw(rad),Rud,Ele,Thr,Ail,S1,6P,S2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)

It does look like you have one sensor (between `GPS` and `Sats`) that doesn't have a name.  Also, can you show the altitude sensor on a telemetry page?  If so, it really sounds like you just don't have that sensor checked to save to logs.  For each sensor you can turn logging on or off.

In any case, if it's not saving and the sensor does exist and you do have that sensor set to log, I would create an OpenTX issue as this sounds like either an OpenTX support issue or bug.

wx4cb, 2019-02-18:
@teckel12 no that blank one is the latlong that i had split out to bring into google earth.

as far as on the telemetry pages, yes it works fine in both the LUA script and also just using a simple "hums" page. i also get the callouts and when i flip my readout switch it reads it out (along with voltage, current, speed, and mah)

I have opened one in the opentx repo: https://github.com/opentx/opentx/issues/6334 

[image]
[image]

[image]

teckel12, 2019-02-18:
@wx4cb I would include the image where you show the logs box checked.  Then, include a sample log file.

closed 2019-02-18 by teckel12