teckel12/LuaTelemetry #256: Change altitude alert mode from max to current

wx4cb, 2019-02-20:
Is it possible to change the altitude alert to only sound if you're higher than the max alt you have set?

Reason why is once you reach max alt (400 for me :P) even if you are then < than that altitude it still sounds. for example I was testing climb performance of my S1100 while testing different props and  I was at 600 feet in class G airspace, obviously the alert is set for 400 but when i got down to 200 feet it was still reading the 600 foot altitude.

Next time i try it, i'll try and record a video to give more information, but basically it kept reading out the max altitude and not the altitude i was at or even stopped once i got below the alert altitude set.

I would be nice to be able to have an option in the config to set it to alert at all altitudes (ie verbal altitude readout) or only say your altitude if you are above the max set.

Hope that makes sense.

teckel12, 2019-02-21:
@wx4cb I'm not following you.  It should only show the current altitude and the alert beep and/or feedback should only sound when you're over the set max altitude.  For example, you can set the max altitude to 600 feet and it should only alert you when you're over 600 feet.  Once you go below 600 feet the alert stops.

teckel12, 2019-02-21:
@wx4cb I just verified that it only gives the alerts (beeps or altitude announcements) when over the max altitude set.  Once you go below the max altitude it stops reporting/beeping as it should.  It also only shows your current altitude (unless you've changed the view to show the max).

Maybe you have the `Variometer` option set to `Voice`?  If so, this is designed to always voice report your altitude, which has nothing to do with the max altitude set.  It's designed for fixed wing pilots.  With this setting, there's the accompanying `Vario Steps` which allows you to set the increments when the altitude change is alerted.  For example, if set to 50 feet, it will announce every time you climb or decline by 50 feet.

wx4cb, 2019-02-21:
let me double check those and get back to you. won't be for a couple days though, out of town

closed 2019-04-14 by teckel12

wx4cb, 2019-05-01:
ok just got around to getting back to this. variometer is off, but once i hit 400 feet and go back down, it still says the max height. also, just as a test, i went up to 700 feet and when i went back down it still was saying that the rest of the flight.

Am I missing something here in the configuration?

[image]
[image]
[image]
[image]

teckel12, 2019-05-02:
@wx4cb Better than pics would be the .cfg file for the model in question. See the cfg folder inside the iNav folder.

teckel12, 2019-05-02:
@wx4cb I tried it with the latest development build and the best I could duplicate your config settings without the cfg file and it seems to work just fine. It beeps, announces the altitude and the altitude flashes when 400' or higher. As soon as I drop belop 400' the flashing, beeping, and voice accouncements all stop. Tried several times, always worked correctly. Went up to 700' with the same result.

Make sure you're using the latest development build and if you're still having problems I'll need the cfg .dat file and your telemetry logs to simulate your flight.

teckel12, 2019-05-02:
Do you happen to not have a barometer? Does it show the correct altitude on the display?

wx4cb, 2019-05-02:
No i have barometer (f722 wing) and just tried it, unable to get latest dev release but its on latest public. Whats your email address so i can send you all this stuff.