teckel12/LuaTelemetry #257: Battery percentage accuracy

Jetrell, 2019-03-04:
This is as much a question as it is an issue.

I often take notice of the main battery percentage level at a glance when flying line of sight. I even prefer it over mAH.
If the battery is fully charged when the telemetry link is established. It will show 100% and count down from there. But if the battery was connected with only 90% capacity, it will count down from that level with poor accuracy, when compared to the battery voltage(under low load).
Would it be possible to note the Voltage of the battery with low current drain after a telemetry connection. And then make a percentage estimate to start the percentage count down from?

I'm not even totally sure if this is an iNav or Lua Telemetry problem?

Thanks

teckel12, 2019-03-05:
@Jetrell For FrSky telemetry, Lua Telemetry doesn't calculate the fuel percent. It uses the value supplied by INAV.  If the battery voltage is below a certain configurable value, INAV will base the fuel percentage by voltage instead of mAh. My guess is that's what you're experiencing.

In any case, Lua Telemetry doesn't do the fuel calc for FrSky telemetry, it uses the INAV fuel value. Unless you're flying TBS Crossfire, in which case fuel is always the mAh used divided by the battery mAh entered in the config menu.

Jetrell, 2019-03-05:
> 
>
> 
@teckel12 Unless you're flying TBS Crossfire, in which case fuel is always the mAh used divided by the battery mAh entered in the config menu.

Hi Tim, I have not used Lua Telemetry until v1.6 because I use crossfire. Its very nice. You've done a great job with it :+1: 

Because of your statement above with respect to Crossfire. Would it be possible for you to take a low current battery voltage sample and base the fuel meters start point on that? According to Lipo/Lion/LiFE voltage vs capacity discharge tables.

teckel12, 2019-03-05:
@Jetrell It sounds easy, but what you're asking is highly prone to errors. All Lua Telemetry gets from Crossfire is the total mAh used and the current battery voltage. That's simply not enough information to make an adjustment. What's the battery's max voltage for example? When would this calculation be made? At arming? What if you disarm and arm mid-battery? And even if more information was known, basing it on the voltage would result in a situation where the fuel percentage would be only based more on voltage instead of mAh used.

Basically, it just wouldn't work well, and be prone to being inaccurate with full batteries. The alternative is to just use full batteries, which is what I would guess 99% of people do.

If you feel there's a method of doing it, I'll be more than happy to review a pull request. But I fear it would be helpful for 1% and negatively effect the 99% that use full batteries.

If there's a solution here at all, it's to have Crossfire ass thr FUEL sensor like FrSky telemetry. That way, INAV could calculate the fuel percentage.

Jetrell, 2019-03-05:
I do fly with a full battery most of the time too. Its just on those occasions when doing testing. That its a hassle to take off and land a half dozen times while tuning. And have to top the battery up 100ma or so, on the occasions a FC power cycle is needed.

The only idea I have Tim, is this.

1) The first time the Lua script is enabled(after a transmitter power up). And a telemetry link is established. 
It would read the battery voltage and the current drain. And only do this ONCE over a sort period of time, with a latch function. 
(The transmitter would have to be power cycled or the script disabled and re-enable to provide another read).

2) If the current drain is under a specific threshold (e.g. 600mA) for an accurate assumption of fuel capacity; based on voltage. Then do a read. And set the capacity accordingly.
If these parameters are not meet. Then go as it does, assuming the battery is at full capacity.

The user would need to remember to cycle the transmitter power or the script. So to get the new capacity value. But if you don't....then it can't be anymore inaccurate than it already is under a FC reboot, when the battery is discharge to some degree.

Does that sound workable?

Thanks

teckel12, 2019-03-06:
@Jetrell What you're proposing is not programmatically possible without causing a nuisance to most users 99% of the time. It doesn't seem that you have a clear understanding of programming or what information is available to base calculations on. Also, user experience is paramount to Lua Telemetry's core goal. 

But like I said, if you submit a PR I'd consider it if it didn't result in additional rules for pilots to follow 99% of the time when unneeded.

Instead, for the rare times when you're tuning, you should just monitor the battery voltage. A mAh fuel gauge isn't that accurate anyway when you're not flying like you normally do. In other words, calibration and normal flying is different, therefore the mAh used will be wrong anyway, so even if there was some way of calculating the fuel percent if the battery wasn't full, the assumed mA draw would be different anyway, so it would still be wrong.

Jetrell, 2019-03-06:
_@teckel12 What you're proposing is not programmatically possible without causing a nuisance to most users 99% of the time. It doesn't seem that you have a clear understanding of programming or what information is available to base calculations on. Also, user experience is paramount to Lua Telemetry's core goal._

I do agree.  
I don't have an understanding of C++ or how this particular code interacts with the iNav and Crossfire codes.
The suggestion was just an idea that I once made work with Picaxe .BAS.
It had 95% accuracy when calculating from a hall sensor. And its Voltage verses Capacity plot to indicate a Capacity level. Reduced the accuracy a farther 10%, depending on the batteries temperature and rest state.  I know that isn't perfect. But its far better than the fuel meter in Lua Telemetry or iNav at present. 
At present I can connect a battery that is 80% discharged, which can be seen from the battery voltage level at rest. While the fuel gauge will still insist that the capacity is at 100% and count down from there.

I know you have limited to work with. And as I said earlier, I don't full understand this system.
But that's a bug to me. Why include something that is misleading. 
If DJI sold a Mavik with a fuel gauge that had an accuracy level like that. The average Joe would be running them flat all the time ;-)

I won't say anymore.
Thank you for your time.

teckel12, 2019-03-06:
@Jetrell The solution for an 80% full battery being connected must happen at the flight controller level, not the transmitter. INAV already does this for SmartPort / F.Port / D-series telemetry. When you connect an 80% full battery with FrSky telemetry, it shows 80% fuel, not 100%. The fuel percentage is then tracked by cell voltage, not mAh used.

The reason you're not seeing this is because Crossfire doesn't send fuel percentage, so INAV can't send it.

As I mentioned, this is really a Crossfire feature request, as a not full battery calculation needs to come from the flight controller, as it needs to calculate this at the time of flight controller startup, which is unknown at the transmitter.

teckel12, 2019-03-16:
@Jetrell I believe I may have figured out a way of doing it on the controller.  There are some conditions, but it does seem to work.  You must plug the battery into the model, then within a fairly short period of time (before 25 mAh is drawn) do a reset flight on the controller.  For my sample quad, it takes quite a while before 25 mAh is used when the motors are not spinning, so this should be a reasonable time.  I may be able to increase this or maybe even not make it dependent on the total current drawn.  But, that will take more testing to verify it still works that way.

I'll be including this in the development branch in a little bit.

Jetrell, 2019-03-17:
Thanks for giving it more thought Tim :+1: 
Sounds like a perfectly fine way of resetting it to me.

teckel12, 2019-03-17:
@Jetrell It's not perfect for sure, it takes a very linear battery voltage to mAh and there's other gross estimations. But, it should be better than showing 100% fuel on a 50% low battery.

Please don't treat it as being overly accurate, as it's an estimation using a linear voltage to mAh scale. So 20% fuel could really be 0%. So also mind your battery voltage accordingly.

PR #261 (2019-03-29): v1.6.1
* Added new option for altitude graph (additional view on Taranis / map overlay on Horus)
* Config option to set max cell voltage for calculating cell count (primary use: LiHV batteries)
* Horus: Fix display bug that caused `disabled` error
* Horus: Further optimized view for faster rendering
* Crossfire: Better fuel percentage estimation when battery isn't full
* Update language files for new config menu options

Fixes #253 
Resolves #254 
Resolves #257 
Fixes #258 
Fixes #259
closed 2019-03-29 by teckel12