teckel12/LuaTelemetry #258: Incorrect cell number detection with high voltage cells

expipiplus1, 2019-03-10:
### Describe the bug

Cell number detection is hard coded to 4.3V,  some high voltage lipo cells can be charged to 4.35V. This manifests itself as an incorrect better level bar in the widget, and spurious low battery warnings.

It's worth noting that inav has configuration parameters for the cell count detection threshold and allows for an explicit number of cells.

### To reproduce

Use the widget with a model with a fully charged lipo hv battery. Observe that the battery bar is near empty and low battery messages are played.

### Expected behavior

The number of cells is detected correctly

expipiplus1, 2019-03-10:
For reference, the battery detection code is here: https://github.com/iNavFlight/LuaTelemetry/blob/master/src/iNav.lua#L131-L132

teckel12, 2019-03-10:
@expipiplus1 I've had this happen as well. I'll look into it today. Seems like an easy fix/enhancement.

teckel12, 2019-03-10:
@expipiplus1 As a side note, this cell count calculation is a backup for the cell voltage on the A4 telemetry sensor. I didn't realize that was yet another sensor missing from Crossfire telemetry (there are many things missing). So this is really only a problem with Crossfire telemetry, not any FrSky telemetry which incluses the A4 cell voltage sensor.

expipiplus1, 2019-03-26:
Thanks for looking at this!

teckel12, 2019-03-26:
@expipiplus1 I just realized that I didn't note in this issue that the enhancement is included in the latest development branch. There's now a config menu option that allows you to set the max cell voltage for cell count calculation. It defaults to 4.3V but you can set this to 4.4V for

expipiplus1, 2019-03-26:
Cool, thanks again.

PR #261 (2019-03-29): v1.6.1
* Added new option for altitude graph (additional view on Taranis / map overlay on Horus)
* Config option to set max cell voltage for calculating cell count (primary use: LiHV batteries)
* Horus: Fix display bug that caused `disabled` error
* Horus: Further optimized view for faster rendering
* Crossfire: Better fuel percentage estimation when battery isn't full
* Update language files for new config menu options

Fixes #253 
Resolves #254 
Resolves #257 
Fixes #258 
Fixes #259
closed 2019-03-29 by teckel12