teckel12/LuaTelemetry #262: add arming disabled reason to screen

wx4cb, 2019-03-21:
is it possible to add underneath where it says "NOT OK" the reason why it's not ok? i'm testing some mavlink stuff and for some reason it won't arm even though it should be fine. 

I don't really want to reconnect up a vtx and goggles :D

Just an idea

teckel12, 2019-03-21:
@wx4cb The only things available to a lua script are the telemetry values sent from INAV to your transmitter.  Please see the telemetry sensors on your transmitter for what is available, which does not include a reason why it's not okay to arm.

Basically, there's no way for Lua Telemetry to know the reason as INAV telemetry isn't sending that information.

closed 2019-03-21 by teckel12

wx4cb, 2019-03-22:
that's what i figured.... oh well

teckel12, 2019-03-22:
@wx4cb It *may* be possible once and if MSP can be implemented. But, still probably not due to limited memory available on Taranis transmitters.