teckel12/LuaTelemetry #266: Voltage Not showing using Crossfire CRSF

turbosi92, 2019-03-30:
I have been trying and trying to get voltage on all my rigs but none work.  All 4 wings are on latest inav as of 2/25/19 on F405 Wing and F411 boards.  Using full sized Crossfire module and CRSF protocol.  All the data shows up fine on the screen EXCEPT voltage.  In the settings of the script there is an RX Voltage but its disabled (and i can't enable it) and that is how i get voltage on CRSF.  Yes, i've done the discover sensors many times to no avail.  Again everything else shows up fine other than voltage.  

I'm open to any suggestions here.  I have another telemetry page setup and it shows the voltage just fine.  I've read the docs but i don't see anything that would cause what i'm seeing.  I am at my wits end as I've spent way to much time already so i'm hoping someone knows what i'm missing!  Thanks!

teckel12, 2019-03-30:
@turbosi92 Try deleting the sensor and then discovering it again. The name should be exactly `RxBt`. If it's anything else other than the default name, Lua Telemetry won't know the battery voltage. My only guess is that you've renamed the battery voltage as there shouldn't be a problem otherwise and no one has reported this problem previously.

turbosi92, 2019-03-31:
Thanks Tim!  Thats exactly what I did but forget to edit my post.  I had 2 RxBts in sensors and I deleted the one that showed 0v and all is well.  I do get blinking GPS coordinates even though I have a lock but seems everything is working so far on all 4 of my wings with F405, F411 and an old Betaflight F3 board on iNav 2.1

teckel12, 2019-03-31:
@turbosi92 I can't stress this enough, if something is blinking or not showing correctly, it should be addressed or it could cause problems. A blinking or red value means something is wrong. Many could also lead to constant vibrations or audio feedback. My guess, is that the GPS is blinking because you have something non-standard with the number of satellites telemetry sensor.

turbosi92, 2019-03-31:
Its blinking now, has 17 satelites and says "Not OK" but whats blinking are the coordinates themself

Sats are showing in sensors.  Coordinates are correct.  From a speaking standpoint...there are no voice alerts going on.    

There is also a sensor that says GPS which is displaying the coordinates correctly.

If there is anything else to check let me know.  This is on a taranis with full size Crossfire module.

teckel12, 2019-03-31:
@turbosi92 Could you maybe not be using INAV 2.1.0?  The latest INAV is required for Lua Telemetry when using Crossfire. Or, maybe your FM telemetry sensor was renamed/duplicated as well? 

Basically, I'll need more info. When you create an issue it asks for certain things, like telemetry logs from your transmitter, your transmitter settings, etc. This allows me to diagnose what's happening instead of just guessing.

teckel12, 2019-04-01:
@turbosi92 I looked into this some more and I couldn't duplicate when you're experiencing.  I'm going to need to see your telemetry logs and your radio and model settings.

The telemetry logs are created on your transmitter by turning on `Logs` for each sensor then creating a special function to create the SD logs when the arm switch is activated.

Below are the instructions for exporting radio and model settings that are needed to simulate (and hopefully reproduce) your problem:

1. Connect transmitter to computer using standard method (trim in, power, connect USB) see: https://youtu.be/GAETmZsKqwY?t=2m13s
2. From your transmitter's '\SCRIPTS\TELEMETRY\iNav\cfg' folder, copy the contents '*.dat' to your computer
3. Run OpenTX Companion
4. From 'Read/Write' menu, select 'Read Models and Settings From Radio'
5. Note the model number/name you're using when having the issue
6. From 'File' menu, select 'Save As...' and save radio models and settings to OTX file
7. ZIP the OTX file and the '*.dat' files and attach it to this issue
8. Make sure you specify the model number/name from your transmitter when having the issue

closed 2019-04-30 by stale[bot]