teckel12/LuaTelemetry #268: nil value error

laurentopia, 2019-04-01:
### Describe the bug
inav moved from display 2 to display 1(which was bar telemetry) = nill value error message
### To reproduce
1. set display1 as bar and give it gps telemetry
2. set display2 as inav
3. add vtx lua
4. set display 2 to vtx and display 1 to inav

now go to the home page and long press PAGE, instead of inav telemetry we get "script syntax error ?:0 attempt to call a nil value"
when i restart the taranis now i'm getting not enough memory on the second screen of the inav telemetry (yes the first screen shows up now)
then I exit that display back to home and long press page again and i'm getting that script syntax error again

some times the qx7 freezes up and we can't exit that screen

### Radio and model settings
qx7 with 2.2.3 with flex build
model: XL8 inav
[qx7.zip](https://github.com/iNavFlight/LuaTelemetry/files/3027336/qx7.zip)
(add the inav telemetry back in display 1 to see those errors)

teckel12, 2019-04-01:
@laurentopia Please see the Wiki: [Tips & Common Problems](https://github.com/iNavFlight/LuaTelemetry/wiki/Tips-&-Common-Problems)

laurentopia, 2019-04-01:
I followed all that and inav is the only display. There is no indication of memory usage on open TX so finding problem source ain't trivial.

Here is something new: now the inav display works but if I go through screens with the enter wheel button, after a few cycles it "call nil value".  could it be memory leak in the scripts?

teckel12, 2019-04-02:
@laurentopia You can see the free memory available to lua scripts. From the main page, hold enter (I believe) then select statistics or something like that, then page to see free memory. Doing it by memory as I'm not at my desk but those are basically the steps. 

You probably have other things taking up memory so you don't have much left. Each Lua Telemetry view uses a different amount of memory but each is cleared out as it moves to the next. What could happen, I guess, is fragmented memory. But I've never run into this.

Maybe you have many models saved or a lot of custom functions or mixes, etc. All that uses memory. On my QX7, I can cycle through all the views seemingly forever and it never runs out of memory.

I'm also using the latest stable release of OpenTX. I believe you said you're running a pre-release which could also be the issue as they are often not stable and not designed to be flown with.

teckel12, 2019-04-03:
@laurentopia I confirmed that to see the free memory from the main OpenTX page do the following:

1) Hold `Enter`
2) Select `Statistics`
3) Press `Page` key
4) Value you're looking for is `Free Mem`

No matter how many times I cycle through all the views, this value never goes below about 14,000 for me.  And I tried flipping through maybe 20 cycles and it stayed at 14,000.

laurentopia, 2019-04-05:
Thanks @teckel12 I didn't know that features.
So I put your stripped down VTX as display 2, Page Page = nil error
Go to statistics and free mem is at 200b
We're getting closer.

teckel12, 2019-04-05:
@laurentopia Nope, no longer possible to even use my stripped down VTx script along with Lua Telemetry. The **only** current solution is to use the OSD for VTx control, wiki changed to no longer suggest my VTx script, now suggests to use OSD.

Until such time that VTx control can be incorporated into Lua Telemetry doing both on the transmitter is impossible.

laurentopia, 2019-04-08:
How about using BF as VTX lua control?

teckel12, 2019-04-08:
@laurentopia The BF lua script uses far more memory, so that will never work.  The reason I created the VTx script was to reduce the amount of memory by getting rid of all the PID stuff.  While Lua Telemetry _could_ work with my VTx script (if you don't have anything else on your transmitter).  There's no possible way the BF script would work with Lua Telemetry as it uses as much memory as Lua Telemetry alone.

Basically, the OpenTX folks have set a limit that just isn't enough to do both, which is the entire reason I have a task to add VTx control to Lua Telemetry.  Because there's no other option at this point other than use the OSD to control VTx, which isn't so bad.  I just don't change my VTx channel at all, that makes it even easier.

closed 2019-04-21 by stale[bot]