teckel12/LuaTelemetry #269: INAV LUA SCRIPT ADI IS REVERSED ON TARANIS 9D

Stringbrakr, 2019-04-02:
### Describe the bug

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

### Screenshots

### Radio and model settings

### Additional context

Stringbrakr, 2019-04-02:
Thank you very much. This can be a real airplane saver for me. The page with the ADI helps me the most. I am having a problem to solve. the roll indicator is reversed on my taranis 9d+. the command in the inav cli has the set osd_artificial_horizon_reverse_roll = ON it works on the osd but it does not change the lua ADI. Any ideas?

teckel12, 2019-04-02:
@Stringbrakr First, the transmitter doesn't see the INAV settings, so it's not correct to assume that a setting in INAV will be sent to the transmitter, it absolutely does not.

Secondly, I have no idea what you mean or what your request is.

teckel12, 2019-04-02:
@Stringbrakr Okay, I think I may be understanding you.  You're looking at it as an attitude indicator and not an artificial horizon.  A Russian attitude indicator keeps the horizon horizontal and moves a plane indicator to show attitude.  That's not what the artificial horizon shows.  It shows the horizon angle.  There's no ability to rotate the plane indicator to show attitude in relation to a fixed horizontal horizon.

Stringbrakr, 2019-04-03:
Thank you.  I knew that I had to be doing something wrong.  I just didn't know I was misunderstanding its  basic function.

closed 2019-04-03 by Stringbrakr