teckel12/LuaTelemetry #271: Configuration can not be saved

mastersurferde, 2019-04-10:
### Describe the bug
After a power cycle all configurations in the script are resetted. I use a X9D with crossfire. Normally I switch the Battery consumption to absolut, the altitude warning off and some other settings.
How can I save the settings permanently? - or is this a bug?

teckel12, 2019-04-11:
@mastersurferde Config settings are saved when you exit the config menu. So I guess if you change settings then power off, the settings won't save. Also, it saves settings for each model name. So each model can have different settings.

I would check in the `cfg` folder inside the `iNav` folder for the model.cfg files. I guess it's possible that it's not working based on the model name if it can't save the model name as a file name.

The reason I ask for information from your transmitter when you open an issue is so I can try to duplicate your issue. Without this, I can only guess. So, if you could go to the page where you opened this issue and selected "bug" and then supply the information asked, I could look into this instead of guessing randomly.

teckel12, 2019-04-11:
https://github.com/iNavFlight/LuaTelemetry/wiki/Issue-Template

closed 2019-04-11 by teckel12

P-I-Engineer, 2019-05-27:
i'm having same issue

everything works great, other than it wont save my settings. I keep trying to turn off the altitude alarm.
[Desktop.zip](https://github.com/iNavFlight/LuaTelemetry/files/3222020/Desktop.zip)

frsky taranis (non plus)
opentx 2.2.2

great script, thank you very much.

mastersurferde, 2019-05-27:
After some hours of testing I found the solution.
Delete all Inav Files and Folders on the SD-Card and write them completely new.
The issue happens if you overwrite a old version on the SD-Card with the new version.

Best regards,
Stefan

P-I-Engineer, 2019-05-27:
that was it!
thank you

teckel12, 2019-05-27:
@P-I-Engineer Maybe a file could become locked? Strange, but you can always just delete the cfg file for the model having a problem and a new cfg file will be created.

P-I-Engineer, 2019-05-27:
ok, thank you for the response.

mastersurferde, 2019-05-27:
Hi,
no - files are not loocked. I have testet serveral SD-Cards with my actual Image. 
The cards have been formated with SD-Formater.
The cfg-Files were always new created if I have deleted them before. (the first time I have opened a modell).
I have tested with a X9D and a X9D+ with OpenTX 2.2.3. Same behaviour on both TX.

After deleteing and reinstalling the Inav-Directories it works perfect.

Best regards,
Stefan

teckel12, 2019-05-27:
@mastersurferde Strange, never had the problem on several controllers, I believe this is the only issue about this too.

What's the model name and the file name as a result. The best I can guess is that your model name is causing a file name problem (as it uses the model name for the cfg file name).

If you still have the problem, monitor your cfg files, as that's where the settings are stored. Remember that settings only save when you exit the config menu. So entering and rebooting (for example) won't save.

P-I-Engineer, 2019-05-28:
i even changed my model name, from "nano goblin" to "nano_goblin". it created a new cfg file and still suffered from the same issue.

once i fully deleted and re-uploaded, it all started working.