teckel12/LuaTelemetry #273: RSSI Display issue (Updated)

frouxel, 2019-04-11:
Hello,
Following your instructions in my first issue #272 "RSSI display issue invalid", I give your more detailed informations in this new one.
Regards,
Florian

### Describe the bug
Using INAV LUA Telemetry V1.61 on Taranis X9D plus under OpenTX 2.2.3 EU configure with FR language, RSSI refresh is OK in the classic view but is stuck to 99dB in Pilot or Radar views (No refresh).
If I use +/- button in de radar/pilot view, RSSI show what seems to be the minus value of RSSI (The value doesn't refresh, except for a lower value (example : when 1 mask TX antenna))

### To reproduce
From de default screen after powering on the Taranis, long press "Page" key to display INAV Telemetry classic view
-> RSSI bar graph on the bottom line is correctly displayed with good refresh rate.

Go to "Pilot view" or "radar view" pressing "ENT" key one or two times
-> RSSI on the bottom right of the screen is stucked to 99dB 

In "Pilot view" or "radar view" pressing "+/-" key display RSSI value which seems to be the minus recorded value.
=> Maybe 99dB is the maximum recorded value instead of the current read

### Expected behavior
Without pressing "+/-" key, RSSI should display current value instead of 99dB (like in INAV Telemetry classic view)

### Screenshots
[image]
[image]
[image]

### Radio and model settings
Frsky R-XSR receiver with EU-LBT FPORT firmware 190311 upgraded before D16 bind with Telemetry ON 
Matek F722STD with INAV 2.1.0 with matek M8Q-5883
	-> FPort SBUS configured on TX2 with (set serialrx_provider = FPORT, set serialrx_inverted = ON, set serialrx_halfduplex = ON, set gps_provider = UBLOX7, set gps_sbas_mode = AUTO, set gps_ublox_use_galileo = ON, set smartport_fuel_unit = PERCENT, set battery_capacity = 1300)
	-> Telemetry ON
Taranis X9d plus EU-LBT from 18/01/2016
	-> OpenTX 2.2.3 freshly upgraded from 2.1.6 using a brand new Sandisk Ultra 16GB micro sdcard (Bootloader 2.2.3)
	-> Did not manage to upgrade XJT internal module firmware to 170317 (it seems to be "normal" for non hardware upgraded radio before 15/11/2016)
	-> SDcard content used from https://opentx-doc.fr/wiki/opentx-2-2-3-2-2v0018 with FR sound only, and FR translated stuff (maybe the origine of the issue?)
	-> Telemetry reset and re-discovered, sensor 420 renamed to Dist, units ajusted for "km/h" and "m"

	-> My only model is "Drone RXL" and \SCRIPTS\TELEMETRY\iNav\cfg\DRONE RXL.dat' contains this value "135342301200110211100203020350030101500043"
	-> See attached "DroneRXL.zip" file
[DroneRXL.zip](https://github.com/iNavFlight/LuaTelemetry/files/3067764/DroneRXL.zip)

### Additional context
Add any other context about the problem here.
	-> I don't know if it can help but I've tried to add "VTX LUA script" and it generate script error with memory issue (I have since disabled this script in display since memory issue is frequent with Tarany, and you suggested to use OSD to set VTX)

d3nyys, 2019-04-11:
I have the same problem. RSSI is displayed ok in the Classic view but stuck on 99 in the Pilot view.

teckel12, 2019-04-11:
Interesting, I wonder if this was introduced in the v1.6.x cycle. I'll try duplicating the issue this evening. Thanks.

wx4cb, 2019-04-11:
One thing i did notice and it might be a conflict between settings i have,
is that it keeps repeating some of the voice alertS over and over again.
Ive been out of town all week, so when i get time to fly tomorrow ill check
it out with a fresh version on the sd card

teckel12, 2019-04-12:
@wx4cb I would open a different issue about a different problem. Also, you'd need to be more specific, like which voice alert was it repeating? Were any values on the display flashing/red/yellow at the time? Maybe your settings cfg file, etc.

Keep in mind that this is obviously not happening for me or everyone or it would have been addressed before release. So when your having a problem, it's specific to your use-case. So, giving your use-case is important.

But still, it's not a good idea to hijack an issue with another question or issue. Please open another issue. The issue system is how issues are tracked so they're organized and issues don't fall though the cracks.

teckel12, 2019-04-12:
@frouxel We're you not armed when it was showing 99dB for RSSI? What did it show once armed and flying?

frouxel, 2019-04-12:
No, I wasn't armed when I noticed the issue, I will test today and report to you.

teckel12, 2019-04-12:
@frouxel Found the issue. I must have left some test code in the build for Crossfire that was setting the RSSI to 100. It fix is included in the latest development build.

frouxel, 2019-04-12:
Thanks a lot Tim, I flew this afternoon and even armed, RSSI was also stuck to 99dB.
According to your advice, I've just tested developpement build and it just works :) 
Is it possible that altitude bargraph in classic view was also stuck to zero ? I didn't notice this before this afternoon, and with developpement build, it seems OK too.
Thanks you very much for this great job.
Regards,
Florian

teckel12, 2019-04-12:
@frouxel Glad it's working for you. I don't see any issue with the altitude bargraph, either with 1.6.1 or with the development branch, and no changes were made in that either.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12