teckel12/LuaTelemetry #275: Lua disable when selected on boot up of TX

jm-wijnholds, 2019-04-21:
i've found a problem with the latest version of the iNav LuaTelemetry:

When using it on an Horus X10(s) you can not boot you TX with this lua script enabled otherwise it will tel you that it is disabled. A little research gave me this error: "refresh():?:0: attempt to compare number with nil"

What does work is making second screen: one with the lua widget, and another one without the lua. Shutdown the tx with the screen without the lua(so the tx will bootup in this screen when you turn it on), en than go to the screen with the lua widget.

In multiple FB groups i found people having the same issue.

I'm using version: v1.6.1 from lua and OpenTX version 2.2.3 on an Frsky Horus X10s. (for the record, the screen with the lua is full screen, without the top bar and trims. Also running only one screen does not help)
[image]

teckel12, 2019-04-21:
@jm-wijnholds I assume these FB groups are not specific to this Lua Telemetry script but to lua in general? Sounds like an OpenTX bug?

teckel12, 2019-04-23:
Fix merged into development.

jm-wijnholds, 2019-04-26:
Hi Teckel, No the FB groups were specific in this Lua Telemetry Script :). It was also mentioned that it was working on version 1.5.1 and was 'broken' in from version 1.6.

but i see that you've added a fix for it. Will try version 1.6.2 when its out. thanks!

teckel12, 2019-04-26:
@jm-wijnholds Yes, someone else reported the same thing and I believe v1.6.2 fixes it.

teckel12, 2019-04-27:
@jm-wijnholds Not sure why there would be groups in a vacuum on FB when they could be communicating directly with the author on Github or Telegram.

I know some are comfortable with FB, but it's not designed for issue tracking. Please encourage those with issues to use the Github issue system. Or, for real-time chat, Telegram works really well. Both are listed in my Wikis.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12