teckel12/LuaTelemetry #276: CRSF INAV LUA: 'NOT OK' MESSAGE

RanChen123, 2019-04-23:
runnig inav 2.1.0 on matek f405 wing with crsf attached to taranis  x9d.

i get a 'NOT OK' message instead of receiving flight mode name. its blink with the gps Coordinates at the same time while  15-16 sats.

beside of that issue i think everything is working perfectly.

thank in advance

ran.

teckel12, 2019-04-23:
@RanChen123 The number of satellites is not considered. The flight mode controls when it's not okay or ready to fly. INAV tells LuaTelemetry via the FM sensor when INAV believes it's ready to fly.

Look at your telemetry sensors on your transmitter. What's the value of the FM sensor? That is set my INAV and tells LuaTelemetry when it's okay to fly, and what your flight mode is.

RanChen123, 2019-04-23:
thank you. i will check it asap. 
also today i notice that the compas is not calibrate (matek f405 wing and bn220 gps unit without compass) when i put the aircraft nose point to the north (0) it show me at the telemetry lua  35 +/-
do you know what can be problem?

thank in advance,

Ran

teckel12, 2019-04-23:
@RanChen123 Again, look at the telemetry sensor. LuaTelemetry takes telemetry values and displays them in a more useful way. Garbage in, garbage out applies. 

As a guess, with fixed wing you don't use a compass, it uses forward flight to determine orientation. My guess is that it would work while flying. But I don't fly FW so it's just a guess. Works fine for me with multi-rotor with a hardware compass.

wx4cb, 2019-04-23:
if it's not on a quad, just disable the compass, it's not needed

gsixx8, 2019-04-25:
What is the FM Sensor?  Mine is showing FM WAIT on my Taranis QX7.  The “NOT OK” message and GPS coordinates flash in the LUA script.  Even after getting the minimum sats, arming and flying, the not ok message and gps coordinates flash.

teckel12, 2019-04-25:
@gsixx8 The `FM` sensor is the flight mode. `WAIT` is what INAV outputs when not all flight checks are good. When LuaTelemetry sees the `WAIT` in the `FM` sensor, it shows you the `NOT OK` message, as it means your model is not ready to fly (according to INAV). LuaTelemetry is just the messenger. 

Contrary to popular belief, the number of satellites has nothing to do with the GPS accuracy (DOP is the accuracy). INAV allows you to set a minimum number of satellites, but that's secondary to the DOP.

There's many other reasons why INAV would still be showing `WAIT` even if your DOP is good and you have the minimum satellites. For example, craft isn't level, sensor error, etc.

Do you have extra_arming_safety turned on? If so, you won't be able to launch till all safety checks are good (it will stay at `WAIT` till everything is good). If you have it turned off, you're flying when something isn't right, which LuaTelemetry is trying to tell you with the `NOT OK` message.

Your INAV `dump diff` would be helpful. Also, if flying FPV with OSD turn on messages so it will tell you any warnings you have (which should be the reason for the `WAIT`). Also, show the HDOP on OSD as that's the important GPS accuracy information, not the number of satellites.

Let me know what your OSD shows for messages. Remember, LuaTelemetry is just showing `NOT OK` because INAV believes it's not okay to fly yet.

gsixx8, 2019-04-25:
@teckel12 my OSD shows Unsafe To Fly until I get enough sats then there are no further messages as far as the Navigation goes.  Here is my diff dump sir:

Entering CLI Mode, type 'exit' to return, or 'help'

# diff dump

# version
# INAV/MAMBAF405 2.1.0 Feb 17 2019 / 18:43:38 (7bdd596)
# GCC-7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 2 2048 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led
led 0 7,15::CO:10
led 1 0,0::CO:10
led 2 15,0::CO:10

# color

# mode_color

# aux
aux 0 0 0 1850 2100
aux 1 1 1 900 1100
aux 2 2 1 1400 1600
aux 3 4 4 1875 2100
aux 4 9 4 1875 2100
aux 5 8 3 1875 2100
aux 6 36 1 900 2100
aux 7 11 2 1900 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 1 10 V
osd_layout 0 1 1 1 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 6 1 11 V
osd_layout 0 7 13 12 H
osd_layout 0 8 17 0 V
osd_layout 0 9 1 2 H
osd_layout 0 10 22 1 V
osd_layout 0 11 1 3 H
osd_layout 0 12 1 4 H
osd_layout 0 13 25 11 V
osd_layout 0 14 12 0 V
osd_layout 0 15 25 10 V
osd_layout 0 20 10 12 V
osd_layout 0 21 10 11 V
osd_layout 0 22 11 0 V
osd_layout 0 23 24 12 V
osd_layout 0 28 23 11 H
osd_layout 0 29 1 12 V
osd_layout 0 30 1 2 V
osd_layout 0 32 1 0 V

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6000
set acczero_x = 42
set acczero_y = 10
set acczero_z = 24
set accgain_x = 4086
set accgain_y = 4110
set accgain_z = 4045
set align_mag = CW90FLIP
set mag_hardware = QMC5883
set mag_declination = -520
set magzero_x = -118
set magzero_y = -18
set magzero_z = -114
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set min_throttle = 1140
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set model_preview_type = 3
set gps_min_sats = 5
set nav_mc_hover_thr = 1275
set osd_video_system = NTSC
set osd_units = IMPERIAL
set name = iXL7
set tz_automatic_dst = USA
set vtx_band = 1
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5865

# profile
profile 1

set mc_p_pitch = 70
set mc_i_pitch = 65
set mc_d_pitch = 40
set mc_p_roll = 70
set mc_i_roll = 65
set mc_d_roll = 40
set mc_p_yaw = 75
set mc_i_yaw = 65
set mc_d_yaw = 5
set dterm_lpf_hz = 100
set yaw_lpf_hz = 40
set dterm_notch_hz = 150
set dterm_notch_cutoff = 100
set tpa_rate = 20
set tpa_breakpoint = 1250
set rc_yaw_expo = 70
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 75
set manual_rc_yaw_expo = 70

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 340
set battery_capacity = 2300
set battery_capacity_warning = 690
set battery_capacity_critical = 230

teckel12, 2019-04-25:
@gsixx8 Try setting `set nav_extra_arming_safety = on`. Probably a good idea anyway (I always fly with this on) and it may isolate the problem if it then works for you.

teckel12, 2019-04-25:
@gsixx8 I think I know the problem, you're not running the release version of INAV 2.1.0.  2.1.0 was released February 25th and your version is from the 17th. You're running a RC before the Crossfire updates were in place or finalized.

teckel12, 2019-04-25:
BTW, for INAV release 2.1.0, this is what must happen if the flight mode sensor `FM` has a value of `WAIT`.

1. Not armed
2. GPS enabled
3. Extra arming safety is set to `on`
4. No GPS fix (nothing to do with the # of sats)
5. No GPS home fix

If all of the above are satisfied, INAV outputs `WAIT`.

gsixx8, 2019-04-25:
Is there a current stable release version of INAV for the MambaF405 FC or will I have to stick with 2.1.0?

teckel12, 2019-04-26:
@gsixx8 I'm a little confused, the latest stable release of INAV is 2.1.0 for all targets. The 2.1.0 you're running is a release candidate. I know the diff says 2.1.0, but that's what RC versions say.

teckel12, 2019-04-26:
@gsixx8 Also make sure you only have one FM sensor. Or, you could just delete all sensors and have it search for sensors again to verify you don't have multiple sensors with the same name. Someone else had this and therefore it wasn't getting good readings as sensors are accessed by name.

gsixx8, 2019-04-26:
I understand what you are saying about the current release.  I don’t see the Mamba FC listed as a choice on the firmware flasher dropdown menu for the “choose a board” option so I’m unsure which way to go as far as a firmware update to possibly correct the issues I’m having.

As far as the sensors, I just did a “delete all sensors” and “sensor search” on the tx yesterday.  I’m showing one FM sensor.

gsixx8, 2019-04-26:
Looking at the list of targets, I see the FuryF3 target, but the target in Betaflight for my FC is FuryF4OSD.

RanChen123, 2019-04-26:
i checked again and the fm sensor sohws OK but the 'NOT OK' message.

teckel12, 2019-04-26:
@gsixx8 The firmware you're using is MAMBAF405, see if you can download it from the Github download page. Maybe it's not in the configurator.

gsixx8, 2019-04-26:
I ordered an AIRBOT F4 V5 today. I like my experience on iNav and figured getting an FC that is recommended is the best way to go. Into the spares bin with the Mamba!!!!

teckel12, 2019-04-27:
@gsixx8 I'm going to have a bit of free time tomorrow. Maybe I'll compile a release 2.1.0 for you for your target.

gsixx8, 2019-04-27:
@teckel12 that would be awesome sir!

gsixx8, 2019-04-28:
@teckel12 any chance you made progress on compiling the MAMBAF405 target?

teckel12, 2019-05-02:
@gsixx8 Not sure where you got that target, it's not in INAV at all as far as I can tell.  It's no wonder it doesn't work, it's not an actual target.

teckel12, 2019-05-02:
@gsixx8 Seems MAMBAF405 wasn't in INAV v2.1.0 at all, which is why it's not in the configurator.  It's no wonder it's not working with Lua Telemetry as it was someone's forked build.

I built the MAMBAF405 target using the development branch of INAV v2.2.0: 
[inav_2.2.0_MAMBAF405.hex.zip](https://github.com/iNavFlight/LuaTelemetry/files/3136507/inav_2.2.0_MAMBAF405.hex.zip)

RanChen123, 2019-05-02:
@teckel12 i checked again and the FM sensor as you told me its 'OK' status but the 'NOT OK' message still shows in the lua pages blink with the gps coordinate

thank in advance,

Ran

teckel12, 2019-05-02:
@RanChen123 Will need to see a screenshot as I'm not sure what you mean by “still shows in the lua pages blink with the gps coordinate“

teckel12, 2019-05-02:
@RanChen123 I'd really like telemetry logs from your transmitter as I'm guessing without getting actual data. Keep in mind it works great for me.

RanChen123, 2019-05-02:
ill post a short video of the issue asap. 

how do i dwonload the telemtry log? 

i really want this to work 🤗😇

gsixx8, 2019-05-02:
@teckel12 Thank You for doing this.  I will flash my Mamba FC this evening and keep you posted on the results.  Anything special about 2.2.0 that i need to be aware of?

teckel12, 2019-05-02:
@gsixx8 A list of changes has not yet been documented from 2.1.0 to 2.2.0.  But, here's something that can be reviewed:

https://github.com/iNavFlight/inav/compare/master...iNavFlight:development

I would suggest dumping diff, clean flash, then load the diff noting any errors that occur.  This will typically tell you if there's any settings that have been refactored which could need to be investigated.

teckel12, 2019-05-02:
@RanChen123 Telemetry logs...  For each telemetry sensor make sure the `Logs` option is checked (I believe its already on by default).  Then create a special function that's triggered buy the arm switch in the arm position.  Set the function to `SD Logs` and the interval to `0.1s`.  That's it!

This will create a log file on your SD card's `LOG` folder each time you arm your model.  The log file name will be your model name followed by the date.  I would suggest doing this for every model as all it takes is adding a single a single special function.  The files are small, but if you're worried about SD card space, you can set the interval to `0.2s`. I just move the contents of my `LOG` folder every so often to my PC, not because they're large, but because I use the telemetry log files a lot.

It's great information that I would suggest everyone logging if using Lua Telemetry or not.  There's all kinds of forensic information that could be helpful.  For example, GPS coordinates.  Even if you lose your model you'll be able to find it or piece together what happened (IE: whoops, I hit the disarm switch, no wonder it crashed).

gsixx8, 2019-05-02:
@teckel12 i am noticing when i look at the inav configurator under the Mixer tab, the mambaf405 build i am using (old one, not the new one you made) has a motor mixer AND servo mixer.  When i was setting up one of my Matek boards, there was no servo mixer.  Just a motor mixer after selecting the 7" Endurance Multirotor preset.  Is a servo mixer necessary on a quadcopter without servos?

gsixx8, 2019-05-02:
@teckel12  How would i see any errors that occurred when adding back in the diff file?

RanChen123, 2019-05-02:
@teckel12 thank you sir! im still waiting to reach the transmitter to take a video of the blinking NOT OK message.

thank you for your help.

teckel12, 2019-05-02:
@gsixx8 You shouldn't need a servo mixer.  You open the CLI console then post the diff file you had before the update.  You then look at the output which shows that each setting was successful (or failed).  You can also do a search to find anything that fails.  It's manual, but an easy process.

gsixx8, 2019-05-02:
@teckel12 not sure what you mean when you say “look at the output”.  I saved the diff to a file using the save to file button after entering the ‘diff’ command in the CLI.  Then I flashed the FC to the version of iNav that you complied.  Then I copy/pasted the contents of the diff starting at #mixer.  I hit enter and all the info scrolled up the screen.  I then hit save.  After reboot, things look like they did as far as config and settings.  I haven’t fine tooth combed it yet, but thing look ok.  I may get the FC installed and wired back up to all the components tonight.

teckel12, 2019-05-03:
@gsixx8 Between when you pasted the diff and you hit save. All that stuff that scrolls up can be selected and copied, then pasted to a file (or you could review it on-screen after the paste is complete.

If there's a breaking change, the old diff command will throw errors which you can then investigate.

teckel12, 2019-05-03:
I'd like to get to the bottom of this `NOT OK` issue if there is one before releasing 1.6.2.  As 1.6.2 fixes a few other issues I'd like to release it sooner rather than later.

Any progress on this problem either being resolved by upgrading INAV and/or LuaTelemetry? Or any insight into why it's happening in your situation? It's only happening on Crossfire, so maybe the problem is with the bug in 2.89+ of Crossfire which is locking telemetry values (keeping it locked in a not okay state).

If we can't diagnose and resolve this in a day or so, I'll probably release 1.6.2 and come back to this error afterwards.

wx4cb, 2019-05-03:
@teckel12 i've been running the latest 1.6.1 since you released it without an issue, obviously when i've been flying i can't look at the screen to see if it freezes like the logging does, but next time i do i'll let you know. that won't be till sunday.

gsixx8, 2019-05-03:
@teckel12 I took my bird out today with your MambaF405 build flashed.  I never got more the  6 SAT’s during a 10 minute wait.  The NOT OK message kept flashing.  On my bird either with the Matek F722 on it, I was able to get enough SAT’s and arm.  Upon throttle up, the bird immediately flipped out.  All the while, the NOT OK message never stopped flashing.  That bird is Crossfire as well.

teckel12, 2019-05-03:
@gsixx8 There's more to being okay to fly than just the number of satellites.  Actually, as previously mentioned, the number of satellites really has nothing to do with a GPS fix.  This is urban legend, and totally untrue.  Waiting till you get 6 satellites is NOT a good indicator.  Your fix could be REALLY bad and you could have 10 satellites.  The minimum satellite count that you can add in INAV is a SECONDARY check.  You shouldn't rely at all on the number of satellites as having a good GPS lock or not.

Also, there's many other things that INAV checks to be sure it's okay to arm.  For example, if the gyro orientation is wrong.  This is why extra arming safety should be activated.  It avoids simple to fix problems, and makes sure you don't launch just because you can see 6 satellites (which again, means nothing) or the compass isn't being read or there's a misalignment or missing calibration.

So it's very possible that the NOT OK was being displayed correctly.  Because you didn't have a good GPS fix or that INAV was detecting something that was wrong.  Maybe I'll add a "raw" flight mode display which could make it easier to see what the FM sensor is showing, which should correlate with what Lua Telemetry is displaying.

gsixx8, 2019-05-04:
@teckel12 that’s a good idea.  Being able to see what the issue is would help a ton!  I’m going back our to try again. Don’t think I mentioned it but the two times I got it to arm and fly there was a new series of tones from the bird that I hadn’t heard before.  After these tones I armed and flew.  I had to wait quite a while both times.

gsixx8, 2019-05-04:
Got it up in the air after waiting for about 10 minutes.  Once again I got the series of tones or beeps and then I could arm.  The NOT OK message never went away, but it still flew.  I activated ALT, POS, and HEADING HOLD.  Got some toilet bowling so I went back into Angle.

gsixx8, 2019-05-04:
I’m guessing the tones means it’s ok to fly?

teckel12, 2019-05-04:
@gsixx8 Tones(beeps) would typically indicate some type of alert, if it's coming from Lua Telemetry at least. Lua Telemetry only does a beep every few seconds, indicating a warning. The only exception is if you have alt hold enabled, and then there's some tones indicating when you hit the center point where it's hovering. This can be disabled to avoid confusion as well. But basically, beeping is bad, not good.

teckel12, 2019-05-04:
@gsixx8 It's really odd that your only getting the NOT OK status. Maybe this simply has to do with the bug introduced in Crossfire 2.89.  I'm running Crossfire 2.88 which has no problems with showing the correct flight mode. If the flight mode stays at NOT OK, I would never arm nor fly. I would also never fly without extra arming safety turned on.

Maybe downgrade to the latest bug free version of Crossfire (v2.88) and enable extra arming safety. It sounds like maybe INAV and Lua Telemetry are trying to save you.

tfreddy23, 2019-05-04:
I am having this same exact problem but my fm sensor changes from angle to horizon to acro and it always works in my osd with no problems but in the lua I am also getting the not ok and the gps coordinates are flashing and flight modes won’t change but they do in the osd and the fm sensor just not on the lua !!

gsixx8, 2019-05-05:
Went out for a proper flight test at the park.  Got GPS fix much faster.  FM went from WAIT to OK in 5 minutes.  Got the beeping tones that I feel indicate things are good to fly, and flew!  Position hold and return to home need work, but things flew well.  The NOT OK message is still there in the LUA even after FM indicates OK.

gsixx8, 2019-05-05:
I put the FM sensor on my Taranis page on my TX so I could continually see its status.

wx4cb, 2019-05-05:
@teckel12 @gsixx8 I downgraded to 2.43 on the crossfire and went out, didnt notice whether the NOT OK was still on there, was flying fpv so kinda difficult, but i had all the normal operations. my FM sensor was 0 all the way through the flight. I have 3 flights logged if you want them

gsixx8, 2019-05-06:
@teckel12 after researching a bit, I believe those "special" tones I hear as the FM Sensor goes from WAIT to OK are the GPS FIX AQUIRED tones.

teckel12, 2019-05-06:
@wx4cb OpenTX doesn't log the FM value.  My guess is that it's designed only for numerical values and it doesn't know what to do with the alpha characters so it outputs zero.  This is most unfortunate because it makes diagnosing this problem difficult.

So for one, there's a bug in OpenTX that doesn't log character-based telemetry.  I've opened an issue for OpenTX: https://github.com/opentx/opentx/issues/6413

teckel12, 2019-05-06:
@gsixx8 Lua Telemetry doesn't play tones when going from different flight modes.  It can play voice audio tracks if you have it turned on which would say something like "Ready to fly" or "GPS good".  It does beep for warnings, like FS, RTH, battery low, RSSI low etc.  The beeps are all the same, a beep every 3 seconds.  It can also vibrate at the same 3 second frequency for warnings.  The config allows you to control if you want beeping/vibrating/voice feedback and for what level (all, warnings only, or off).

In any case, there's no "tones" that play in Lua Telemetry.  Not sure what you're hearing or what is setup on your transmitter, but it's not Lua Telemetry.  I also don't get any tones when flight modes change on my Horus running Crossfire 2.88.

teckel12, 2019-05-06:
@tfreddy23 Are you running the latest release of INAV (2.1.0?) and are you on a Taranis or Horus transmitter?

gsixx8, 2019-05-06:
@teckel12 the tones are coming from the beeper on the bird, not the tx.  I see in the Configurator options for the beeper GPS Fix Acquired as one of the switches you can turn on or off.  Mine is on.  That’s whats leading me to believe that the tones I’m hearing (one short, one long, two shorts all very quickly) are telling me that GPS Fix has been acquired and the FM Sensor backs that up in my mind by saying OK.

teckel12, 2019-05-07:
@gsixx8 I've got to conclude that this problem has to so with the firmware version of Crossfire. I flew today with both Horus X10S and Taranis Q7X and both go from NOT OK to READY. When I arm it reports that it's armed and goes to ANGLE, HORIZON, ACRO or whatever correctly.

gsixx8, 2019-05-07:
@teckel12 same behavior I’ve experienced except mine goes from WAIT to OK.  The flight modes such as ANGLE, HORIZON, ACRO, and HOLD show up under the FM Sensor after that when they are activated.

teckel12, 2019-05-07:
@gsixx8 I'm speaking of what's shown in Lua Telemetry.  It says NOT OK, READY, then ANGLE or whatever flight modes.  You're speaking of the FM sensor values.  I assume that while the FM sensor was changing, it wasn't changing in Lua Telemetry?  With Crossfire 2.88 Lua Telemetry correctly matches the FM sensor values.

What I'm asking is if anyone is using 2.88 and can confirm it works correctly.  Or, if running a newer version (as they have known telemetry bugs anyway) would downgrade to 2.88 and see if that resolves the problem.  I don't really want to upgrade to a known buggy version so I'll stay with 2.88 till TBS believes they've fixed telemetry.

tfreddy23, 2019-05-07:
Yes I am running inav 2.1.0 and I am using a taranis x9d plus.   I am also running crossfire 2.93.   I will try to downgrade the crossfire to 2.88. Like you suggest and I will let you know what happens.   I would like to thank you for all the time and effort you put in this to try and help us all out !!

teckel12, 2019-05-11:
@tfreddy23 FYI: Crossfire 2.95 beta was released which is supposed to fix some telemetry issues.

gsixx8, 2019-05-13:
@teckel12 I have been flying the iNav build you did for the Mamba F405 FC and I’m happy with it.  I have a small LED ring that I wanted to use.  Right now it is connected to the LED pad.  All LED’s are white.  I have set it up in the LED tab to be a different color, but they remain white.  Any ideas?

[36 more replies]

closed 2019-06-03 by teckel12