teckel12/LuaTelemetry #277: speed sensor,View Mode and GPS HDOP View can not be set

87040109qq, 2019-05-02:
Speed sensor,View Mode and GPS HDOP View can not be set ! Even after I got " GPS OK" voice prompt !
here is my set:
hours10(2.2.3)+D4R II (Frsky Telemetry)+INAV2.0.1 +Lua Telemetry 1.6.2 
How can I change those parameters ?

87040109qq, 2019-05-02:
What kind of conditions would make those three parameters addressable?

teckel12, 2019-05-02:
@87040109qq You can't set options which don't apply to your situation.  For example, `View Mode` can't be set because you're using Horus which combines all views into one view.  On Taranis, you must switch between (for example) pilot and map view because you can't see both at the same time.  On Horus, you can see all views all the time, so that option is disabled.

Same is true with the rest of the options.  `Speed Sensor` is only an option if you have a pitot air speed sensor.  You don't have that so the option is disabled.

`GPS HDOP View` is also disabled because you're using Horus which shows all options so there's no reason for the option to exist.

Please see the Wiki for more information on the configuration menu options: 

https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings

Also, please notice the note on the Wiki page which states the following:

> Options without selections have either been disabled by another option setting or are not available on your transmitter or telemetry protocol

teckel12, 2019-05-02:
> @87040109qq What kind of conditions would make those three parameters addressable?

If you had a pitot tube speed sensor and a Taranis transmitter those options would be available.

closed 2019-05-02 by teckel12

87040109qq, 2019-05-03:
Thank you for your detailed reply!