teckel12/LuaTelemetry #278: Altitude Alerts (update)

wx4cb, 2019-05-02:
as described in previous issue: Illusion - https://github.com/iNavFlight/LuaTelemetry/issues/256

had a flight today and the issue was still observed with crossfire. alt limit set to 400' climbed higher max altitude reported by voice alert was 525 and was constantly repeating that until i landed. even when < 400 feet.

Another thing i noticed was that when i disarmed, the script never called out "disarmed" until i pulled the battery which then i got the "telemetry lost" alert (as normal) but then it immediately said "aircraft armed" (not sure the exact phraseology)

non of the voice alerts mentioned are in a special function. the only ones i have are completely different and mainly for flight mode changes. The only thing i can think of is that there's a current bug in the crossfire system that when it switches from FMD2 to FMD1 the taranis stops logging data.

This appears to be exactly what it's doing. maybe there's a way to have a menu option to stop announcements on fmd change? i've created an issue in opentx as well - but i don't hold out much hope.

https://github.com/opentx/opentx/issues/6406

taranis logs are attached: 

harrier.dat - logfile from inav directory
Harrier-2019-05-02.bin - model file for companion
Harrier-2019-05-02.csv - flight log from the taranis

[taranis log.zip](https://github.com/iNavFlight/LuaTelemetry/files/3139114/taranis.log.zip)

wx4cb, 2019-05-02:
turns out it's a bug in crossfire. remi said it will be fixed in 2.95 - whenever that is

teckel12, 2019-05-02:
@wx4cb It's always a good idea to look at actual sensor values when there's a question about something.  Most of what Lua Telemetry does is take the sensor values and display them in a more user-friendly format, it doesn't invent data.  Looking at the `Alt` sensor it's fairly easy to see that it gets stuck at your highest altitude.  It doesn't seem that it ever goes down, but maybe that's because you went straight up?  Looking at my Crossfire logs I don't see that (the altitude goes up and back down).  Maybe I'm running an older version (I'll be sure to not upgrade till 2.95 is released).

I'll look into the disarmed sequence.  The problem with the `FM` sensor is that it's a string instead of a number and OpenTX can't log telemetry sensors with string values.  So, in the telemetry logs it always records `0` for the `FM` sensor value, not the flight mode, which is a real buzz kill.

wx4cb, 2019-05-02:
what version are you running on your crossfire. and don't get me wrong, i know it wasn't a bug in the script, but was trying to give more information as to what it is now that we've finally figured it out. and yes, i'm running 9.93 right now and it's definitely in there

teckel12, 2019-05-02:
@wx4cb You have voice alerts set to Critical only, so it won't be very chatty but should give you "engines armed" and "engines disarmed" voice messages.  Since the altitude alert is considered critical, you're not stopping those alerts by setting it to Critical.  Instead, you may want to set the voice alerts to "All" and set the Altitude Alert to "Off", at least until Crossfire 2.95 is released.  That's the setting that will quiet the altitude alerts.

I'll have to check my Crossfire version, off the top of my head I'm not sure, it's from a couple months ago for sure.  I don't like to update unless there's a feature or bug fix that I'm interested in/experiencing.

wx4cb, 2019-05-02:
i realise that, and i was just trying to make it less chatty to try and isolate down what the issue is. thanks for the tip i'll do that.  but yea i upgraded to try to fix another issue on the crossfire. lol

Jetrell, 2019-05-02:
Thanks for the write up guys. I have noticed this issue too, with the Crossfire. 
I like voice alerts. They tend to get my attention faster than a flashing OSD character.
The one that really bugs me. Is when she repeats 'low battery' over and over in short succession.

But like you said @teckel12 its a CRSF issue.  But I understand @wx4cb wanting to make it less chatty, by other means. Due to Crossfire 2.95 taking its time to get here.

wx4cb, 2019-05-02:
@Jetrell yea for sure lol

this was the reply to the same question in the FB group

Robert Criteser It started happening in 2.88 i believe. Or maybe the release after that. 2.95 is supposed to fix it.
Remo Masina 2.95 will fix it.
Charles Blackburn any news on when's that due to at least be betad in agent-x
Remo Masina Anytime soon. Working on it those days.

teckel12, 2019-05-03:
@wx4cb BTW, I'm on v2.88 or 2.89.  I couldn't tell which on the tiny screen that only shows the version for a second. Tried twice.

teckel12, 2019-05-03:
@Jetrell Make sure you're using the latest development build. There was a voltage bug which could cause frequent battery warnings if you don't have filtering turned on.

wx4cb, 2019-05-04:
@teckel12  ok. plug it into agent X and see what it says for the version. from what I've seen, 2.42 is the one we need to be on according to tbs unless you're using the newer receivers.

Agent X goes from 2.42 -> 2.93 -> 2.94

wx4cb, 2019-05-05:
@teckel12 - it works fine on 2.43

closed 2019-05-05 by wx4cb

teckel12, 2019-05-07:
@wx4cb I'm running 2.88, I have no issues either.