teckel12/LuaTelemetry #280: If run the Lua script in a widget,I can not flip between compass-based and launch/pilot-based orientation and radar with the dial or +/- buttons

87040109qq, 2019-05-03:
If run the Lua script in a widget,I just can not flip between compass-based and launch/pilot-based orientation and radar with the dial or +/- buttons.
But if I run LUA script  from SD card directly ,the dial works!
Is that a bug?
my sets:
horus x10s opentx 2.2.3   lua 1.6.2

87040109qq, 2019-05-03:
If there is a text note show on the screen  when flip between different mode would be great.

87040109qq, 2019-05-03:
If can not flip between different mode in widget ,add a menu option  may be good ideal or flip with stick before armed.

teckel12, 2019-05-03:
@87040109qq That's correct. You can't do that in a widget. The reason is that OpenTX has (for some reason and on purpose) blocked all non-flight controls from being monitored on Horus when running scripts as a widget. This is why the config menu can only be accessed while not armed and uses the sticks for navigation.

At some point (if OpenTX comes to their senses and removes the purposeful blocking of non-flight controls to widgets) that feature will be available. But currently, it's not, 100% due to an OpenTX decision.

teckel12, 2019-05-03:
@87040109qq Every config menu option takes a surprising amount of memory, which not much is available. When not armed, the stick toggles the min/max or current values.

The real solution to this is for OpenTX to allow widgets to see non-flight controls. There's not really a good reason why they block it anyway.

teckel12, 2019-05-03:
@87040109qq Here's the OpenTX issue for exposing key events to widgets on the Horus:  https://github.com/opentx/opentx/issues/3821

It seems that a UI refactoring is happening with OpenTX which will resolve this issue.  So, this issue will be resolved at that time.  I'll try to come up with another solution in the meantime.

87040109qq, 2019-05-04:
OK！ just wait for the time.

teckel12, 2019-05-11:
@87040109qq Latest development build adds a option for a preferences file `prefs.dat` in the `cfg` folder.  In the `prefs.dat` file you can specify your default model orientation.  See #288 for details.

For you, you probably want your `prefs.dat` to be `011`, unless you also want to try one of the new aircraft symbols (which you select with the first digit, currently `0`).

teckel12, 2019-05-16:
BTW: Global preferences have been deprecated and incorporated in the config menu due to memory enhancements from refactoring #289.  Please see the [config menu](https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings) to set these options.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12