teckel12/LuaTelemetry #281: Switch display to X-lite.

Basti77, 2019-05-04:
Short description:
There is no option for the X-lite to switch between the displays.

No key or combination switches.
you only get into a menu.

otherwise the script works fine on the Xlite

teckel12, 2019-05-04:
@Basti77 You can switch views from the config menu. Quick switching between screens is only available on transmitters that have a key to support it. Also, it's not available on the Horus as there's no key events on Horus widgets at all.

I'll look to see if there's something different with the X-lite which can be enabled. But I have my doubts due to the very limited keys.

Basti77, 2019-05-04:
To my shame, I had an older version on the transmitter.
Now I see the option in the menu to change.
for the first I can live with it.
thanks for the note

teckel12, 2019-05-06:
@Basti77 I don't have an X-Lite transmitter, but simulating it on the OpenTX Companion I can switch between views by short pressing <Enter> just like on all Taranis transmitters.  Maybe you could check again with the latest version of Lua Telemetry to see what happens when you short press <Enter> from inside Lua Telemetry?

Basti77, 2019-05-07:
> short pressing just like on all Taranis transmitters. Maybe you could check
> short press from inside Lua Telemetry?

I made an update to OTX 2.3. Hurray, it works. Thank you very much

teckel12, 2019-05-07:
@Basti77 Excellent!

closed 2019-05-07 by teckel12