teckel12/LuaTelemetry #283: New radio

wx4cb, 2019-05-07:
Just an FYI Tim.

Just got my Jumper T16 running jumper TX. it's a cheaper version of the horus X10.
base functionality seems to be there.  update rate on the AHI is slow (about 3/4 sec updates) but as a base setup it runs.

[image]

wx4cb, 2019-05-07:
Note: I don't know how to get out of the screen once it's in it though lol

teckel12, 2019-05-07:
@wx4cb Amazing that it works at all! Are you running it as a widget or a one-time script? It should work the same as on Horus OpenTX no? If a widget, long-press `Tele` button. If a one-time script, long-press `RTN`.

The update rate is no good. It's super fast on the Horus (~10 fps).

wx4cb, 2019-05-07:
It is running as a widget and it has to be set up as full screen that is the only way to run it that I can tell from what I can figure out you can only run it that way.

Its a. Fork of open tx so not officially supported by the opentx devs (this one is based on 2.2.3)

If you try and run it in any other way, it says. Must be run fullscreen. 

Im going to do some more testing when i can, im in MD the rest of the week (5am flight tomorrow)

teckel12, 2019-05-08:
@wx4cb The "Must be run full screen" message is from Lua Telemetry. You need to make Lua Telemetry full screen, just like on Horus.

wx4cb, 2019-05-08:
Yup. Thats the only way to run it. You cant run it as a normal script that
i can tell

teckel12, 2019-05-08:
@wx4cb For Horus, you should only run it as a widget anyway, running as a one-time script will work, but there's problems like resetting the flight that can't be done.

Sounds like the speed is the main issue. But that seems like something only thr radio firmware can fix. That is, unless they're using slower hardware, in which case compatibility may not be possible.

wx4cb, 2019-05-08:
yea, i have seen a couple of vids where the update rate is faster, so i
might try the release version. the ones i saw were using 1.6.0. when i get
around to testing it i'll try that too

87040109qq, 2019-05-09:
I have a jumper T16 too.The LuaTelemetry 1.6.2 can work with it.  the update rate on the AHI seems same as my OSD update rate, transfered by VTX!

wx4cb, 2019-05-09:
@87040109qq your ahi update rate. is that about 1.2-3/4 second or is it
faster than that ?

teckel12, 2019-05-10:
The screen updates are in the 8-10fps range.  It's not like a 60fps video game, but it should seem fairly smooth.  Telemetry speed itself limits this a bit, so Crossfire is much faster because telemetry updates are quicker.  As you go longer range with Crossfire, the screen will appear to slow down, but it's really the speed of the telemetry that's causing it to look like the script is slower.

If you could run the latest development version of Lua Telemetry v1.6.2 and at least Crossfire 2.88 (if you're running Crossfire) and confirm that the screen is updating fairly quickly in close range it would be appreciated.  I'd like to add this transmitter to the supported list as I feel it will be very popular.

teckel12, 2019-05-10:
Also, anyone try Lua Telemetry with the Jumper T12?  I'm guessing it will also work with that transmitter.

teckel12, 2019-05-10:
https://www.youtube.com/watch?v=pSch24lmqzw

wx4cb, 2019-05-10:
mine wasnt updating that fast... TBS just released 2.95 i believe so when i
get home i'll try it

87040109qq, 2019-05-11:
> @87040109qq your ahi update rate. is that about 1.2-3/4 second or is it faster than that ?
> […](#)

I'm not sure about that ,but it not quick enough,may be only about  1/2 second ? no idea!

teckel12, 2019-05-11:
@87040109qq Are you running Crossfire? If so, with 2.88 or later, Crossfire telemetry should be VERY fast. Look at telemetry values for the accelerometer. If they update slowly, the problem is outside of Lua Telemetry for sure.

I could post a video that shows the speed of Lua Telemetry on Horus. But as an example, there's already a video of Lua Telemetry on the main Github page. If it doesn't look similat in speed, the transmitter must be at fault.

wx4cb, 2019-05-11:
With 2.95 Beta release update rate is as it should be nice and fast hopefully they fixed all the other crap that we needed to get fixed I will find out probably tomorrow. 

Hopefully the issues with flight modes and mavlink spaffing are no longer an issue, but so far it is working as it should on the bench

teckel12, 2019-05-11:
@wx4cb I'm really interested to see if this also fixes #276

wx4cb, 2019-05-11:
I flashed it this morning and it works ok with the script on the bench so
we shall see

teckel12, 2019-05-17:
@87040109qq The latest development build of Lua Telemetry shows the frames per second in the title bar on Horus. Use OpenTX's "reset flight" to reset the FPS to get a good reading. On the Horus X10S, it's about 7 fps when receiving telemetry.

I would be interested to see what the reading is on the Jumper T16. If similar, it means telemetry is being received slow. If much lower, probably a transmitter firmware or hardware limitation.

wx4cb, 2019-05-17:
Latest dev version, Latest crossfire version on a new micro receiver and full module - was indoors so that's why the "not ok"

[image]

here's a video "in action".

https://youtu.be/8hikjiUvvDA

teckel12, 2019-05-17:
@wx4cb Can't view video, maybe it's private? The pic with a frame rate of 9.5 with not telemetry seems about right. I assume it dropped a bit once it was getting telemetry data. Like I said, not 60fps first person shooter refresh rates, but seems fairly smooth all things considered.

wx4cb, 2019-05-17:
@teckel12 sorry tim, i thought i changed it... now it's unlisted, but no the number didnt change at all went down to 9.3 but still not updating very fast, but then i was "swinging it around" so for general use without such huge updates it should be fine><

hopefully going to go fly tomorrow so we shall see

teckel12, 2019-05-17:
@wx4cb The FPS is the average since the last flight reset. So you really need to do a flight reset to see the current fps.  I have an unused switch on my transmitter configured as a special function to do a flight and telemetry reset. If running as a Widget you can also long-hold [Enter] and select reset flight. 

This shows what I was suggesting, that it could be Lua Telemetry is updating quickly, but Crossfire telemetry is what's lagging. It doesn't matter how quickly the screen updates, if the telemetry is only updating once a second, it will appear the screen is refreshing at 1 FPS as well, even thought it's not.

I'm not near a controller currently, but I believe there's a Crossfire setting for telemetry speed. There's also an OpenTX setting (having to do with the hardware hack required on Horus to get full-speed telemetry).

I looking to get an evaluation Jumper T16 to look into this first-hand as well.

teckel12, 2019-05-18:
@wx4cb It could be if you've turned on forced telemetry mode or set force RF 50Hz. I know some Crossfire long-range modes make telemetry very slow, and prioritize certain sensors over others (roll and pitch are not high priorities). Also make sure you've upgraded your Crossfire firmware.

teckel12, 2019-05-18:
@wx4cb Here's a video of Lua Telemetry on the Horus X10S with non-forced Crossfire telemetry:

https://youtu.be/qVLpryiGMLU

wx4cb, 2019-05-18:
i'll take a look to see if FT is on.

wx4cb, 2019-05-20:
it works perfectly fine, however the update rate on the screen is the same, but that's not a big deal. but as far as running on the T16. it seems to work great.

I have noticed is that the battery alerts were going down too much, but i think that's an issue with the way i have it setup in the config. Other than that, the Only thing i noticed was that i couldn't figure out how to get into the configuration menu. I tried holding the button on the roller down, and likewise with the menu button etc. but it just goes to the usual opentx menus (model menu, model select etc).

teckel12, 2019-05-21:
@wx4cb  This issue is about the Jumper T16. If you have other issues, please open other issues otherwise issues can easily be lost.

See the wiki on the config. Also, slow updates means you're in 4k telemetry mode with Crossfire. Seek Crossfire support for slow telemetry updates as it has nothing to do with Lua Telemetry.

wx4cb, 2019-05-21:
@teckel12 I know, i was just giving you observations. on the T16 it seems to work great with the exception that I can't figure out how to get into the config on the T16 as what works on the taranis doesn't seem to work on the t16

teckel12, 2019-05-22:
@wx4cb For Horus (and Jumper T16 as it's a Horus clone) you need to access the config menu using the sticks (so this only works when not armed). There's a Wiki that shows how to access the config menu on Horus (which also would be the Jumper T16).

I haven't got word on when I'm receiving my T16. When I do it will be much easier to debug issues.

wx4cb, 2019-05-26:
Quick Question, i'm getting disabled message on one of the models (latest dev version), but if i switch to my MD model, it works fine. Can't figure out why, I have the widget full screen from what I can tell.

When i go in after selecting the widget I get refresh() ?.0: attempt to index field 'I' (a nil value). could that be a "model memory" issue considering I can go to the mini drak model and it works?

teckel12, 2019-05-30:
Flew a few times with Jumper T16 and May 23, 2019 firmware. No issues.

closed 2019-06-03 by teckel12

Babych, 2019-07-08:
> Also, anyone try Lua Telemetry with the Jumper T12? I'm guessing it will also work with that transmitter.

Can confirm that on Jumper T12 works well.

teckel12, 2019-07-08:
@Babych Thanks!

scousethief, 2019-09-20:
> @Babych Thanks!

Did you get telemetry working with the T12 ?

Babych, 2019-09-20:
> > @Babych Thanks!
> 
> Did you get telemetry working with the T12 ?

Yes I did, but with frsky r9m module and r9mm receiver. With T12 multiprotocol module I couldn't get telemetry but i didn't spend much time with it.