teckel12/LuaTelemetry #284: Map view has home and model position switched on Crossfire?

teckel12, 2019-05-08:
Somehow, home and model positions were swapped on Crossfire. No idea how that could be. Need to fly FrSky and Crossfire back to back to verify. Very odd!

87040109qq, 2019-05-09:
In my case ,both  the home and model icon moves at the same time! 
In my opinion, should'nt the home icon stays at the center,and the model icon keep moving around the home?

teckel12, 2019-05-09:
@87040109qq They both can move because if home stayed in the center there wouldn't be much distance between home and the model.  In other words, it's trying to center both home and the model in the map view.  In any case, this is the correct behavior.

teckel12, 2019-05-09:
Tested with multiple controllers (Horus and Taranis) and multiple models and the home and model positions are switched. Not sure when I screwed that one up, maybe it was always that way and I just never payed attention?

teckel12, 2019-05-10:
So the bug was only on Horus and only when in the launch-based map orientation.  I use the `tmp` value too much for too many things to try and cut down on memory usage and in this case caused the bug.  Anyway, figured it out and fixed in the development branch.

87040109qq, 2019-05-11:
> @87040109qq They both can move because if home stayed in the center there wouldn't be much distance between home and the model. In other words, it's trying to center both home and the model in the map view. In any case, this is the correct behavior.

I still think that keep the home icon motionless,instead of moving, would be better! Because the pilot stay at the home location,keeping still.

teckel12, 2019-05-11:
@87040109qq Don't think of them as both moving, think of it as centering the view. It's very useful for the Q7X or Taranis in general because of the small screen size.

In reality, the code keeps home in the center position. It then has additional code to slightly shift the home position to give more granarity (so you can tell it's further away or closer). At first I had home fixed to center, it didn't work as well as you're probably guessing.

Do you have a Taranis or Horus transmitter?

87040109qq, 2019-05-11:
I have an Horus x10s

teckel12, 2019-05-11:
@87040109qq On the Horus, with the higher resolution screen and more screen size for the map, I can see how centering home and the craft location may seem different.  But, there's still two good reasons.  First, for consistency with Taranis.  Secondly, to show a better range of distance from home and not overlap the altitude graph.

There's also the issue of the fact the map is really faked for speed.  If you can believe it, GPS locations are not part of the map programming.  It draws the map with 3 piece of data.  Craft orientation, direction of craft to home position, and distance from home.  With those 3 pieces of data, it magically (with the help of some trigonometry) creates a totally fake map not based on GPS coordinates at all.

But if you really want this, you can compile your own with the following changes to the `horus.lua` file, these replace the existing define of the `d`, `hx, `hy` variables:

```
			-- Craft location
			d = data.distanceLast >= data.distRef and math.min(math.max((data.distanceLast / math.max(math.min(data.distanceMax, data.distanceLast * 4), data.distRef * 10)) * 50, 7), 50) or 1

			-- Home position
			local hx = X_CNTR + 2
			local hy = Y_CNTR
```

So while it could be done, the problem again is memory.  Each config menu option uses a lot of VERY limited memory (this isn't a desktop computer or your phone).  If it will ever be possible to incorporate VTx control, I can't be using the last bit of memory on every little tweak to the UI.  Basically, I need to set priorities because not everything will fit.  We're already bumping up against a memory and computation speed wall with this script.  Think of it as already at 99% of capacity, which isn't much of an exaggeration.

But in any case, if you'd like to open an issue for this (because it's not the same as this current issue) it can at least be tracked if memory will allow it or if there's another way.  The problem with chiming in with a different subject in an issue is that this issue will be closed automatically when v1.6.2 is released, as this issue is about something different.  So, creating individual issues is very important to prevent things being closed and forgotten.

teckel12, 2019-05-11:
@87040109qq See #288

teckel12, 2019-05-16:
BTW: Global preferences have been deprecated and incorporated in the config menu due to memory enhancements from refactoring #289.  Please see the [config menu](https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings) to set these options.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12