teckel12/LuaTelemetry #287: Additional aircraft symbols on attitude indicator for Horus

teckel12, 2019-05-11:
### Is your feature request related to a problem? Please describe.
Current aircraft symbol is boring!

### Describe the solution you'd like
Option to switch aircraft symbols

### Describe alternatives you've considered
Dealing with it

### Additional context
Here's the current options I've put together:

[image]

I was going to try and name them, but having a hard time.  The current one (first one in the pic) would be the `Boeing`, the next would be `Classic` maybe?  The next two would be `Garmin1` and `Garmin2`, while the second to last (plane) would be `Dynon`.  The last one is really troubling, I can find no name for this type of aircraft symbol, so I'm calling it `W` till someone tells me otherwise.

But, if I've made a mistake in my current naming, let me know.  I asked an actual pilot friend of mine and he had no idea what the different types of aircraft symbols were named.  Maybe I'm just too much a stickler for details being right.

Thoughts, opinions?

wx4cb, 2019-05-11:
The last one is just a level flight and bank indicator combined. The slants
are usually at 20 degrees iirc.

2nd to last is usuallt a dual indicator with the circle and lines split
horizon line and flight path indicator where the circle represents the
actual path  as you could show level but be descending like in a flat
stall. Same for the Boeing and the airbus ones 1st/2nd

teckel12, 2019-05-11:
@wx4cb For sure, the aircraft symbols are not sophisticated like commercial-grade glass cockpits.  It's meant to be more of a style option as they're all fixed in place.  Eventually, I'd really like to incorporate indication of orientation vs. flight direction as well as wind speed (or at least direction).  INAV is dabbling with these now, although I need to verify if Crossfire is already technically supplying orientation vs. flight direction.  With orientation vs. flight direction, it's possible that I could calculate wind direction.  I only fly quads, so it may be a bit of stretch for me as it's really irrelevant.

Anyway, it seems naming the last one `Waterline` seems like a good move.  The second one I see in "classic" attitude indicators, I know it's used in other cases, but would `Classic` be a good name?  Should I remove the second to last as it would be confusing, especially if  we later have additional telemetry which could indicate flight path vs orientation?

wx4cb, 2019-05-11:
crossfire telemetry has rotation - by default in rads but can be set to degrees (Ptch/Roll/Yaw) so obviously if your roll > 100 degrees you're almost upside down etc

wx4cb, 2019-05-11:
as far as the icons... personally I find the waterline too big.... but i'm used to having just a dot or the 2nd to last one for the middle. maybe keep them as the same ones you can use in inav might be the way to go?

teckel12, 2019-05-11:
@wx4cb I made the Waterline aircraft symbol a bit thinner.  INAV is character based and B&W, so it would be limiting to copy INAV.

wx4cb, 2019-05-11:
> @wx4cb I made the Waterline aircraft symbol a bit thinner. INAV is character based and B&W, so it would be limiting to copy INAV.

no what I meant was, copy the designs that are on there so people can use the same ones on the osd, on the radio.

teckel12, 2019-05-11:
@wx4cb Is there a page which outlines the aircraft symbols in INAV OSD?

teckel12, 2019-05-11:
BTW, I've added multiple aircraft symbols to the development branch.

teckel12, 2019-05-12:
[image]

wx4cb, 2019-05-12:
i would guess it would be referenced some how by

OSD_CROSSHAIRS

typedef enum {
    OSD_CROSSHAIRS_STYLE_DEFAULT,
    OSD_CROSSHAIRS_STYLE_AIRCRAFT,
} osd_crosshairs_style_e;

^^ those are in osd.h in main/io there's no wiki pages in the configurator and no other mention in the main wiki.

teckel12, 2019-05-12:
@wx4cb Those two I remember, I figured there was more that I wasn't privy to. As long as the ones I've shown here are not broken in some way I'll add them. We can always fix/extend later.

wx4cb, 2019-05-12:
there were more in the development version, oliverC was working on some and then that got pulled in favour of the lora thing.

teckel12, 2019-05-12:
@wx4cb That must have been what I was thinking of. I'll look around.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12