teckel12/LuaTelemetry #288: Create preferences file for basic style options that wouldn't change per-model

teckel12, 2019-05-11:
@87040109qq Regarding #284 I've been giving this "Preferences" some consideration and thought today.  Maybe certain options could be done in a preferences file instead of the in-script configuration menu.

teckel12, 2019-05-11:
Latest development build adds an optional global preferences file `prefs.dat` in the `cfg` folder.  The file allows you to set certain features globally to your own preference.  It currently supports 3 settings (each one digit so 3 digits long).  The 3 settings are as follows:

1. Aircraft symbol (Horus only): 0 = Boeing/Airbus (default), 1 = Classic, 2 = Garmin1, 3 = Garmin2, 4 = Dynon, 5 = Waterline
2. Radar home position (Horus only): 0 = Adjust to fit (default), 1 = Fixed in center
3. Default orientation: 0 = Launch/pilot-based (default), 1 = Compass-based

Example: `prefs.dat` with the content: `511` would use the waterline aircraft symbol, fix home position to the center in the radar view, and make the default orientation compass-based.  All options default to `0` so if the file doesn't exist or doesn't have enough digits it will use the default option.

87040109qq, 2019-05-13:
Thanks for your great idea! Now we can  select home positon through prefs.dat ！

teckel12, 2019-05-13:
@87040109qq Not sure what your question is referring to.

teckel12, 2019-05-16:
BTW: Global preferences have been deprecated and incorporated in the config menu due to memory enhancements from refactoring.  Please see the [config menu](https://github.com/iNavFlight/LuaTelemetry/wiki/Configuration-Settings) to set these options.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12