teckel12/LuaTelemetry #291: Fuel Percentage incorrect (slowly counting up exceding 100%)

jm-wijnholds, 2019-05-15:
After following the steps in de wiki, i still can't get the fuel gauge to work correctly. When i plug in the battery it show's 141% when a 4S battery is at 15.4v. (It''s also counting up, exceeding 100% as it was still using Mah instead af percentage). iNav is configured in precentage also: set smartport_fuel_unit = PERCENT

The percentege shown in iNav is also different than the values shown by the lua script

wx4cb, 2019-05-15:
just a thought off the top of my head, in the telemetry sensor page. is it
in MAH or percent? You might need to change the unit of the sensor itself?

jm-wijnholds, 2019-05-15:
My bad, after doing a dump in the CLI, set smartport fuel unit was not set to percent. Did it again and now its working as it should. Sorry for the hassel :)

closed 2019-05-15 by jm-wijnholds