teckel12/LuaTelemetry #298: crossfire Diversity not working 100%

advancetom, 2019-05-26:
hey today im tested the lua with Diversity and not working 100% CF Version is 2.95
,but if connect to a other crossfire receiver Micro V2 it working fine :(
Seems like it hang or else with Diversity connected?

wx4cb, 2019-05-26:
technically it shouldn't matter what receiver is connected as csrf is csrf. when you say it's not working 100%, what part isnt?

teckel12, 2019-05-26:
@advancetom Don't follow what isn't working.  It's also a bit confusing as at first you say it's not working 100% (suggesting it works but not perfectly) and later you say something about it hanging (which suggests it's not working at all).  Seems like you're both saying that it's working 99% and 0%.

If you could describe what isn't work that would be very helpful.  Also, what exact transmitter and receiver are you using?  (there's a bunch of other important issue questions that were also glossed over).

The INAV Lua Telemetry script just reads the telemetry data and displays it.  If something doesn't work with a particular receiver, it sounds like the receiver isn't outputting that information.

advancetom, 2019-05-26:
im tested now with development branch and newest Firmware from Jumper.xyz
Micro V2 Crossfire works awesome.
If i switch to Diversity Receiver from Crossfire telemetry is not find all sensors and feels like interupted.
I cannot better explain the behaviour but im confused why?
 Crossfire version is 2.95

wx4cb, 2019-05-26:
make sure you delete and rediscover the sensors between radio changes,

advancetom, 2019-05-26:
So i solved the problem Inav shows me that receiver works but TX and RX was crossed on receiver ...OMG
After discovering new sensors it worked !!!!So only FM shows !ERR failure without modes configured yet i think :)

teckel12, 2019-05-26:
@advancetom Glad you got it working. Note that the Jumper T16 isn't on the supported list yet. I won't be getting one till later in the week.

The FM sensor will show `!ERR` (and the telemetry script will report `NOT OK` if you don't have a GPS fix or there's some kind of sensor error (for example, you haven't calibrated the accelerometer or magnatometer yet).

advancetom, 2019-05-27:
I can report That the Jumper T16 works with development Version 1.6.2 and Crossfire Version 2.95 and latest Firmware from Jumper.xyz website

closed 2019-06-03 by teckel12