teckel12/LuaTelemetry #301: Horus: Preview airplane symbols in config menu

teckel12, 2019-05-28:
### Is your feature request related to a problem? Please describe.
Hard to know what the airplane symbols look like till you change them.

### Describe the solution you'd like
Show what they look like as you change them.

### Describe alternatives you've considered
Living with the way it is now.

### Additional context
Maybe move the config menu to the left when on the Airplane Symbol option so a sample can be shown to the right?  I added code to show a preview at the bottom of the screen, but that no longer works after other changes were implemented. Should mostly be refactoring code.

wx4cb, 2019-05-28:
I was wondering why they disappeared when i upgraded. Personally (and im guilty of this) just point people to the wiki, or at least use the name if you can and not a number, then point them to the wiki heh

teckel12, 2019-05-28:
@wx4cb I had already written the code to show a preview, I just needed to tweak it a bit after other changes were made.  It's in the latest development branch now.

PR #274 (2019-06-03): v1.7.0
* Logs altitude graph in background so changing views doesn't affect altitude graph
* Refactored config options which saved a ton of memory
* Total rewrite of artificial horizon and pitch ladder, faster and better looking
* Horus: Optional roll indicator at top of attitude indicator
* Horus: 5 different aircraft symbols on attitude indicator (preview in config menu)
* Added support for [Jumper T16](https://www.jumper.xyz/) transmitter (operates just like Horus)
* Resolved bug that caused RSSI to always show 99dB on pilot, radar and altitude views
* Resolved bug that caused low voltage v [... cut]
closed 2019-06-03 by teckel12