teckel12/LuaTelemetry #302: Update language file for new config menus

teckel12, 2019-05-28:
### Is your feature request related to a problem? Please describe.
New config options added that need to be translated to other languages.

### Additional context
The following config menu options need translation:
- Aircraft Symbol
- Center Map
- Default Orientation
- Roll Indicator

wx4cb, 2019-05-28:
I can help woth german if you give me the English language text you want

teckel12, 2019-05-28:
@wx4cb For the menu items there's a max of 16 characters.  For the options (Launch/Compass) the max is 8 characters.  Only the Orientation options need translation because the others are Yes/No or a number.

teckel12, 2019-06-29:
I'm going to let Google Translator do this. Then if wrong someone can make a correction later.

PR #310 (2019-06-29): v1.7.1
- Uses "flat earth" algebra for faster orientation/distance calculation
- Crossfire on Horus: Adds flight path vector indicator (currently only horizontal axis)
- Crossfire: Resolved default orientation bug
- Horus: Nice 15% speedup 
- Horus: Show satellite and HDOP (on S.Port) on config menu
- Horus: Changes to the config menu controls
- Horus: Trim 6 can be used while flying to switch between launch and compass-based orientation
- Horus: Added alternative location for speed and altitude units when certain options are enabled
- Horus: Distance value in radar always shows current distan [... cut]
closed 2019-06-29 by teckel12