teckel12/LuaTelemetry #311: Memory issue X9D

gmacfpv, 2019-06-07:
Having what seems like a memory issue with the X9D. There is a lot of lag with audio alerts and general telemtry information.

Worst case scanario was out flying today and was constantly recieving "disarmed" messages and flight mode alerts while flying in manual/passthrough. Better case scanario is that it all works fine but a lot of delay between mode change and mode change alert.

On some other occasions I don't have any of these issues which makes me think it is purely a memory issue.

Is there a way to improve this?

Thanks

teckel12, 2019-06-07:
@gmacfpv. Are you running INAV LuaTelemetry 1.7.0, OpenTX 2.2.3, and INAV 2.2.0-RC2? If not, upgrade first.  Also, what telemetry protocol are you using? Crossfire? FrSky? 

I've never encountered anything like that and I've never heard a report of anything like that either. While it could have something to do with very limited memory, it seems like something else. 

The only way to reduce memory usage is to remove other Lua scripts that may be using memory.  For example, if you're trying to use the Betaflight Lua Script for VTx control that will never work with INAV Lua Telemetry.

Other Lua scripts will also use memory as well. To see the available memory, from the transmitter's home page long-press ENT, select Statistics, press PAGE. The `Free Mem` value is what you're looking for.

But like I said, it sounds like something else, and I'd first suggest upgrading everything.

mastersurferde, 2019-06-11:
@gmacfpv: If you use crossfire please update to 2.95 beta. The telemetry in this version is much better then the versions bevore. It is not perfect but flyable.

teckel12, 2019-06-11:
@mastersurferde @gmacfpv It does sound more like a Crossfire issue.  Every so often I get some lag for a few seconds just after power-up of the quad/receiver.  It's really laggy (maybe a full second delay) and then clears up in a few seconds.  It follows the startup of receiver, so it's clearly a Crossfire protocol issue. 
 I don't recall if the problem has surfaced since upgrading to 2.95 beta.  But it's really not much of an issue as it typically happens when waiting for satellites anyway so I don't even notice.  It's never been a problem during flight.

wx4cb, 2019-06-11:
i've found it's only while i'm waiting for sats that I notice it, but then, before I arm, i always do a reset flight and it seems fine after that. until of course the RFMD bug happens :D

teckel12, 2019-06-11:
@wx4cb I'm sure that lag has nothing to do with INAV Lua Telemetry, it's like Crossfire is pushing too much data and is buffering or maybe INAV is trying to push telemetry info to Crossfire too quickly while it's booting or whatever.  In any case, it's something to do with Crossfire, doesn't affect flights, and doesn't sound the same as this issue.

I thought 2.95 no longer makes `RFMD` stick?  It causes your RSSI to drop and stay low, right?  Maybe if failsafe was working on INAV I'd fly far enough away to test this.  But I'm staying close to home till failsafe is sorted out.

wx4cb, 2019-06-11:
@teckel12 yea it was SUPPOSED to fix it, but it hasn't. as far as failsafe, that works fine :heh:

wx4cb, 2019-06-11:
@teckel12 it's for sure the crossfire issue i don't think it's anything to do with the script

closed 2019-08-02 by stale[bot]