teckel12/LuaTelemetry #315: Excessive audio warnings with X9D and CRSF

Jetrell, 2019-06-19:
I used Lua telemetry 1.7 today for the first time. With the hardware mentioned in the title. And got an ear full of repetition.
I can't think of the exact wording... But Amber kept repeating motors arm/motors disarmed. And GPS good/ GPS lost.   
It only did it in Horizon mode. And would stop in Air mode.

She has often had a few things to say before. But this was beyond the normal repetition....  It was more like being nagged at :-)  It just went on and on and on; when there was no issue.
So I thought I better mention it,

Thanks

teckel12, 2019-06-19:
@Jetrell Sounds like maybe you're flying an older version of INAV or Crossfire?  The kind of messages you're getting means that the `FM` telemetry sensor is all over the place.  There's telemetry bugs in older versions of Crossfire, make sure you're using v2.95+.

Jetrell, 2019-06-19:
@teckel12  Well actually. I'm using the latest Crossfire firmware 2.93 and iNav 2.2 rc2. The X9D has OTX 2.2.3.
 When I was at the field yesterday. I placed LT 1.61 back into the Taranis. With no other changes to anything.... And the repetitious audio warning ( GPS and Arming) went away.

teckel12, 2019-06-20:
@Jetrell There's known telemetry issues with Crossfire 2.89-2.93. The pre-release 2.95 may correct these problems (it does for me and others). Rolling back to 1.6.1 may be coincidental, as the Crossfire telemetry issues are only under certain circumstances. 

I'm running Lua Telemetry 1.7.0, Crossfire 2.95, INAV 2.2RC2, and OpenTX 2.2.3.  So we have a similar setup (except for the Crossfire firmware version).

Technically speaking, the arm/GPS messages are just being reported based on what the Crossfire `FM` sensor is reporting. What this means is that this sensor's value is jumping all over the place, my guess is going from real values and `0` or empty.

You could also try the latest Lua Telemetry development branch, (pre-release 1.7.1). Although I don't believe there's been any changes that would change your situation.

jasc76, 2019-06-20:
Thanks for the tip.
CRSF 2.95 fixed it for me

closed 2019-07-07 by Jetrell