teckel12/LuaTelemetry #320: Not sure if intended

wx4cb, 2019-06-28:
Last stable release, just something i noticed. Maybe an unintended consequence and irrelevant,  but i just noticed this. Apologies for the weird way my phone put the video up to YouTube 

[Video](https://youtu.be/oPfkOEdTRjY)

teckel12, 2019-06-28:
@wx4cb That's a feature.  I believe you're referring to the min/max values?

When you're not armed and you move the right stick up/down it toggles between current values and the min/max values.  It uses the stick because on a Horus widget you can't use the control buttons.  The version you are running denotes this with the small arrows.  The development release now has one central `Min/Max` message in the center of the screen and can also be selected using the Trim 6 button (which is also used while flying to switch between launch and compass-based orientation).

It's not very useful before you fly, but after you fly and disarm you can use this to not only show the last values, but also the min/max values.  For example, your max speed, lowest battery voltage, lowest signal strength, or highest current (among other things).

The ability to show the min/max values has been around since the beginning of Lua Telemetry, and is supported on Taranis and Horus.  The Taranis uses the control wheel or arrow keys, but with Horus that's not an option due to limitations that OpenTX put on Horus widgets.  If you run Lua Telemetry as a telemetry script instead of a widget, you can use the wheel to toggle between current and min/max like on the Taranis.

In any case, a feature, totally intended, but doing circles with the stick isn't what's causing it, just the up and down part of the circle.

closed 2019-06-28 by teckel12

wx4cb, 2019-06-28:
ahh that makes sense.... i figured it was supposed to do that, just
never noticed it before LOL