teckel12/LuaTelemetry #328: GPS data in event of loss of signal.

mikekenyon99, 2019-06-30:
### Describe your question
I have a Taranis x9D and I use the log function to store my data in case I lose my craft.  Is it possible to review this in this tool? Or does it freeze the last known position of the craft?

Thanks 

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. Should it be doing this?

### Expected behavior

### Screenshots

### Additional context

teckel12, 2019-06-30:
@mikekenyon99 It shows the last good GPS location (designed to help locate in case of a "hard landing").

With that said, the ability to playback the telemetry logs is something that I've considered. It wouldn't work perfectly (for example, with Crossfire flight mode information isn't logged due to an OpenTX bug).  But, it could be useful or just interesting to review.

I was going to add this as a future feature anyway, so I'll mark this issue as such.

mikekenyon99, 2019-06-30:
I think the ability to review the logs for the currently used model would be great. 😀 Unfortunately I have had lost planes and the only way I found them was by taking the SD card out of the transmitter and using the gps values from the log file.   This meant I had to drive home and then go back to flying location.

teckel12, 2019-06-30:
@mikekenyon99 Ultimately, wouldn't the plane be closest to the last GPS location?  I get that there could be unique situations where maybe telemetry is lost on the way down and getting the direction, altitude, etc. could be used to better calculate where the model would be.  But, INAV Lua Telemetry also shows that information too (all the last telemetry values are shown, not just the GPS coordinates).  So, without any changes, you'll be able to see the GPS location, direction from your launch position, altitude, speed, direction flying, etc.  Wouldn't this be what you needed?  What additional information would you get from the telemetry logs?

mikekenyon99, 2019-06-30:
You are completely right and if it behaves like that then perfect :-) Though I'm unsure, what happens in the scenario where there is no signal from the Crossfire (plane buried in ground, powered off etc..) does the Taranis X9D write zeros every 5 seconds as it writes data?  If so then you would have to scroll back through the log to find the last non-zero record.

teckel12, 2019-07-05:
Heh, the logs playback is a riot! Super fun to play with!

I'd say it works fairly well on Horus. On Taranis it works too, but it's really eating a bunch of memory! I'll see what I can do, but it may be a Horus-only feature. Or, if I can't reduce the memory on Taranis, I may have to scrap it totally :(

teckel12, 2019-07-05:
@mikekenyon99 Log playback is in the `logs` branch if you want to try it out. I'm on my phone now so I can't give detailed instructions. But you basically download the `logs` branch zip file, and copy the `TELEMETRY` folder to your transmitter's SD card. I can give detailed instructions tomorrow.

But it does work and it's really fun to playback flights! Needs a bit of polish and seek controls, but you can stop. Also, arming will stop automatically.

mikekenyon99, 2019-07-05:
Hey great work there! unfortunately Im away from my gear for a few days (holiday) so I will have to test next week.  I hope you can get it to work on the Taranis without the memory crash.  I will report back. :-)

teckel12, 2019-07-07:
To upgrade to the log build, use the following instructions:

1. Download [LuaTelemetry-log.zip](https://github.com/iNavFlight/LuaTelemetry/archive/log.zip)
1. Open or extract the `LuaTelemetry-log.zip` file
1. Open the `LuaTelemetry-log` folder and navigate to the `dist` folder
1. Copy the contents of the `dist` folder (`SCRIPTS` and `WIDGETS` folders) to the transmitter's SD card's root

Jetrell, 2019-07-10:
@teckel12 Do I read this right. The log build will work with an X9D?
Mine comes up as -  `Review logs --`
Am I missing something?

mikekenyon99, 2019-07-10:
Hi, I just uploaded this to my Taranis X9D.  I turned on TX, then went to the settings and selected a previous log file.  I then backed out of the settings and got a 

> Script syntax error
> ?:0
> attempt to call a nil value

Mike

teckel12, 2019-07-10:
@mikekenyon99 Out of memory, or you don't have the `luac` build option checked for your OpenTX firmware.

mikekenyon99, 2019-07-10:
Thanks - just built with luac option to make sure its not that.  I still get error message about nil value and once I have had out of memory.  Interestingly it's not listing all the log files to review, in the example below the model "Defiant28" has log files but none are listed?    

[image]
[image]

teckel12, 2019-07-10:
@Jetrell I'm running the `logs` branch on my Q X7, and it runs on the X9D simulator.

On startup, it looks for logs that have been created in the last 2 weeks for the model you currently have selected.  The `--` means it couldn't find any log files in the last 2 weeks.  Your log files will be in your `LOGS` folder and will start with your model's name.

I just thought of something.  If your model name includes a space, it appears OpenTX changes spaces to underscores.  I've changed the latest `logs` branch to work with model names with spaces.

Also, the latest `logs` branch includes other features, like fast forward/reverse using the pitch/elevator right stick and a timer indicator on Taranis.

teckel12, 2019-07-10:
@mikekenyon99 It's the space in the model's name.  OpenTX changes spaces to underscores, which I wasn't doing.  Try the latest `logs` branch, it now works with models with spaces in the name.

Also, the latest commit includes other features, like fast forward/reverse using the pitch/elevator right stick and a timer indicator on Taranis.

mikekenyon99, 2019-07-10:
Really appreciate your work here, but Im getting these errors, guessing I don't have enough memory :-(
"not enough memory"
"attempt to perform arithmetic on a nil value"

teckel12, 2019-07-10:
@mikekenyon99 You must have something using memory, a different firmware version of OpenTX, or something as it's working on my Q X7.

Startup your transmitter and before doing anything, long-press `Enter`, select `Statistics`, hit `Page` and note what it says for `Free Mem`.  If it's less than 30,000, you must have something eating up memory.  Do you have any other lua scripts on other pages?  Excessive models?

teckel12, 2019-07-10:
@mikekenyon99 Oh, it would also be more possible to work if you were on a view that used less memory, like the classic view.  The pilot view uses the most memory so if you were on that view when you started the  log playback it has a higher possibility of running out of memory.

The Taranis doesn't have much memory to begin with, and OpenTX doesn't allocate much for lua scripts.

mikekenyon99, 2019-07-10:
@teckel12 thanks for that my free memory is 28336, so that's my problem.  Interestingly, I only have 5 named models, using Crossfire Lua scripts that's all.  I will get Googleing to sort that first and then report back.  Cheers!

[image]

teckel12, 2019-07-10:
@mikekenyon99 What version of OpenTX are you running?  I believe I'm running v2.2.3.  I think older versions of OpenTX didn't allocate as much memory to lua scripts, but I'm not 100% certain.

mikekenyon99, 2019-07-10:
@teckel12 using v2.2.3, just checked in classic view and I either get the:

> "attempt to perform arithmetic on a nil value" 
or
> "out of memory"

happy to send you the log files if you want to test?

teckel12, 2019-07-10:
@mikekenyon99 The log file would at least allow me to rule other things out.  It's possible (for example) that you have missing sensors in your logs.  It's expecting the log file to match the sensors you currently have.  If something is missing, it could also cause this error.

mikekenyon99, 2019-07-10:
[Archive.zip](https://github.com/iNavFlight/LuaTelemetry/files/3379391/Archive.zip)
Here you go, Im off to bed now (UK based) :-) thanks for your help!

teckel12, 2019-07-11:
@mikekenyon99 Okay...  Reviewed your logs.  First off, you're not recording telemetry data.  If you look at your logs, the telemetry sensors are all blank or zero (except for the `RFMD` sensor which appears to be set to log).  You need to set each telemetry sensor to log or it won't.  So, that's the first problem which will cause this not to work, as it's expecting all of the telemetry sensors to be in the log file.

So, you'll need to set each telemetry sensor to log and then do another flight.  Looking at the logs, you'll see the difference right away as there will be data in your logs after you do this.

Also, your memory situation is probably going to make it difficult to work anyway.  When I boot my Q X7 before running anything else I have over 37,000b of free memory.  That's the free memory after running Lua Telemetry's bootup and background process.  It will go about 10k less once you view Lua Telemetry as the view needs to load.

mikekenyon99, 2019-07-11:
Thanks for helping, I have just checked and I do have logs ticked against the telemetry sensors.  This file is an example (attached), so I just renamed the filename so it would show up.  I loaded the file and got the:
> "attempt to perform arithmetic on a nil value"

looking at the file, sure there are 0's because I turn on the TX and it starts logging, I then plug in RX, fly, then land.  So the first rows are 0's, however further down there are values. 

[CaipirinhaX-2019-07-01.txt](https://github.com/iNavFlight/LuaTelemetry/files/3383834/CaipirinhaX-2019-07-01.txt)

wx4cb, 2019-07-11:
@mikekenyon99 don't have "SD Log" turned on all the time, only have it on when it's armed. i have mine in a special function so that when the arm switch is "on" the sdlogs is active, you'll avoid that issue then

mikekenyon99, 2019-07-11:
@wx4cb I thought about that - what happens if something goes wrong and it powers off into the distance? I had this very incident when my RTH failed to work.  I hit "disarm" which cut motor (thankfully) but the logs stopped.  It only occurred to me about 5 minutes later to undo this so I could get the latest GPS position from telemetry.    The solution is to use another switch, I know... :-)

teckel12, 2019-07-11:
@mikekenyon99 The previous logs you provided had zeros all the way to the bottom of the file. I'll look at your latest ones.

Try using the latest `logs` branch, as changes just today made it more robust to avoid errors. If you're getting an error with today's commit, it's almost surely not enough memory.

Maybe you have an OpenTX build option selected that's causing a bunch of memory to be used? You can uncheck all options (including lua and luac even) and checking all the 'no' options as well as checking the 'eu' option will save a bit of memory. Not sure why you have 10k less than me.

teckel12, 2019-07-11:
@wx4cb I have it on my arm switch too, and I would only disarm in a crash, and then nothing is moving anymore anyway. But I only fly quads.

wx4cb, 2019-07-11:
@teckel12 there should be no reason to disarm while flying. as you say if it's crashed, nothing's moving anyway.

in the case of a failsafe, then it should be RTH (imho) which if it's gone into failsafe then you're not getting data anyway most likely otherwise it wouldn't be in failsafe.

Jetrell, 2019-07-11:
Great read guys.
Thanks for all your work here Tim :-)

@wx4cb  I have my log enable with a mixer that needs arming and the throttle over a set value.
Plus iNav won't disarm a plane if the throttle is above Min throttle value, even if the arm/disarm switch is accidentally bumped.

teckel12, 2019-07-12:
And the problem with disarm is that you need to remember to put the throttle low to arm, so I can then be in a panic trying to arm and it won't because your throttle isn't low.  I'm in the process of building my first FW, so I'm sure I'll have different opinions on this from a FW perspective.

Jetrell, 2019-07-12:
@teckel12 it can be an issue, until you get used to it. But having the model disarm accidentally, can also be a problem. 
At least with a FW, you have time to think it through and then re-arm. Because unlike a quad, a FW still glides on.

mikekenyon99, 2019-07-13:
Just downloaded the latest build - works a treat!!!  Thank you very much :-)

PR #336 (2019-08-25): v1.7.2
* [Playback log files](https://github.com/iNavFlight/LuaTelemetry/wiki/Playback-Telemetry-Log-Files) (may run out of memory on Taranis)
* Runs without luac or lua build options (broke in v1.7.1)
* Refactored code for higher speed and reduced memory usage
* Fix for french translation of OpenTX with the `RxBt` sensor labeled `BtRx`
* Crossfire: Add vertical speed option
* Crossfire: Fix for Q X7 Tx battery options
* Crossfire: Fix distance calculation errors
* Crossfire: Resolved edge-case yaw issue due to Crossfire rollover bug
* Horus: Home indicator now shows as a HUD indicator instead [... cut]
closed 2019-08-25 by teckel12