teckel12/LuaTelemetry #329: 1.7.1 cannot enter/exit menu (T16)

wx4cb, 2019-07-04:
### Describe the bug
Part 1:
Cannot enter configuration menu for script on T16 when used as a widget and the aircraft is unplugged - eg just after landing and unplugging battery.

When plugged back in you can enter the config menu.

Part 2: Cannot use RTN (button above scroller) to exit configuration menu.

### To reproduce
Plug in aircraft
arm
wait a few seconds (or go fly)
disarm
unplug battery
get the haptic feedback once per second and "telemetry lost" voice alert
try to enter config menu using stick commands - thr: top right, right stick neutral

teckel12, 2019-07-04:
@wx4cb I can't duplicate. But what concerns me (and I believe the real problem) is that after you disarm then unplug the battery, it will mention telemetry lost, but there shouldn't be any vibrations once per second. It should only do that if it believes you're still armed.

So, do your test again, but this time when you disarm note what happens. What does the flight mode say? Does the flight timer in the title stop?

wx4cb, 2019-07-05:
Yea ignore the haptic stuff that works fine, but i still cant exist the
config, unless im doing something wrong

teckel12, 2019-07-05:
@wx4cb The one thing you mention (the return above scroll bar) will never work as enter, nor will exit or any other control buttons. None of the controls will work in a Widget. This is an OpenTX limitation. Which is why you must use the stick commands and it only works when not armed.

When running as a widget, you must use the stick commands to enter/exit the config menu as well as change settings. There's a graphic in the wiki on how to navigate the config menu as a widget. None of the buttons will ever work (OpenTX doesn't expose the buttons to widgets for some stupid reason).

teckel12, 2019-07-05:
@wx4cb If your talking only about exiting the config, you just move your right stick to the left. Basically, right is enter, left is exit, and up/down are as you expect.

wx4cb, 2019-07-05:
ok, that makes sense then... I'll look at it again, sorry to be a p.i.t.a lol

closed 2019-07-05 by wx4cb