teckel12/LuaTelemetry #330: Not a bug, just an fyi

wx4cb, 2019-07-07:
The beta for crossfire 3 was released yesterday,  i ha ent flown it yet, but hopefully they fixed the logging bug. Just thought id let you know.

Ill be flying it either later today or tomorrow so will keep you updated

teckel12, 2019-07-07:
@wx4cb I hope they also fixed the heading and yaw rounding overflow bugs. But, it's only a issue between a heading of 327-359 degrees and it didn't seem like TBS believed the problem was them (even though it happens in both Betaflight and INAV and I made a custom INAV build and couldn't get around the overflow issue).

Anyway, I'll check out 4.0 and make sure things are working correctly (and hopefully bugs fixed). Thanks for the heads up!

teckel12, 2019-07-08:
At least the `YAW` rollover bug is still there, and I assume `Hdg` as well.  Everything else seems to work with 3.00, but I'll test later today to be sure.

I do like the new `RF Profile` option where you can specify 150Hz!  Keeping the telemetry smooth when not doing long-range as a plus.  Maybe I'll try longer-range and see how 150Hz works.

wx4cb, 2019-07-08:
be careful with that. i just did 2 flights, dynamic rf on and the FMD bug is still there (taranis stopping logging and telemetry over BT stops too - comes back on when it goes back into rfmd2)) HOWEVER.... I then flew it with force 150hz on and the inav rssi never left 99 but I failsafed at a mile which it shouldn't do in this wide open area and don't have any issues when it drops down.

RF Power output was at 100mW so shouldn't have an issue with that, and i tried multiple antennas.

I have video and logs i'll send you when i get home

teckel12, 2019-07-08:
@wx4cb I'm confident with my INAV failsafe RTH, but thanks for the warning.  Maybe I'll set the output to 10mW and keep the quad low and over grass as a test for the force 150Hz.  Then maybe pump it up to 2W and go a mile out and see what happens.  What could possibly go wrong? ;)

I've never experienced the FMD bug, but I was always using CRSF 2.88 and later.  I also typically do a 2W output power when going long-range, so maybe I just don't trip FMD that frequently.

wx4cb, 2019-07-08:
@teckel12 according to remo (and bear in mind 3.0 is in beta) is a known issue that he will fix so that's a start.

i've never needed to go any more than 100mW as i usually only go 2 miles out max. I don't know where to get 2.88 from, agent-x only goes from 2.43 to 2.93/2.95.

I just got home so i have to pull all the info/logs etc from the laptop, taranis and the video

wx4cb, 2019-07-08:
here's my taranis logs if they're helpful for you and also the MWP logs for the first flight.

all in all it's a lot better, but that thing is still bugging  the crap out of me - pardon the pun.

[07.08.2019 - crossfire rfmd taranis and other logs.zip](https://github.com/iNavFlight/LuaTelemetry/files/3369704/07.08.2019.-.crossfire.rfmd.taranis.and.other.logs.zip)

teckel12, 2019-07-09:
@wx4cb For fun, I reviewed your telemetry logs with the new log playback feature in Lua Telemetry. It's almost as fun as actually flying (well, maybe not with your logs because of the telemetry lockup). Here's a video of it in action:

https://youtu.be/r2ChtoZG7k0

Note that you had two flights on the same day so it plays back both back to back. Also, you had some corruption in the file that I had to remove. Could be a bug in OpenTX or maybe an SD problem?

Anyway. I don't experience what you do. When I go to 50Hz, telemetry just DRASTICALLY slows down (like to maybe 1-2 changes a second). Which is (I believe) by design. I've been running 2.88 and 2.95 (plan on trying 3.0 in a bit). 2.88 seems to be no longer available, but it was at one time with the old TBS flashing app (could still be, I didn't check). 

Thinking maybe this is a receiver issue? What Rx do you have? I use the micro receiver v2 and have never had a telemetry lockup like your logs show.

wx4cb, 2019-07-09:
Unable to watch you don't work however I do have one question when you played the logs back the telemetry froze correct I'm wondering whether the telemetry freezes when it happens actually in flight I haven't looked at that that's 1 thing that ramo wanted to know

 I spent most of the morning chatting with ramo from TVS the logging bug apparently is a bug and that should be fixed in the actual next release as far as my FMD1 bug goes it's not actually a bug it turns out that when it goes into FMD1 and you have the module in bridge mode bridge mode does stop the ony supported protocol is mad link in FMD1

closed 2019-08-30 by stale[bot]