teckel12/LuaTelemetry #331: Frsky X9 lite attempt to call a nill value

P-I-Engineer, 2019-07-08:
This is on opentx 2.3, 
frsky firmware x9lit_190605

script syntax error ?:0
attempt to calla nil value

has anyone tested the x9 lite yet?

thanks

teckel12, 2019-07-08:
See: https://github.com/iNavFlight/LuaTelemetry/issues/new/choose, first issue option.
Also see: https://github.com/iNavFlight/LuaTelemetry/wiki/Tips-&-Common-Problems

So it's out of memory.  Remove any other Lua scripts.  I guess it's possible that the X9 Lite has less memory?  Maybe the protocol differences uses more memory making the X9 Lite not appropriate for Lua.  It's also possible that the OpenTX team hasn't sorted the memory out yet.

P-I-Engineer, 2019-07-08:
thanks for response. I checked all your recommendations before posting.
I have no other LUA scripts on the radio.
There are only 5 models.
I've tried a couple different firmwares, my guess is the same as yours, opentx issue.

my spirit flybarless lua script works though. I added and tried that after seeing your script not work.

teckel12, 2019-07-08:
@P-I-Engineer Lua Telemetry uses a LOT of memory.  You could try using the latest `logs` build which I have done a lot of memory cleanup.  To upgrade to the logs build, use the following instructions:

1. Download [LuaTelemetry-logs.zip](https://github.com/iNavFlight/LuaTelemetry/archive/logs.zip)
1. Open or extract the `LuaTelemetry-logs.zip` file
1. Open the `LuaTelemetry-logs` folder and navigate to the `dist` folder
1. Copy the contents of the `dist` folder (`SCRIPTS` and `WIDGETS` folders) to the transmitter's SD card's root

Make sure all other Lua scripts are not on any screen, reboot Tx.  You may be able to use it, but then it will throw this error when entering the config menu or going to a particular view (the pilot view uses the most memory).  Noting where it throws the error would also be helpful.

P-I-Engineer, 2019-07-08:
same error, doesn't startup.

is it because it calls to figure out what radio it is, and it's not being identified correctly?

teckel12, 2019-07-08:
@P-I-Engineer Good point!  Let me see if I can simulate the X9 Lite

teckel12, 2019-07-08:
@P-I-Engineer I had to use OpenTX 2.3 to simulate the X9 Lite and the OpenTX 2.3 companion is really jacked up.  The transmitter detection wasn't correct in Lua Telemetry, but it didn't cause any harm either.  I updated the `logs` branch do detect the X9 Lite but I don't think that will resolve the issue.

As OpenTX 2.3 is a pre-release, my guess is one of two things:

1) This will be sorted out by the time OpenTX 2.3 is released
2) OpenTX 2.3 reduces the amount of memory available to Lua scripts for all Taranis transmitters, rendering Lua Telemetry no longer functional on Taranis

As a fun experiment, maybe try the following to see what kind of memory is available before running any Lua scripts:
1) Remove Lua scripts from all pages
2) Cycle power
3) Long-press `Enter`
4) Select `Statistics`
5) Press `Page Up/Dn`
6) Note the `Free Mem` available

P-I-Engineer, 2019-07-08:
lua scripts [D]70[I]70
free memory 63236b

teckel12, 2019-07-09:
@P-I-Engineer On my Q X7
Free Mem: 78184b

Maybe you can use OpenTX firmware with all the `no` options selected (for example `noheli` and nomulti module  or whatever it's called). You can also uncheck both the `lua` and `luac` options. Then see if the available memory (with no Lua scripts on any pages) changes.

P-I-Engineer, 2019-07-09:
nightly N41 is working! cool, a step in the right direction.

i have it set
-no heli
-multimodule
-lua
-luac

teckel12, 2019-07-09:
@P-I-Engineer You can uncheck multimodule and lua too. The lua option is only for lua scripts for models.  I know it seems strange to disable the lua option, but only the luac option is needed.

I believe all of the options that start with `no` will reduce the memory usage a bit. On the Q X7 the three to select are `nooverridech`, `noheli` and `nogvars`. Select
 those 3 and `luac` for the most free ram.

r3dsn0wz, 2019-07-09:
I am having the same issue with OpenTX 2.2.3 and a Taranis QX7S.
Version 1.7.0 works flawlessly
Version 1.7.1 throws the mentioned error by P-I-Engineer

teckel12, 2019-07-09:
@r3dsn0wz It's one of two things:

1) Out of memory on your transmitter (you need to remove any other lua scripts from the other pages)
2) You don't have `luac` checked in your transmitter firmware

P-I-Engineer, 2019-07-09:
i believe my issue was related to luac not being checked. I'm assuming it wasn't checked when Frsky released their firmware.

teckel12, 2019-07-09:
@P-I-Engineer Awesome!  I made some changes to the readme and Wiki to insure it's known to check the `luac` build option (which was an assumption mistake on my part).

r3dsn0wz, 2019-07-11:
I just updated to opentx 2.2.4 which has the luac build option checked.
Now everything is working again

teckel12, 2019-07-12:
@r3dsn0wz Just an FYI, it seems v1.7.1 of Lua Telemetry requires the luac build option (I didn't realize this).  v1.7.0 and v1.7.2 won't need the luac build option checked.  I made a change in v1.7.1 for my own debugging and didn't realize this at the time.  But, it's been corrected in the next release (or in the the `logs` branch if you want to use that).

P-I-Engineer, 2019-07-14:
roller wasn't working in 1.7.1 also, it works in logs 1.7.2 though...fyi

teckel12, 2019-07-14:
@P-I-Engineer "roller"?

P-I-Engineer, 2019-07-14:
The + - buttons on x9 lite are a roller like the jumper t16

teckel12, 2019-07-14:
@P-I-Engineer Ah yes, It's detecting that now.  It seems in 2.3 of OpenTX they've created new control constants that will make it easier on my end for new transmitters.

PR #336 (2019-08-25): v1.7.2
* [Playback log files](https://github.com/iNavFlight/LuaTelemetry/wiki/Playback-Telemetry-Log-Files) (may run out of memory on Taranis)
* Runs without luac or lua build options (broke in v1.7.1)
* Refactored code for higher speed and reduced memory usage
* Fix for french translation of OpenTX with the `RxBt` sensor labeled `BtRx`
* Crossfire: Add vertical speed option
* Crossfire: Fix for Q X7 Tx battery options
* Crossfire: Fix distance calculation errors
* Crossfire: Resolved edge-case yaw issue due to Crossfire rollover bug
* Horus: Home indicator now shows as a HUD indicator instead [... cut]
closed 2019-08-25 by teckel12