teckel12/LuaTelemetry #332: I'm using a Taranis X9D, with FW 2.2.3, (bootloader included). The good news: I can see the iNav screens as expected. The bad news: as soon as I assign Scripts>iNav to the display (screen 3), the transmitter becomes really slow, both, in accepting push button commands, servo response, or trying to change the model. When a model change (without the iNav script screen) is achieved, that model works normally.

Richrfl, 2019-07-11:
### Describe your question

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. Should it be doing this?

### Expected behavior

### Screenshots

### Additional context

teckel12, 2019-07-12:
@Richrfl I've not heard of this problem.  What's on screen 1 and 2?   Also, you don't happen to have multiple Lua Telemetrys running or something strange like that (maybe one on screen 3 and another as a one-time script)?

In any case, this is NOT something that should be happening and it's not something others have reported.  Lua Telemetry runs at 20 fps on Taranis and OpenTX prevents lua scripts from using too many resources so this shouldn't be possible.  Sounds like some kind of conflict or something non-standard.

Be sure you're following the first-time install instructions here:

https://github.com/iNavFlight/LuaTelemetry/wiki/Installation

Richrfl, 2019-07-12:
Thank you for your response Tim. Based on your questions and comments, I did some "clean up" to start taking issues out of the way. I modified my Screen 1 to Scripts>iNav, Screens 2 and 3 to NONE.
Assuming the system may be trying permanently to poll the telemetry device and that was not powered, I connected the receiver and the MATEK board. I was able to see telemetry values in the Telemetry Page under Model, but still, apparently, the LUA is cycling with few interruptions that allowed me the move between the configuration pages. I had to power it off and re-load the models into the TX, with a different model as default, otherwise, I was not able to change the model selected.
My next intent will be to take screen 1 to NONE, delete all sensors, Discover them back and return the screen 1 to iNav. Will report later.
BTW, I do not have any other Lua loaded in the TX, and yes, I followed the installation in the link you pointed out. Now, though, the TX is not taking the long page to show me the screen 1, anymore.

Richrfl, 2019-07-12:
Well, didn't work either. I have a question for you anyway: how would the Lua react if for whatever reason one sensor fails to respond? Although I deleted from the telemetry screen all sensors related to the GPS, because I have not signal where I am, I wonder if that may be causing the Lua to loop, or if it's expecting the Board to be armed, which I cannot do if the GPS does not link with the satellites. 
May I be missing something in the SD card?

teckel12, 2019-07-12:
@Richrfl Lua Telemetry works fine if there's no telemetry or even no sensors. It's actually quicker as it doesn't need to do as much processing.

What version of INAV, OpenTX, and Lua Telemetry are you using? What telemetry protocol (S.Port or Crossfire)?

Richrfl, 2019-07-12:
OK, then I have to have something wrong. Answering your questions: 
iNav:
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074] 
Lua: 
INAV Lua Telemetry Flight Status for Taranis/Horus/Jumper T16 - v1.7.1
I'm using S.Port
I did follow the order in the instructions, with the exception that given that I was using telemetry before installing the Lua, Step 1 in the instruction was not needed. ( and telemetry was running as expected)
However, I missed one issue in Step 2 that was to rename the parameter 0420. Will change it and report results.

Richrfl, 2019-07-12:
No changes. Unless you have some other idea, I think its time for me to delete this model and define a new one from square 1.
Should I see the screens also in the companion simulator?

teckel12, 2019-07-12:
@Richrfl The OpenTX companion simulator also runs Lua Telemetry.  That's how I do all development.  I'll try a few things when I get home, but it's kinda a mystery.  I would like to figure out what's different, as this is a totally unique situation.  

We we don't get anywhere, I would like to get your radio and model settings.  See the following instructions for dumping your radio and model settings:

https://github.com/iNavFlight/LuaTelemetry/wiki/Issue-Template#radio-and-model-settings

Richrfl, 2019-07-12:
Well, I'm having some bigger problem then. The simulator is giving me Script syntax error. Althought I have selected luac y lua in the radio configuration. What I do not know, as I did this selection after using Companion (2.2.3) for few days, if it never loaded the correct sw support, or its looking in a different directory.  I'll look into the link you gave me to send you the data I have, and in the mean time I will review the directories issue.

teckel12, 2019-07-13:
@Richrfl Now that I think about it, you may not be able to use the companion unless you also have all the source files (ending with .lua) along side the luac files. It's for development, not testing, so there's probably other things that would need setting up (like the widgets folder and the build.lua file). So maybe it's not as easy to test in the companion as it's designed for development. 

I'm any case, you can just create a new blank model on your transmitter and set Lua Telemetry on screen one. It will happily run even without any telemetry sensors. It will easily run at 20 fps (the OpenTX max).

Richrfl, 2019-07-13:
Well, this doesn't look good. I don't have any dat file in the SD, anywhere.  On the other hand, I created a new model, but I still have to discover the sensors. Without at least have to sensors discovered the screen 1 shows for a couple of seconds and then comes with the script error. One thing you must know is that the original model files were created with OpenTX 2.0, and were converted to 2.2.0 and now 2.2.3.

Richrfl, 2019-07-13:
In other words, I don't have a clue where the models are stored in the SD.

teckel12, 2019-07-14:
@Richrfl Models are in the MODELS folder, but not sure why you need to mess with them. Not sure what's up with your setup, but maybe just start over?

Richrfl, 2019-07-14:
Well, you suggested me to send you some data and the link you sent me with instructions pointed me to a folder under Scripts. I think the idea was for you to load those models and see if you were able to make it work. Anyway, for the time being, let me try with the model from scratch, and we go from there.

teckel12, 2019-07-14:
@Richrfl I believe my instructions were to just dump everything using the companion. The only other thing was to dump the contents of your "SCRIPTS/TELEMETRY/iNav/cfg" folder, which would only exist if you actually got Lua Telemetry running (which may not apply to you). The instructions are general for anyone opening an issue, not specific to every situation. If you have not yet run Lua Telemetry and not yet gone into the config menu, you wouldn't have *.dat files in that folder, so you would just skip that step (the VAST majority of people will have files there as you are the only person ever not able to run Lua Telemetry).

Richrfl, 2019-07-17:
Well, Tim, it happens that I am the guy that does the field test of "foolproof" stuff. Because I am a real fool. LOL.
Anyway, I got some good news. I created a new model, made a sensor discovery, and voila! I got the Telemetry screen! I only have a small issue, with the roll sensor. I'll verify CLI settings and sensor naming, but I think we can give this case as closed and solved. Why did not work before, not sure, but I guess is something to do with models being created by a former OpenTX rev. WIll verify that later, but I already know what to do if something doesn't work as it should be. Thanks a lot for your help.

teckel12, 2019-07-17:
@Richrfl Glad a new model sorted things out.  It sounds like some OpenTX thing (which is triggered somehow by Lua Telemetry).

As far as your roll indicator.  The INAV default is to send Acc_X, Acc_Y and Acc_Z values.  With these values, Lua Telemetry calculates roll and pitch, but this is prone to accuracy issues.  I added a CLI switch in INAV to send actual roll and pitch via telemetry.  Enable this feature with `set frsky_pitch_roll = ON`.

See the install wiki: https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#add-telemetry-sensors-to-transmitter (specifically `2` item `iii`) for setting the CLI value (and do a discover telemetry sensors).  Also, renaming sensors as described will be super helpful.  After doing this, you can remove the Acc_X, Acc_Y and Acc_Z sensors as they're no longer sent with `frsky_pitch_roll` set to `ON`.

If you still do have a problem with roll, create a new issue which can greatly help the next guy with your same problem or question.

Richrfl, 2019-08-01:
Well, the reason that originated reported problem was finally found. In all my models, I have a 3 seconds timer that will request a screenshot. No doubt about, this timer  (timer 3) for some reason interferes with the Lua script and makes the system to become extremely slow. Once I eliminated the Special Function that was controlling the timer reset, the problem disappeared. A pity, because given that you cannot watch the plane and the screen at the same time, I was looking for having screenshots. Anyway, I'll use a small video camera to walk around this issue.
Thank you Tim for your invaluable help.

teckel12, 2019-08-02:
@Richrfl Yes, you can't use timer 3 with Lua Telemetry. Lua Telemetry uses timer 3 to detect when you do a flight reset. Until OpenTX exposes the flight reset event to lua scripts, you can't use timer 3 for other purposes.

Why do you need to take screenshots? I've never heard of someone don't that (nor even using timer 3 at all). Couldn't you just use timer 2 if you had to? 

BTW, you can playback your entire flight with Lua Telemetry, and at up to 10 frames per second (30 times the framerate). See the development build. Just log your telemetry and Lua Telemetry can play it back (you can also fast forward, rewind, pause).

Richrfl, 2019-08-02:
Well, no wonder...... Never thought about using the sim to review the screen. Much simpler and healthy.
I already deleted all the special functions related to timer 3 in all my models. I already tested the sim to view the screens in previous logs, (Telemetry Numbers) and it gives me what I need. Thanks a lot again. I think I may have some issue to run the Lua in the OpenTX. Will review your previous comments and try to get the lua running. Will report.

Richrfl, 2019-08-02:
I meant Companion rather than  OpenTX.

Richrfl, 2019-08-02:
Tim you stated this in a previous message (21 days ago):
"Now that I think about it, you may not be able to use the companion unless you also have all the source files (ending with .lua) along side the luac files. It's for development, not testing, so there's probably other things that would need setting up (like the widgets folder and the build.lua file). So maybe it's not as easy to test in the companion as it's designed for development."

In the widget folder, I have an iNav subfolder containing the files: main.lua and main.luac
In the Script folder I have the files: HUD iNav test.lua, HUD.lua and some old lua files.
In the subfolder Telemetry (Script>telemetry) I have the files inav.luac and inav.lua and a subfolder iNav, which contains 19 luac files, language folders and cfg that has a readme.txt file. Also a pics folder with 11 png files. 

Please, can you help me with these? is there or should I open another thread for this?
Thanks

teckel12, 2019-08-02:
@Richrfl There's no reason to use the OpenTX Companion to review a flight (which would require some development environment setup).

Instead, you can review a flight right from your transmitter using Lua Telemetry.  Use the latest development branch with the following these instructions:  https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

Once you're running the development version of Lua Telemetry (v1.7.2) you can review your logs by accessing the config menu, scrolling to the very last item, and selecting the date of the flight.  When you select the date, it will start playing back the flight on your transmitter.

Richrfl, 2019-08-04:
Tim, I followed the instructions on the link that you gave me, but I cannot replay the log. This is what I am doing: 
Power TX on
Long press for Radio Setup
Page to SD card
Open LOG folder
Select the Model where I have enabled the iNav script.
Nothing Happens.
Option 2 Long press on the log file. 
I have no options for run it.
Therefore, I am pretty sure I am missing something important, but I don't know what. I guess there has to be some process to run the log in the TX that I am not aware of. Remember I am the "fool proof tester".

teckel12, 2019-08-04:
@Richrfl Playback of a log file is a feature in the development branch of Lua Telemetry, not OpenTX.

First, you need to be running the development branch. A link to upgrading is listed above. You know your running the development branch because on startup of Lua Telemetry it will show version 1.7.2.

Then, from inside Lua Telemetry you go to the config menu of Lua Telemetry (depends on your transmitter how this is done, see the wiki if you don't know how). From the Lua Telemetry config menu, scroll to the last item which allows you to select a log file you wish to playback (it will show the latest 5 log files from the last 2 weeks for the model you have currently selected). Whe you select the date, it will playback the log file in Lua Telemetry.

For more details, see the pull request (the only open pull request at the moment). This also gives details on the Lua Telemetry feature for log file playback.

Richrfl, 2019-08-08:
Tim, I was looking into the Development branch, and I have a couple of questions about what I read:
1) The zip file includes a Telemetry folder with lua and luac files that seems the same as I installed before. Should I assume they are slightly different to allow the playback?
2) The zip file does not include the widgets folder that I got when I installed from the LUA Telemetry.zip.
3) Should I delete or overwrite all files I downloaded before for this project? (those that were included in the LUA Telemetry.zip)
Finally, and just to be sure, because I had a weird behavior in my last flight, although I was not using the iNav model: is it possible that the lua may be running, even if the model does not have it selected in any screen? or at least running some interrupt now and then?. I found out that suddenly I have delays between the stick and the plane reaction to the command.
Thank you

teckel12, 2019-08-09:
1. To install the development branch you want to use the [instructions here](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build).
    1. Basically, you open the zip file (or extract it) then open the `LuaTelemetry-development` then `dist` folder.  In that folder will be two folders `SCRIPTS` and `WIDGETS`.
    1. When upgrading you won't typically see new file names, they will be replacements to existing files.  So the fact that the file names are all or mostly all the same is just because it's an upgrade.  For this particular upgrade there are 3 new files: `log.luac`, `log_c.luac` and `log_s.luac`.  But **ALL** files need to be replaced, not just the new files (just as the instructions tell you to do).
2. The `LuaTelemetry-development\dist` folder in `LuaTelemetry-development.zip` does include the `WIDGETS` folder (`dist` contains a `SCRIPTS` and `WIDGETS` folder).  I would make sure you're following the [Upgrade to Development Build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build) instructions. There's really just 4 simple steps that work every time.
3. Just *Copy the contents of the `dist` folder (`SCRIPTS` and `WIDGETS` folders) to the transmitter's SD card's root* as noted on step 4 of the install instructions.  That would of course mean you're overwriting all files.  There's no reason to delete anything, when you copy the files it will delete the old copies and install the new.

Basically, a lua script can never cause a slowdown.  OpenTX doesn't allow a lua script to use more than about 1/20 of a second.  If it does, OpenTX will shut the lua script down.  Having to do with the subject of this issue, the problem wasn't a script slowing your transmitter down, it was the constant saving of screenshots (really designed to do from a physical switch, not triggered from a logical switch set to a timer).  Kinda cool, but SD access is notoriously slow on Taranis.

Anyway, no, if your model doesn't have lua scripts assigned to telemetry pages, they're not running or even loaded at all.  Lua scripts on telemetry pages **do** run in the background (even when not visible).  But only if that model has lua scripts assigned to telemetry pages.  If this model didn't have telemetry lua scripts setup, the slowdown is from something else.  Are you saving screenshots for this model too?  Any kind of SD card access should be looked at as that's a common reason for a transmitter slowdown.  Look at your logical switch and special functions.

It seems like you keep having slowdown issues, but other's don't.  The only (unique to you) difference I can see is the screenshots triggered via timer.  I've never heard of anyone doing that, sounds like a problem waiting to happen.

teckel12, 2019-08-09:
Maybe this image will help?  This is literally the only thing you need to do to update:

[image]

Richrfl, 2019-08-12:
LOL, Tim, I really call this foolproof. You are great!
I am putting this project on the fridge for some time. I am focusing on flying EDF now, and I don't want to mess up with anything else until I get proficient on EDF.
I am filing the link with all our exchange so that when I take this back I don't start from square one.
Thanks a lot for your help.

closed 2019-08-15 by Richrfl