teckel12/LuaTelemetry #333: icon and some element missing after install LuaTelemetry on JumpterT16

ltkrbkhd2, 2019-07-14:
After Install LuaTelemetry on JumperT16 the model name on the model selection page is not display. And the system configure page the slider is not display correctly as well. And if I remove the LuaTelemetry on the T16, the model name and the slider is showing correctly again. Any suggestion?

The pics shows the problem.
[https://imgur.com/a/jnn5vWU](url)

teckel12, 2019-07-14:
That's because of the way the Horus works as a widget.  Lua Telemetry is changing the font color to white, but for whatever reason, OpenTX changes that globally instead of locally to one widget.

In any case, it's happening because you're not setting what font color you want.  See the Horus install instructions here: https://github.com/iNavFlight/LuaTelemetry/wiki/Installation#installsetup-lua-telemetry-on-transmitter  Specifically, you missed step `g`.  As you're using a theme with a white background, you need to enable `Restore` to restore your theme's colors and set your theme's `Text` and `Warning` colors.  For example, you probably want a black text color and maybe a red warning color.

The other option is to just select a dark-colored theme, which already uses white text.

closed 2019-07-14 by teckel12

teckel12, 2019-07-15:
@ltkrbkhd2 FYI, the goal eventually is to create an INAV theme that would by default work well with Lua Telemetry (just as the dark blue one does now). This could also slightly increase speed and free up some memory. But I've had issues getting custom themes to work and have found little documentation.

ltkrbkhd2, 2019-07-15:
@teckel12 Thanks for the reply. I simply follow the instruction on the installsetup page and have now issue now. For me only the default theme works properly. The other two just flicking. Was trying to create custom theme, as you say there's no much document about it.

teckel12, 2019-07-15:
@ltkrbkhd2 AI believe you're saying you got it working. Is that correct?

The flickering sounds like you don't have the correct SD card contents for the version of OpenTX you're running. For example, upgrading OpenTX but didn't also update SD contents.

But, that's not a Lua Telemetry issue,  just trying to help.

wx4cb, 2019-07-15:
@teckel12 no i have the same flickering issue on the T16 when im using a theme that's NOT the default. it's an issue with opentx i think. I managed to make a theme that works for me but i still get it occasionally. the theme support is lacking for sure.

@ltkrbkhd2 I would suggest reaching out to jumper about that. I still haven't had any answer when i asked :D

teckel12, 2019-07-15:
The TBS theme is broken even on Horus. I use the dark blue theme on Jumper T16 and Horus X10S without any flicker. The TBS theme seems incomplete.