teckel12/LuaTelemetry #334: OpenTX 2.2.4 LuaTelemerty 1.7.1

dsaboot, 2019-07-18:
I've update Taranis QX7 to OpenTX 2.2.4
Update SD_card data to new one (delete all data before update)
Install LuaTelemetry 1.7.1
Only 1 lua script installed on my Taranis.
If I start script - "attempt to call a nil value"

I've also update to 2.2.4 my Horus X12S
And luaTelemetry 1.7.1 also didn't work on X12S.

At OpenTX 2.2.3 all was correct.

closed 2019-07-18 by dsaboot

teckel12, 2019-07-18:
@dsaboot Was this problem resolved?  If so, what was the solution?  Or, did you accidentally close the issue?

dsaboot, 2019-07-18:
Problem solved.
I've flashed Q7X with other list option - luac, nooverridech, nogvars

>
>@dsaboot Was this problem resolved?  If so, what was the solution?  Or, did you accidentally close the issue?

С уважением,

dsaboot@mail.ru

teckel12, 2019-07-18:
@dsaboot Ah yes!  v1.7.1 does require the `luac` build option.  The next release won't require `luac`.