teckel12/LuaTelemetry #338: NV-14 Nirvana

Rec0iL, 2019-07-24:
I tried using your script on the Nirvana NV-14 with a Lua script enabled Firmware (Betaflight scripts do work) and it says "Opentx 2.2+ required" when running the script although it is a 2.2 version. The widget just says "Disabled".
Would be happy to help with trying the builds.

teckel12, 2019-07-24:
@Rec0iL To get around the 2.2 version check, try [upgrading to the development build](https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build).  I happened to just remove that last night as everyone should now be running 2.2 anyway.

That will get you past that error.  The "Disabled" error means that it crashed.  As the Nirvana uses custom firmware, it's possible that something doesn't work.  Running as a widget adds another level of complexity (and difficulty tracking down the problem).  Running as a one-time script will be easier.

So...  After updating to the development build with the above instructions, then try running Lua Telemetry as a one-time script.  Long-press `Sys`, short-press `Page`, navigate to `SCRIPTS`->`TELEMETRY` and select `iNav.luac`.

I just thought of something.  It's possible that it doesn't work because you could have the `luac` build option turned off and v1.7.1 requires it.  v1.7.2 (the development build) doesn't require the `luac` build option so going with the development build is probably the best way to go anyway.

Rec0iL, 2019-07-28:
I've downloaded the latest version just now.
When i try running it it flashes for a brief moment white whit a bit of text
and then returns back to the menu.

For a split second i can make out the numbers 1.7.2 when running the iNav.luac.
And when running the iNav.lua file i see some numbers i think like 8.888888
Both have INav Lua Telemetry written above the mentioned text.

If a video helps I can make one.

Maybe there is some Lua stuff still missing on the Nirvana. I'm really unsure.

teckel12, 2019-07-28:
@Rec0iL Because Lua Telemetry is not designed for the screen size of the Nirvana, it's probably crashing because it's trying to draw outside the screen.

For the Nirvana to work, a custom Lua Telemetry view would need to be designed. It would be close to that of the Horus, but basically everything would need to be relocated.

It sounds like it's trying to run now, but it will never work without a totally custom designed view.

I can reach out to see if I can get a Nirvana transmitter for development. But honestly it will be difficult as I can't run a simulator of the Nirvana for development.

teckel12, 2019-07-28:
Possible Nirvana NV-14 view:

[image]

Just a rough idea, moving a few things around on the Horus.  Suggestions welcome.

Rec0iL, 2019-07-28:
This looks great! This is exactly how i immagined it!
Btw. I messaged Warrick Dainter and Jakub Klimasz to get in touch with you.
I hope you hear something soon from them!
In the meantime I'll happily continue to play guinnea pig for you :D

qba667, 2019-07-29:
@teckel12 here is Jakub, as you mentioned main issue is drawing out of screen bitmap - there should be some protection mechanism but I am not sure if it his working. 
To obtain radio you probably need to contact FlySky or UFPV - but I can tell how to run simulator. 
For me development on simulator is easier because of limited debug possibilities on radio.

teckel12, 2019-07-29:
@Rec0iL Here's another mock-up:
[image]
I've moved some things around so it will actually work (that altitude graph on the first draft would take far too long to render).  Also, stretching out the attitude/glass cockpit would require Nirvana-specific images.  While not difficult, it could be avoided and would also render faster.

In any case, while I wait to see if FLYSKY is interested in assisting, doing a few iterations of the design is helpful to work out the kinks.

Rec0iL, 2019-07-29:
Looking good, i like the switch to the smaller altitude graph and the bigger map!
Couldn't you unify the Fuel and Battery Bar? I mean they should idealy display both the same percentage. Oh well if your flying a gas plane then i guess not....

Also maybe a question to both of you and probably something for the far future: Would it be possible to use the internal bluetooth module of the nirvana to send the gps data to an arduino to create an antenna tracker?

teckel12, 2019-07-29:
@Rec0iL The fuel is based on the mAh used while the battery is the voltage. While they basically give the same info, they can be quite different due to voltage sag. I tend to pay more attention to the fuel (mAh used) and only use the battery voltage to monitor voltage sag.

As far as your Bluetooth question, sure it would be possible. However, it would be on FLYSKY's shoulders to develop it as this is currently not exposed. I would think there's a higher chance that Team Black Sheep would be able to do this as their transmitter module also supports Bluetooth and it would be an additional product to sell. If you're flying Crossfire, I'd reach out to Trappy as he may get a kick out of doing it. The trigonometry isn't that challenging, and everything needed is already in place.

qba667, 2019-07-30:
@Rec0iL  Bluetooth is little problematic  - as far I know there was 2 different chips used. One has only Chinese documentation so it will be hard to do anything. Second one is fine to use because there is English documentation. Are you aware of any existing implementation where it can be eventually tested? If there is anything existing it will be easier to use it as reference.

@teckel12 Let me know if you need help on implementation of UI.

FlyskySoft, 2019-07-30:
> @Rec0iL Because Lua Telemetry is not designed for the screen size of the Nirvana, it's probably crashing because it's trying to draw outside the screen.
> For the Nirvana to work, a custom Lua [...]
@teckel12 Hello, here is flysky, thanks for your interest in the development of Nirvana, if you need a Nirvana transmitter for development, you can add my skype account (Ankeryang), thank you!

qba667, 2019-07-30:
@teckel12 I invented FlySly to discussion, so you can get transmitter. Currently it seems I am only one who cares about touch interface of opentx - but maybe you will love it as well. I did some other LUA rendering already for this tx so in case of questions let me know.

Rec0iL, 2019-07-30:
@qba667 I know of the Ghettostation, but it takes input from a ground station reciever as far as I understood. In theory it would allow for any input. https://github.com/KipK/Ghettostation
If that is what you meant with a testing implementation.

teckel12, 2019-07-30:
@qba667  Touch interface is nice, even better if exposed to lua scripts.  Currently, with OpenTX on Horus, if running a script as a widget, there's ZERO interface with the widget through lua.  This is why you must use the sticks to navigate the config menu on Horus as a widget.  Really, controls as well as touch events should pass to widgets and lua scripts.  But, this is not currently the case.

There's an open ticket on [IBUS support for INAV Lua Telemetry](https://github.com/iNavFlight/LuaTelemetry/issues/312).  Not sure about your hacked version, but proper IBUS/INAV compatibility could be possible and the above ticket is where you can join that discussion.

@qba667 With Crossfire, I believe you can set the receiver to send MSP to the transmitter.  And the Crossfire transmitter module can output GPS and other data via Bluetooth (with several protocols).  So, it should be possible to connect a Crossfire transmitter module via Bluetooth to a Bluetooth-compatible Arduino-like device which then controls an antenna.  I haven't dove deep into this, but if you're running Crossfire, this is the route I would take as it seems like everything is already in-place.  And it would logical if you're doing long-range you're probably running Crossfire.

In any case, this is really not a Lua Telemetry issue at all.  It's something the Nirvana could do on it's own or I believe you can do today with Crossfire.

teckel12, 2019-08-25:
Received the FLYSKY Nirvana NV-14, upgraded the firmware and got things working with my Crossfire test model.  Working on Lua Telemetry now to get it to at least fire up.

ankeryang, 2019-08-26:
Perfect, Nirvana NV-14 will be better under the efforts of everyone, thank you for everyone who works hard for this!

ankeryang, 2019-08-27:
> 玻璃驾驶舱工作：
> 
> [image]

So perfect!

teckel12, 2019-08-31:
Made more progress with Nirvana NV-14 support.  I did run into a few snags.

First, it seems `lcd.drawRectangle` isn't working.  I don't use it much, but I needed to build a hacky work-around.  Probably an easy fix for FLYSKY, but will require a new firmware.

Secondly, as there's no menu control buttons/dial on the transmitter (as it's a touch interface), there's no event constants either.  However, I use the event constants like the following in HORUS:

```
EVT_SYS_FIRST = 1542
EVT_ROT_LEFT = 57088
EVT_ROT_RIGHT = 56832
EVT_ENTER_BREAK = 514
EVT_EXIT_BREAK = 516
```

Had to build a hacky working-around for this too, which I'm not sure how I'll be able to get around.

Finally, it seems `lcd.drawFilledRectangle` is slow (and maybe other lcd functions).  I know a recent version of OpenTX optimized all of the lcd functions making them much faster, but maybe these changes didn't make it into the Nirvana firmware.

I'll see about contacting FLYSKY and reporting the `lcd.drawRectangle` issue, suggestion on optimizing the lcd functions, and maybe see if they've considered the event constants that scripts commonly use.

qba667, 2019-08-31:
@teckel12 drawRectangle  is is throwing invalid argument count?
There are virtual events defined related to touch gesture, or trim buttons that can be used in LUA.  I will provide you details tomorrow at at Monday.

teckel12, 2019-08-31:
@qba667 No invalid argument count, it seems to try to work, but fails to render anything or kind of tries to do a fill rectangle. It's also extremely slow. I couldn't identify the problem so I built a rectangle replacement function. I'll try a few small rectangles and see if I can diagnose it further.

Touch events would be awesome if they were exposed. If it works well, using the touch interface completely for the config menu would be possible.

Thanks.

teckel12, 2019-08-31:
Most everything is working now.  RSSI/Link quality will go in the empty upper right corner.  The altitude graph is untested (turned off in the below sample).  But it's about time to try some test flights now.

[image]

teckel12, 2019-09-01:
Seems to work, merging into the development branch if anyone wants to test

[image]

qba667, 2019-09-02:
@teckel12 here you have list of available event used in latest firmware (but they are there since month). For trims used as joysticks - they have also click function - it iscompatible with current OpenTX.

`
  { "EVT_PAGEUP_FIRST",  EVT_KEY_FIRST(KEY_PGUP) },
  { "EVT_PAGEUP_LONG",  EVT_KEY_LONG(KEY_PGUP) },
  { "EVT_PAGEUP_BREAK",  EVT_KEY_BREAK(KEY_PGUP) },

  { "EVT_PAGEDN_FIRST",  EVT_KEY_FIRST(KEY_PGDN) },
  { "EVT_PAGEDN_LONG",  EVT_KEY_LONG(KEY_PGDN) },
  { "EVT_PAGEDN_BREAK",  EVT_KEY_BREAK(KEY_PGDN) },

  { "EVT_TELEM_FIRST",  EVT_KEY_FIRST(KEY_TELEM) },
  { "EVT_TELEM_LONG",  EVT_KEY_LONG(KEY_TELEM) },
  { "EVT_TELEM_BREAK",  EVT_KEY_BREAK(KEY_TELEM) },

  { "EVT_MENU_FIRST",  EVT_KEY_FIRST(KEY_MENU) },
  { "EVT_MENU_LONG",  EVT_KEY_LONG(KEY_MENU) },
  { "EVT_MENU_BREAK",  EVT_KEY_BREAK(KEY_MENU) },

  { "EVT_RTN_FIRST",  EVT_KEY_FIRST(KEY_EXIT) },
  { "EVT_EXIT_BREAK",  EVT_KEY_BREAK(KEY_EXIT) },

  { "EVT_ROT_BREAK", EVT_KEY_BREAK(KEY_ENTER) },
  { "EVT_ROT_LONG", EVT_KEY_LONG(KEY_ENTER) },
  { "EVT_ENTER_BREAK", EVT_KEY_BREAK(KEY_ENTER) },
  { "EVT_ENTER_LONG", EVT_KEY_LONG(KEY_ENTER) },

  { "EVT_ROT_LEFT", EVT_KEY_FIRST(KEY_RIGHT) },
  { "EVT_ROT_RIGHT", EVT_KEY_FIRST(KEY_LEFT) },

  { "EVT_PLUS_FIRST", EVT_KEY_FIRST(KEY_UP) },
  { "EVT_PLUS_BREAK", EVT_KEY_BREAK(KEY_UP) },
  { "EVT_PLUS_LONG", EVT_KEY_LONG(KEY_UP) },
  { "EVT_PLUS_REPT", EVT_KEY_REPT(KEY_UP) },
  { "EVT_MINUS_FIRST", EVT_KEY_FIRST(KEY_DOWN) },
  { "EVT_MINUS_BREAK", EVT_KEY_BREAK(KEY_DOWN) },
  { "EVT_MINUS_LONG", EVT_KEY_LONG(KEY_DOWN) },
  { "EVT_MINUS_REPT", EVT_KEY_REPT(KEY_DOWN) },
`
Touch events - not yet approved by OpenTX, for EVT_TOUCH_UP and EVT_TOUCH_DOWN, they are additional 2 parameters passed to script - x coordinate and y coordinate. 
`
  { "EVT_SLIDE_LEFT", EVT_TOUCH(TOUCH_SLIDE_LEFT) },
  { "EVT_SLIDE_RIGHT", EVT_TOUCH(TOUCH_SLIDE_RIGHT) },
  { "EVT_TOUCH_UP", EVT_TOUCH(TOUCH_UP) },
  { "EVT_TOUCH_DOWN", EVT_TOUCH(TOUCH_DOWN) },
`
Virtual events (from keyboards/dialogues displayed in LUA scripts) - also not yet approved:
`
  { "EVT_VK_MIN", EVT_VK(VKEY_MIN) },
  { "EVT_VK_MAX", EVT_VK(VKEY_MAX) },
  { "EVT_VK_INC", EVT_VK(VKEY_INC) },
  { "EVT_VK_DEC", EVT_VK(VKEY_DEC) },
  { "EVT_VK_INC_LARGE", EVT_VK(VKEY_INC_LARGE) },
  { "EVT_VK_DEC_LARGE", EVT_VK(VKEY_DEC_LARGE) },
  { "EVT_VK_DEFAULT", EVT_VK(VKEY_DEFAULT) },
`

For dialogues it should be also EVT_VK_DIALOG_RES_OK and EVT_VK_DIALOG_RES_CANCEL but they are part of popup interface and currently not exposed as string to LUA. 
What more it is possible to display a keyboard that provides input possibility - but it is also not approved by Opentx yet.

qba667, 2019-09-02:
@teckel12 would it be not powerful if we would be able to repaint only invalidated region of LUA screen? (In NV14 it is a little problematic because double display buffer is used - so we need to copy it first and then paint over desired region if needed , then swap buffers again.

hugsgui, 2019-09-02:
Hi there! I tried to launch the developement build of iNav LUA on my Nirvana, it started,  but when i plugged the quad, and the telemetry connect again, the LUA is closing. If I let the quad plugged and try to launch the LUA, nothing happen... It really seem to be link with telemetry, any idea why?

teckel12, 2019-09-02:
@qba667 It seems none of the input events are exposed to Lua when running as a widget (as INAV LuaTelemetry is designed to work). Only one-time Lua scripts can get the input events. I verified this also on the Nirvava. I believe this was a design decision by OpenTX (a poor decision in my opinion).

teckel12, 2019-09-02:
@qba667 I tried this technique on the Horus X10S and it didn't work at all. The reason is that for each page refresh, OpenTX first paints all the standard display stuff. Then, the lua routine runs and it needs to start with a full clear screen. If not, the OpenTX display will be seen, not the previously generated screen the lua script painted.

I can try this on the Nirvana, but if it works at all like on the Horus, it won't work at all.

teckel12, 2019-09-02:
@hugsgui I ran with crossfire telemetry without an issue. I haven't tried S.Port yet, but I may be able to with a multi-protocol transmitter module that I have. IBUS won't work as it's not supported with Lua Telemetry at this time. What telemetry protocol are you using? Also, are you running the latest firmware?

If you could enable telemetry logging via the arm switch, could you also supply your telemetry logs? Make sure all telemetry sensors are set to log (that's the default). With this, I'll be able to see all the telemetry sensors you have, what their values are, and hopefully reproduce the issue.

teckel12, 2019-09-02:
@qba667 I just tested, the Nirvana "cloud" background is painted first on every screen refresh, so if I don't update the entire screen each time, the cloud shows through and flickers. This is identical to what happens with a widget on every transmitter I've tried (Horus X10S, Jumper T16 & Nirvana NV14). Therefore, the entire page needs to be printed every time.

It sounds like OpenTX shouldn't paint the screen (or it be an option) when using a full-page widget. Similar to exposing the events to widgets, it seems OpenTX needs to be convinced to implement a few features.

What I hear from OpenTX is that a full UI rewrite is in the works, and these types of things will be addressed at that time, so there's no interest in enhancing widgets currently.

teckel12, 2019-09-02:
@hugsgui I tried but I can't test S.Port on the Nirvana with my multi-protocol module. The Nirvana doesn't have a multi-protocol option like on the Jumper T16, so I can't bind.

I may be able to fake it by setting up all the S.Port telemetry sensors and playing back a log file (which simulates the telemetry). But, that still doesn't touch all the code. So it may not hit the place which is causing your error.

Another option is to get you a special version that's not compiled which should give you an error message instead of simply terminating.  It would also need to be run as a one-time script to see the message, which I haven't even tested on the Nirvana.

hugsgui, 2019-09-02:
Hi teckel12!
I'm running a crossfire full tx, and I update the tx with the last flysky firmware, so I'll try to send you a telemetry log when i can.
Thanks for your answer.

teckel12, 2019-09-03:
@hugsgui You also may want to be sure your running Crossfire 3.0 (I believe it's pre-release). There have been some bugs in recent versions of Crossfire (pre 3.0).

qba667, 2019-09-03:
> the input events are exposed to Lua w

That is right - 2.4 is addressing it - but we can do it in NV14 in current version - it is just about passing event to active widgets. Later I will add experimental version.

qba667, 2019-09-03:
> @qba667 I just tested, the Nirvana "cloud" background is painted first on every screen refresh, so if I don't update the entire screen each time, the cloud shows through and flickers. This is [...]

That can be addresses as well, widget zone should be defined as transparent or not and only redrawn if needed. The problem is known new "windows" approach will solve it - I can work son something experimental for NV14 as well. What we need to is way to get if widget must be repainted or not (invalidated flag). 
I

teckel12, 2019-09-03:
@qba667 I believe what makes sense would be to allow an additional option when a single widget view is selected.  If a single full-screen widget is selected, there would be an option to not repaint the background.

Also, there's no reason why key events shouldn't be passed to all widgets.  I can understand the argument that it should only happen for full-screen widgets, but I think that in those rare cases the user may need to control what widgets go on what pages if there is a conflict.

I've had a few conversations with @lshems about this in the following OpenTX issues:

https://github.com/opentx/opentx/issues/3821
https://github.com/opentx/opentx/issues/6415

Also, there's discussion about how reset telemetry and reset flight should be put in the event stream along with key events.  This would also allow this vital information to be exposed to widgets (and one-time scripts on Taranis for that matter).

teckel12, 2019-09-03:
@qba667 One more thing, I couldn't get lcd.drawRectangle() and lcd.drawGauge() to work on the NV14. I tried a few things, but it would either not do anything or print lines everywhere.  I wrote my own replacements so it's not a blocker, but should be looked into as they are common functions.

hugsgui, 2019-09-03:
> @hugsgui You also may want to be sure your running Crossfire 3.0 (I believe it's pre-release). There have been some bugs in recent versions of Crossfire (pre 3.0).

Hi! I'm Running the last version stable, 2.93, how can i flash the 3.0 and does it need some specific things to know befor doing it?

teckel12, 2019-09-03:
@qba667 I see you fixed `lcd.drawRectangle()` on Aug 11.  I'm just running older firmware.

teckel12, 2019-09-03:
@hugsgui There's some switch or something that allows you to use pre-release firmware.  I'm not at my workbench to let you know exactly the method.  There's no breaking code in 3.0.  I just did a flash and fly.

qba667, 2019-09-03:
@hugsgui  you need AgentX to flash new firmware.
@teckel12  `lcd.drawGauge()` fixed now as well. I will check for you passing events and not drawing background - probably we need to draw white one at lest once after view is selected but that should be all, or we can add it to widget config - probably that will be more generic solution.

teckel12, 2019-09-03:
@qba667 A single white screen would be fine.  Although, thinking about how this would work at the lua script level, the lua script would need to clear the part of the screen that needs to be updated with new content.  And currently the `lcd.drawFilledRectangle()` function is fairly slow when filling a large area.  I've found that it's MUCH faster to have a bitmap with a large solid color than have the script do a `lcd.drawFilledRectangle()`.  This may just be how it is.

In any case, what would be idea is that it would only update values that changed.  But in reality, it would need to be done in more of a group because erasing the old data with `lcd.drawFilledRectangle()` which probably end up not saving any speed as the function is so slow when filling larger areas.

hugsgui, 2019-09-04:
@teckel12 I finally found my issue, after flashing the 3.0 it was not working, but i try to create a new model and the LUA was working on it, so i found why, when i deleted all discovered sensors on my first model, and search for new sensor, then it work. It's nice to known, cause i may not be the one to try this with existing models, so it may become a known bug.

teckel12, 2019-09-04:
@hugsgui It's not the rediscovery that solved it, there must be a difference between the sensors on the old model and the new model.  Basically, it's not that a rediscovery is the solution, the solution is to correct whatever was wrong with probably a single sensor.

If you could compare the sensors with the old and new model, it would be nice to know what the difference is.  There shouldn't be a situation where the script crashes, no matter what the sensors are.  So it would be nice to know what edge-case the old model was in that cause it to prevent it for others.

hugsgui, 2019-09-05:
@teckel12 I feel bad that I deleted all sensors, so I can't help to know where come from the bug...
But I have another issue now, the minimap isn't showing the home point, althougt my OSD is, and i have 12 sat and gps fix, RTH working... does it work on yours?

teckel12, 2019-09-05:
@hugsgui it won't show the home position till you move away from home a bit. Before or at launch, it will just show the craft orientation.

Basically, it's to avoid icons being on top of each other and hard to view. You're to assume that if you just see the model with no home icon, you're really close to home.

qba667, 2019-09-06:
@teckel12 
I think we are getting event support in official OpenTX as well- I found that @raphaelcoeffic already implement it. I will check it and provide you build for NV14 as well.

raphaelcoeffic, 2019-09-06:
Is that about the events in widgets?

teckel12, 2019-09-23:
@raphaelcoeffic Yes, events being exposed to widgets. Is this an OpenTX 2.3 pull request?

raphaelcoeffic, 2019-09-23:
https://github.com/opentx/opentx/pull/6688

qba667, 2019-10-09:
@teckel12 
I added my own version of propagation of events to widgets - it is based on @raphaelcoeffic change.
Touch events contains additional parameters for x and y. Currently I am forwarding those events to all widgets on screen, coordinate translation should be done before in case we touch widget.

Standard events are also passed - currently trim buttons are not translated to keys - because this is allowed only when full screen lua is running - but touch should be sufficient for now.

teckel12, 2019-10-11:
@qba667 What about only doing one screen refresh (to white) initially for full-screen witgets? So only changed values need updating (or painting only parts of the screen). 

I believe this could greatly increase the frame rate.

qba667, 2019-10-11:
@teckel12 working on it right now - but I need to synchronize it with official OpenTX because it contains already  some changes in "Windows" implementation. We should add probably configuration option for widget  - so it will not applied to widgets. Btw I have crazy idea but I did POC implementation long time ago - rendering Open Street map on MCU.

qba667, 2019-10-11:
The other option to get performance  kick would be exposing separate layer to widget - all paints will be allowed only by owner widget. On refresh the layer can be painted on top of other layers. This seems to be simplest approach to me right now.

qba667, 2019-10-15:
@teckel12  Today I made first attempt to run the script as widget with new events - events are forwarded correctly. I looked at hardware keys you used, we need concept at this place. 
We can use gestures (swipe left/right is changing custom screen) so this use is limited, or define touchable menu button so we can trigger all actions via touch interface. 
The background part reinquires a little bit more experimenting, the Widget must extend class Window,to do that I need to solve first other issue in layout - currently main view is painted constantly without chancing if areas get invalidated. To make it right, I need to reimplement all items that can be displayed on main screen.

spinkb, 2019-10-23:
When running the script...how do you then quit, exit...to get back to the normal Nirvana screens?

I can't find any button combo that does it...

teckel12, 2019-10-24:
@spinkb It's a swipe jesture. Like swipe left or right? It's the transmitter not my script that controls this.

spinkb, 2019-10-24:
They said its left trim button press...but your script has that as the exist for the settings menu...so its in conflict.  So there is no way to exit.

Swipe seems to work in others, but not iNav...and left trim button press is already captured, so its out too.

davidabevan, 2019-10-24:
@teckel12 
Hi Nice to speak to you.
Thanks for your great work trying to improve functionality of NV14.
I to am trying this on my NV14, Have had much success with the Inav lua on QX7 etc,
My Nv14  is  running  "nv14_firmware-11102019-commit8d48d05-txpf-BL"  from the nv14 updater, re Stefan 73 RC groups.
I also cant seem to exit either for some reason.
BF lua you exit by a press down on left trim button. I have been starting your script by going to radio SD Card, Scripts, Inav Lua. Hold and execute.
Cant exit with a swipe on screen or any other way but reboot so far, Am I executing your code correctly, maybe our firmware is different. Also i am new and getting to grips with this particular radio
Advice welcomed please.
your work is greatly appreciated making our great hobby even better.
Kind Regards Dave...

teckel12, 2019-10-26:
@spinkb Lua Telemetry doesn't listen to the trim buttons (only T6 which only exists on Horus). And even if I did, it doesn't "capture" a button press.

What are you trying to do exactly? Exit Lua Telemetry? If so, you just do what you would do in any widget on the Nirvana, you swipe right to exit a widget.

If you're having problems exiting a widget, ask for help from FLYSKY, as it's not the script causing the problem. That is, unless I'm not understanding you correctly.

teckel12, 2019-10-26:
@davidabevan I'm using a release version of the Nirvana firmware flysky-NV14-2.1.7 from 2019-06-03, not something downloaded from someone on a forum. That's probably your problem.

As the install documentation states, the script is designed to run as a widget. It doesn't run at all as a one-time script with the firmware I'm using (2.1.7).

[20 more replies]

PR #345 (2021-02-17): v1.7.3
- Adds support for FLYSKY Nirvana NV-16 transmitter
- Adds support for X9D+ 2019 edition
- Small adjustment for OpenTX 2.3 compatibility
- Refactored many things for faster speed and reducing memory used
- Crossfire: Satellite count sanity check

Fixes: #344 
Closes: #338 
Fixes: #347
closed 2020-08-29 by stale[bot]