teckel12/LuaTelemetry #341: Crossfire telemetry FM  !ERR

Jetrell, 2019-08-13:
I vaguely remember reading something about this before. And am unsure if it has always been a problem with the X9D and CRSF.
Is this a Crossfire problem or LuaTelemetry ?

Thanks

[image]

wx4cb, 2019-08-13:
if it's in the opentx telemetry screen as that, then it's not a lua telemetry issue. that's what's being sent from the FC. Have you tried deleting all the sensors and rediscovering them?

Jetrell, 2019-08-13:
> 
> 
> Have you tried deleting all the sensors and rediscovering them?

Yep, I have. But it still remains. Sometimes it will not be a problem and the FM telemetry will work correctly. Then other times, this appears.

teckel12, 2019-08-13:
@Jetrell When `FM` (flight mode) is coming directly from INAV, nothing to do with Lua Telemetry.

`FM` shows not only the flight mode (`ACRO` and `RTH` for example), but also the status.  There's 3 status values for `FM` other than flight modes:

- `OK` = Ready to arm
- `WAIT` = Waiting for GPS lock
- `!ERR` = Arming disabled

Basically, `!ERR` means INAV is reporting that arming is disabled, here's the code in INAV:

```
    } else if (isArmingDisabled()) {
        flightMode = "!ERR";
    }
```

Jetrell, 2019-08-13:
That explains it. I got confused with seeing it in iNav and not with other flight software using CRSF. And I couldn't work out why it played up a few times in iNav as well. This left me wondering, At least I now know how it should act.
Thanks for the clarification Tim :-)

closed 2019-08-14 by Jetrell