teckel12/LuaTelemetry #342: same ol' disconnected but i think i'm connected issue

wx4cb, 2019-08-13:
ok @teckel12 so i had the same thing happen the other day. even rolled back to 2.95 and not 3 but the same thing happened.

everything was fine. landed as normal... carried the plane back and my voice alert said "disarmed". as soon as I yanked the battery I got "telemetry lost" (correct, obviously) and then the lua script said "armed" (or whatever it says) started with the haptic feedback and beeping thinking that it (i'm guessing) is a failsafe (all icons red etc).

If you look at the attached log, you'll see everything at 0 where I disconnect the battery and then it goes back into weird data. It's not an SD card issue - New card, i've deleted and recreated the model. I am at a loss what it can be. It doesn't happen every time.

any thoughts? If there's a specific testing procedure you'd like me to run through i can do that. i'm not saying it's a lua issue, just confused as to what's causing it, because to stop it i have to restart the tx.

LOG (Remove .txt extension):
[Harrier-2019-08-10.csv.txt](https://github.com/iNavFlight/LuaTelemetry/files/3498809/Harrier-2019-08-10.csv.txt)

teckel12, 2019-08-14:
@wx4cb I've seen similar issues with Crossfire (not a problem with S.Port). When you disconnect the power, the Crossfire Rx can send some final bogus info. Or, maybe the transmitter module intrepets it wrong. In any case, it could be anything. I've received both armed and low RSSI, for example, which causes Lua Telemetry to give you an alert.

The problem with the Crossfire telemetry logs is that OpenTX has a bug that doesn't record the `FM` sensor value (it's always zero). So there's no way to see what's going on with the flight mode.

In any case, this is a combination of a bug in Crossfire where sometimes when you disconnect power it sends garbage, and an OpenTX bug which makes it difficult to document the bug to submit to TBS.

It's not much of a problem as it's only when the battery is removed and you can just reset the flight on your transmitter to make it stop. It's also not really an issue with Lua Telemetry, it's just reporting what Crossfire telemetry is outputting.

With this said, I do have an idea that could prevent the issue in Lua Telemetry.

teckel12, 2019-08-25:
@wx4cb I made a change in the just released v1.7.2 which could resolve this issue.  Try it out and let me know if it works any better.

closed 2019-09-24 by teckel12